open62541/open62541 #4807: Method call timeouts

tboeckel, 2021-11-30:
Is there a way to define the maximum call timeout for methods? UaExpert has a default timeout of 10 seconds, but this timeout can be changed via Settings -> Configure UaExpert -> General.CallTimeout

The point is that we have some PLC based servers which take a quite reasonable amount of time to handle a method call and eventually return a successful result. The current default timeout for method calls in open62541 seems to be around 3 seconds, which is definitely too short for these slow PLCs.

So, is there a way to modify this timeout? And where is it defined at all?

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.2.2
 - [X] Other OPC UA SDKs used (client or server): UaExpert 1.5.1 and 1.6
 - [X] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

PiotrBzdrega, 2021-11-30:
I noticed such a delay in the examples, I don't know if that's what you mean.

>     
    UA_Client *client = UA_Client_new();
    UA_ClientConfig *cc = UA_Client_getConfig(client);
    UA_ClientConfig_setDefault(cc);
    /* we use a high timeout because there may be other client and
    * processing may take long if many method calls are waiting */
    cc->timeout = 60000;

jpfr, 2021-11-30:
You can set a timeout in the client settings.
Additionally, if you have long running methods, I'd advise to use async methods (requires multithreading enabled) in the server.
So that you can process other requests in parallel to the long-running method.

tboeckel, 2021-12-01:
Thanks! That is exactly what I was looking for!

closed 2021-12-01 by tboeckel