open62541/open62541 #4813: Problem with communication recovery (BadSecureChannelIdInvalid)

AlejandroAntonio1998, 2021-12-01:
Hello,

I´m trying to develop a code to **recover connection** after losing it. My **client configuration** is as follows:
```
	cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
	cc->timeout = 10 * 60 * 60 * 1000;  // ms
	cc->secureChannelLifeTime = 15 * 60 * 1000;  // ms
	cc->requestedSessionTimeout  = 30 * 60 * 1000 ;  // ms
	cc->connectivityCheckInterval = 2000;  // ms
```

My **function** to check recovery is:
```
        void testCheckComunicacionOK () {        
	        establecerComunicacion(ClienteOPCUA);
	        while (true) {
		        actualizarBitDeConexion(ClienteOPCUA);
		        std::this_thread::sleep_for(std::chrono::seconds(1));
		        UA_Client_run_iterate(ClienteOPCUA, 1000);
	        }
        }
```
Where:
      **ClienteOPCUA** -> is a UA_Client* global variable
      **establecerComunicacion(ClienteOPCUA)** -> function to create the connection
      **actualizarBitDeConexion(ClienteOPCUA)** -> function to change a bool at the server

**ISSUE**: If I delete the connection (removing the ethernet wire) and after a few seconds I set it on again it works fine.
If I wait more time (about more than half a minute) I get these messages and it cannot recovery:
```
[2021-12-01 12:09:40.723 (UTC+0100)] warn/channel       Connection 3 | SecureChannel 359576861 | Receiving the response failed with StatusCode BadConnectionClosed
[2021-12-01 12:09:40.723 (UTC+0100)] info/client        Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good
[2021-12-01 12:09:41.723 (UTC+0100)] info/client        SecureChannel must be connected before sending requests
[2021-12-01 12:09:42.724 (UTC+0100)] info/client        SecureChannel must be connected before sending requests
[2021-12-01 12:09:43.724 (UTC+0100)] info/client        SecureChannel must be connected before sending requests
[2021-12-01 12:09:44.724 (UTC+0100)] info/client        SecureChannel must be connected before sending requests
[2021-12-01 12:09:45.726 (UTC+0100)] info/client        SecureChannel must be connected before sending requests
[2021-12-01 12:09:46.731 (UTC+0100)] info/channel       Connection 3 | SecureChannel 359576862 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 and a revised lifetime of 900.00s
[2021-12-01 12:09:46.731 (UTC+0100)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2021-12-01 12:09:46.733 (UTC+0100)] info/client        Received a ServiceFault response with StatusCode BadSecureChannelIdInvalid
```
The last message keeps appearing every 1 second approximately.

Does anybody have any idea about why the created channel has an invalid ID??

Thank you all!!!

```
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_ENCRYPTION=ON -DUA_MULTITHREADING=199 
```
 - [ ] open62541 Version (release number or git tag): 
 - [ ] Other OPC UA SDKs used (client or server): Server with PLC from Siemens (S7-1500)
 - [ ] Operating system: LINUX 
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-12-01:
please enable debug logging for both server and client.
that makes it much easier to see what went wrong and where.

AlejandroAntonio1998, 2021-12-01:
How can I enable it??

jpfr, 2021-12-02:
Set UA_LOGLEVEL to 100.
For example with the ccmake gui tool.
[image]

AlejandroAntonio1998, 2021-12-03:
Hi, I´ve solved it with the **stateCallback** function.
The problem was that the Channel that was generated after lossing communication was not valid, so I check that situation with stateCallback and try to reconnect until the communication is recovered:
```
if ((channelState == UA_SECURECHANNELSTATE_CLOSED) && (sessionState == UA_SESSIONSTATE_CREATED)) {
		bool Comunicacion_Recuperada = false;
		while (!Comunicacion_Recuperada) {
			desconectarCliente(client);
			Comunicacion_Recuperada = establecerComunicacion(client);
		}
	}
```
I hope it may help someone.
Thanks for your time Julius!!

closed 2021-12-03 by AlejandroAntonio1998