open62541/open62541 #4821: Offer an example of using the library to get tag data

AlexanderZhirov, 2021-12-06:
I installed the library and built a client application based on the tutorial. It started and, as can be seen from the result, the connection was successfully received and the program was executed without errors.

```
[2021-12-06 10:16:35.187 (UTC+0300)] warn/userland  AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2021-12-06 10:16:35.194 (UTC+0300)] info/eventloop Starting the EventLoop
[2021-12-06 10:16:35.196 (UTC+0300)] info/channel   Connection 292 | SecureChannel 2 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2021-12-06 10:16:35.196 (UTC+0300)] info/client    Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2021-12-06 10:16:35.196 (UTC+0300)] info/client    Selected endpoint 0 in URL opc.tcp://WIN-RRMNF8HHDN5:62544 with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2021-12-06 10:16:35.196 (UTC+0300)] info/client    Selected UserTokenPolicy anonymous with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2021-12-06 10:16:35.197 (UTC+0300)] info/client    Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2021-12-06 10:16:35.197 (UTC+0300)] info/client    Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2021-12-06 10:16:35.198 (UTC+0300)] info/userland  date is: 6-12-2021 7:16:35.197

[2021-12-06 10:16:35.198 (UTC+0300)] info/client    Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
[2021-12-06 10:16:35.198 (UTC+0300)] info/eventloop Stopping the EventLoop
[2021-12-06 10:16:35.198 (UTC+0300)] info/eventloop The EventLoop has stopped
```

I have an existing server with data that I need to read over a time interval (for example, an hour) for a specific tag. The structure of the server is as follows:

[image]

And I need to read the data on the specified tag - **TNPZ.P_NEFT.R1.M**.
If I view the data through the native client going with the server, then I see this data, they exist. It is possible to see the dynamics on the chart and upload data in the form of a table.

[image]

The question is:
I need to get the same table, only using the library. I don't even know where to try to start, since I've never worked with OPC. I have difficulties understanding English, so I translated the specifications the OPC Foundation through a translator, but I absolutely did not understand the principle of operation.

From the attached code as an example for reading historical data, it is also not entirely clear how to use it, because it does not even compile for me:

```
10:34:54 **** Incremental Build of configuration Debug for project open62541 ****
Info: Internal Builder is used for build
gcc -std=c99 -DUA_ARCHITECTURE_WIN32 -DUA_ENABLE_HISTORIZING -O0 -g3 -Wall -c -fmessage-length=0 -o main.o "..\\main.c" 
In file included from C:/msys64/mingw64/include/open62541/client_config_default.h:12,
                 from ..\main.c:1:
C:/msys64/mingw64/include/open62541/client.h:397:18: error: unknown type name 'UA_HistoryReadResponse'
  397 | static UA_INLINE UA_HistoryReadResponse
      |                  ^~~~~~~~~~~~~~~~~~~~~~
C:/msys64/mingw64/include/open62541/client.h:398:56: error: unknown type name 'UA_HistoryReadRequest'
  398 | UA_Client_Service_historyRead(UA_Client *client, const UA_HistoryReadRequest request) {
      |                                                        ^~~~~~~~~~~~~~~~~~~~~
C:/msys64/mingw64/include/open62541/client.h: In function 'UA_Client_Service_historyRead':
C:/msys64/mingw64/include/open62541/client.h:399:5: error: unknown type name 'UA_HistoryReadResponse'; did you mean 'UA_ReadResponse'?
  399 |     UA_HistoryReadResponse response;
      |     ^~~~~~~~~~~~~~~~~~~~~~
      |     UA_ReadResponse
C:/msys64/mingw64/include/open62541/client.h:400:53: error: 'UA_TYPES_HISTORYREADREQUEST' undeclared (first use in this function); did you mean 'UA_NS0ID_HISTORYREADREQUEST'?
  400 |     __UA_Client_Service(client, &request, &UA_TYPES[UA_TYPES_HISTORYREADREQUEST],
      |                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                     UA_NS0ID_HISTORYREADREQUEST
C:/msys64/mingw64/include/open62541/client.h:400:53: note: each undeclared identifier is reported only once for each function it appears in
C:/msys64/mingw64/include/open62541/client.h:401:30: error: 'UA_TYPES_HISTORYREADRESPONSE' undeclared (first use in this function); did you mean 'UA_NS0ID_HISTORYREADRESPONSE'?
  401 |         &response, &UA_TYPES[UA_TYPES_HISTORYREADRESPONSE]);
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                              UA_NS0ID_HISTORYREADRESPONSE
C:/msys64/mingw64/include/open62541/client.h: At top level:
C:/msys64/mingw64/include/open62541/client.h:405:18: error: unknown type name 'UA_HistoryUpdateResponse'
  405 | static UA_INLINE UA_HistoryUpdateResponse
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~
C:/msys64/mingw64/include/open62541/client.h:406:58: error: unknown type name 'UA_HistoryUpdateRequest'
  406 | UA_Client_Service_historyUpdate(UA_Client *client, const UA_HistoryUpdateRequest request) {
      |                                                          ^~~~~~~~~~~~~~~~~~~~~~~
C:/msys64/mingw64/include/open62541/client.h: In function 'UA_Client_Service_historyUpdate':
C:/msys64/mingw64/include/open62541/client.h:407:5: error: unknown type name 'UA_HistoryUpdateResponse'
  407 |     UA_HistoryUpdateResponse response;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
C:/msys64/mingw64/include/open62541/client.h:408:53: error: 'UA_TYPES_HISTORYUPDATEREQUEST' undeclared (first use in this function); did you mean 'UA_NS0ID_HISTORYUPDATEREQUEST'?
  408 |     __UA_Client_Service(client, &request, &UA_TYPES[UA_TYPES_HISTORYUPDATEREQUEST],
      |                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                     UA_NS0ID_HISTORYUPDATEREQUEST
C:/msys64/mingw64/include/open62541/client.h:409:30: error: 'UA_TYPES_HISTORYUPDATERESPONSE' undeclared (first use in this function); did you mean 'UA_NS0ID_HISTORYUPDATERESPONSE'?
  409 |         &response, &UA_TYPES[UA_TYPES_HISTORYUPDATERESPONSE]);
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                              UA_NS0ID_HISTORYUPDATERESPONSE
..\main.c: At top level:
..\main.c:65:15: error: unknown type name 'UA_HistoryData'
   65 | readRaw(const UA_HistoryData *data) {
      |               ^~~~~~~~~~~~~~
..\main.c: In function 'readRaw':
..\main.c:66:61: error: request for member 'dataValuesSize' in something not a structure or union
   66 |     printf("readRaw Value count: %lu\n", (long unsigned)data->dataValuesSize);
      |                                                             ^~
..\main.c:69:35: error: request for member 'dataValuesSize' in something not a structure or union
   69 |     for (UA_UInt32 i = 0; i < data->dataValuesSize; ++i)
      |                                   ^~
..\main.c:71:29: error: request for member 'dataValues' in something not a structure or union
   71 |         printDataValue(&data->dataValues[i]);
      |                             ^~
..\main.c: In function 'readHist':
..\main.c:120:49: error: 'UA_TYPES_HISTORYDATA' undeclared (first use in this function); did you mean 'UA_NS0ID_HISTORYDATA'?
  120 |     if (data->content.decoded.type == &UA_TYPES[UA_TYPES_HISTORYDATA]) {
      |                                                 ^~~~~~~~~~~~~~~~~~~~
      |                                                 UA_NS0ID_HISTORYDATA
..\main.c:121:25: error: 'UA_HistoryData' undeclared (first use in this function); did you mean 'UA_SignatureData'?
  121 |         return readRaw((UA_HistoryData*)data->content.decoded.data);
      |                         ^~~~~~~~~~~~~~
      |                         UA_SignatureData
..\main.c:121:40: error: expected expression before ')' token
  121 |         return readRaw((UA_HistoryData*)data->content.decoded.data);
      |                                        ^

10:34:57 Build Failed. 17 errors, 0 warnings. (took 2s.919ms)
```

Is there an example for connecting to a similar server to read such data? Where can I even start to try to read data from the server or get the server structure? Are there any use cases?

AlexanderZhirov, 2021-12-06:
193 authors and none will answer?

AlexanderZhirov, 2021-12-06:
Why can't I compile this example? After all, this is the example that lies in the repository. There is no description of which flags to use. Why are data types not recognized? Where can I read about what is needed and where to replace it?

```
#include <open62541/client_config_default.h>
#include <open62541/client_highlevel.h>

#include <stdio.h>
#include <stdlib.h>

static void printTimestamp(char *name, UA_DateTime date)
{
    UA_DateTimeStruct dts = UA_DateTime_toStruct(date);
    if (name)
        printf("%s: %02u-%02u-%04u %02u:%02u:%02u.%03u, ", name, dts.day, dts.month, dts.year, dts.hour, dts.min, dts.sec, dts.milliSec);
    else
        printf("%02u-%02u-%04u %02u:%02u:%02u.%03u, ", dts.day, dts.month, dts.year, dts.hour, dts.min, dts.sec, dts.milliSec);
}

static void printDataValue(UA_DataValue *value)
{
    if (value->hasServerTimestamp)
        printTimestamp("ServerTime", value->serverTimestamp);

    if (value->hasSourceTimestamp)
        printTimestamp("SourceTime", value->sourceTimestamp);

    if (value->hasStatus)
        printf("Status 0x%08x, ", value->status);

    if (value->value.type == &UA_TYPES[UA_TYPES_UINT32])
    {
        UA_UInt32 hrValue = *(UA_UInt32*) value->value.data;
        printf("Uint32Value %u\n", hrValue);
    }

    if (value->value.type == &UA_TYPES[UA_TYPES_DOUBLE])
    {
        UA_Double hrValue = *(UA_Double*) value->value.data;
        printf("DoubleValue %f\n", hrValue);
    }
}

static UA_Boolean readRaw(const UA_HistoryData *data)
{
    printf("readRaw Value count: %lu\n", (long unsigned) data->dataValuesSize);

    for (UA_UInt32 i = 0; i < data->dataValuesSize; ++i)
    {
        printDataValue(&data->dataValues[i]);
    }

    return true;
}

static UA_Boolean readHist(UA_Client *client, const UA_NodeId *nodeId, UA_Boolean moreDataAvailable, const UA_ExtensionObject *data, void *unused)
{
    printf("\nRead historical callback:\n");
    printf("\tHas more data:\t%d\n\n", moreDataAvailable);
    if (data->content.decoded.type == &UA_TYPES[UA_TYPES_HISTORYDATA])
    {
        return readRaw((UA_HistoryData*) data->content.decoded.data);
    }

    return true;
}

int main(int argc, char *argv[])
{
    UA_Client *client = UA_Client_new();
    UA_ClientConfig_setDefault(UA_Client_getConfig(client));

    UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:62544");
    if (retval != UA_STATUSCODE_GOOD)
    {
        UA_Client_delete(client);
        return EXIT_FAILURE;
    }

    printf("\nStart historical read (1, \"myUintValue\"):\n");
    UA_NodeId node = UA_NODEID_STRING(2, "MyLevel");
    retval = UA_Client_HistoryRead_raw(client, &node, readHist, UA_DateTime_fromUnixTime(0), UA_DateTime_now(), UA_STRING_NULL, false, 10, UA_TIMESTAMPSTORETURN_BOTH, (void*) UA_FALSE);

    if (retval != UA_STATUSCODE_GOOD)
    {
        printf("Failed. %s\n", UA_StatusCode_name(retval));
    }

    UA_Client_disconnect(client);
    UA_Client_delete(client);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}

```

```
01:15:31 **** Incremental Build of configuration Debug for project open62541 ****
Info: Internal Builder is used for build
gcc -std=c99 -DUA_ARCHITECTURE_WIN32 -O0 -g3 -Wall -c -fmessage-length=0 -o main.o "..\\main.c" 
..\main.c:40:33: error: unknown type name 'UA_HistoryData'
   40 | static UA_Boolean readRaw(const UA_HistoryData *data)
      |                                 ^~~~~~~~~~~~~~
..\main.c: In function 'readRaw':
..\main.c:42:62: error: request for member 'dataValuesSize' in something not a structure or union
   42 |     printf("readRaw Value count: %lu\n", (long unsigned) data->dataValuesSize);
      |                                                              ^~
..\main.c:44:35: error: request for member 'dataValuesSize' in something not a structure or union
   44 |     for (UA_UInt32 i = 0; i < data->dataValuesSize; ++i)
      |                                   ^~
..\main.c:46:29: error: request for member 'dataValues' in something not a structure or union
   46 |         printDataValue(&data->dataValues[i]);
      |                             ^~
..\main.c: In function 'readHist':
..\main.c:56:49: error: 'UA_TYPES_HISTORYDATA' undeclared (first use in this function); did you mean 'UA_NS0ID_HISTORYDATA'?
   56 |     if (data->content.decoded.type == &UA_TYPES[UA_TYPES_HISTORYDATA])
      |                                                 ^~~~~~~~~~~~~~~~~~~~
      |                                                 UA_NS0ID_HISTORYDATA
..\main.c:56:49: note: each undeclared identifier is reported only once for each function it appears in
..\main.c:58:25: error: 'UA_HistoryData' undeclared (first use in this function); did you mean 'UA_SignatureData'?
   58 |         return readRaw((UA_HistoryData*) data->content.decoded.data);
      |                         ^~~~~~~~~~~~~~
      |                         UA_SignatureData
..\main.c:58:40: error: expected expression before ')' token
   58 |         return readRaw((UA_HistoryData*) data->content.decoded.data);
      |                                        ^
..\main.c: In function 'main':
..\main.c:78:14: warning: implicit declaration of function 'UA_Client_HistoryRead_raw' [-Wimplicit-function-declaration]
   78 |     retval = UA_Client_HistoryRead_raw(client, &node, readHist, UA_DateTime_fromUnixTime(0), UA_DateTime_now(), UA_STRING_NULL, false, 10, UA_TIMESTAMPSTORETURN_BOTH, (void*) UA_FALSE);
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~

01:15:34 Build Failed. 7 errors, 1 warnings. (took 2s.534ms)
```

AlexanderZhirov, 2021-12-07:
I've already made the code as simple as possible, and errors keep pouring in. Does anyone have any suggestions in which direction to dig? I don't understand what to do.

```
#include <open62541/client_config_default.h>
#include <open62541/client_highlevel.h>

#include <stdio.h>
#include <stdlib.h>

//static void printTimestamp(char *name, UA_DateTime date)
//{
//    UA_DateTimeStruct dts = UA_DateTime_toStruct(date);
//    if (name)
//        printf("%s: %02u-%02u-%04u %02u:%02u:%02u.%03u, ", name, dts.day, dts.month, dts.year, dts.hour, dts.min, dts.sec, dts.milliSec);
//    else
//        printf("%02u-%02u-%04u %02u:%02u:%02u.%03u, ", dts.day, dts.month, dts.year, dts.hour, dts.min, dts.sec, dts.milliSec);
//}
//
//static void printDataValue(UA_DataValue *value)
//{
//    if (value->hasServerTimestamp)
//        printTimestamp("ServerTime", value->serverTimestamp);
//
//    if (value->hasSourceTimestamp)
//        printTimestamp("SourceTime", value->sourceTimestamp);
//
//    if (value->hasStatus)
//        printf("Status 0x%08x, ", value->status);
//
//    if (value->value.type == &UA_TYPES[UA_TYPES_UINT32])
//    {
//        UA_UInt32 hrValue = *(UA_UInt32*) value->value.data;
//        printf("Uint32Value %u\n", hrValue);
//    }
//
//    if (value->value.type == &UA_TYPES[UA_TYPES_DOUBLE])
//    {
//        UA_Double hrValue = *(UA_Double*) value->value.data;
//        printf("DoubleValue %f\n", hrValue);
//    }
//}
//
//static UA_Boolean readRaw(const UA_HistoryData *data)
//{
//    printf("readRaw Value count: %lu\n", (long unsigned) data->dataValuesSize);
//
//    for (UA_UInt32 i = 0; i < data->dataValuesSize; ++i)
//    {
//        printDataValue(&data->dataValues[i]);
//    }
//
//    return true;
//}

static UA_Boolean readHist(UA_Client *client, const UA_NodeId *nodeId, UA_Boolean moreDataAvailable, const UA_ExtensionObject *data, void *unused)
{
    printf("\nRead historical callback:\n");
    printf("\tHas more data:\t%d\n\n", moreDataAvailable);
//    if (data->content.decoded.type == &UA_TYPES[UA_TYPES_HISTORYDATA])
//    {
//        return readRaw((UA_HistoryData*) data->content.decoded.data);
//    }

    return true;
}

int main(int argc, char *argv[])
{
    UA_Client *client = UA_Client_new();
    UA_ClientConfig_setDefault(UA_Client_getConfig(client));
[... 52 lines cut ...]
  397 | static UA_INLINE UA_HistoryUpdateResponse
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~
C:/msys64/mingw64/include/open62541/client.h:398:58: error: unknown type name 'UA_HistoryUpdateRequest'
  398 | UA_Client_Service_historyUpdate(UA_Client *client, const UA_HistoryUpdateRequest request) {
      |                                                          ^~~~~~~~~~~~~~~~~~~~~~~
C:/msys64/mingw64/include/open62541/client.h: In function 'UA_Client_Service_historyUpdate':
C:/msys64/mingw64/include/open62541/client.h:399:5: error: unknown type name 'UA_HistoryUpdateResponse'
  399 |     UA_HistoryUpdateResponse response;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
C:/msys64/mingw64/include/open62541/client.h:400:53: error: 'UA_TYPES_HISTORYUPDATEREQUEST' undeclared (first use in this function); did you mean 'UA_NS0ID_HISTORYUPDATEREQUEST'?
  400 |     __UA_Client_Service(client, &request, &UA_TYPES[UA_TYPES_HISTORYUPDATEREQUEST],
      |                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                     UA_NS0ID_HISTORYUPDATEREQUEST
C:/msys64/mingw64/include/open62541/client.h:401:30: error: 'UA_TYPES_HISTORYUPDATERESPONSE' undeclared (first use in this function); did you mean 'UA_NS0ID_HISTORYUPDATERESPONSE'?
  401 |         &response, &UA_TYPES[UA_TYPES_HISTORYUPDATERESPONSE]);
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                              UA_NS0ID_HISTORYUPDATERESPONSE

18:19:07 Build Failed. 10 errors, 0 warnings. (took 2s.515ms)
```

After enabling UA_ENABLE_HISTORIZING, the preprocessor now gets errors elsewhere.
[image]

Is the library even working?

Trunkshot, 2021-12-07:
+up
Faced the same problem as the author in the process of reading historical data. Is there an instruction or source code that could compile?

AlexanderZhirov, 2021-12-08:
So, I rebuilt the branch library 1.2. The build was successful:
```
User@WIN-RRMNF8HHDN5 MINGW64 ~/open62541
# git submodule update --init --recursive
Submodule path 'deps/ua-nodeset': checked out '393b633468a5d1d062dd253e1488d1d8ba335b6f'

User@WIN-RRMNF8HHDN5 MINGW64 ~/open62541
# mkdir build && cd build

User@WIN-RRMNF8HHDN5 MINGW64 ~/open62541/build
# cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL ..
-- Building for: Ninja
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/mingw64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/msys64/mingw64/bin/c++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: C:/msys64/mingw64/bin/python.exe (found version "3.9.7")
-- Found Git: C:/msys64/usr/bin/git.exe (found version "2.34.1")
-- open62541 Version: 1.2.3
-- The selected architecture is: win32
-- Test CC flag -std=c99
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -pipe
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wall
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wextra
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wpedantic
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Werror
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wno-static-in-inline
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wno-overlength-strings
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wno-unused-parameter
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wc++-compat
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wformat
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wformat-security
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wformat-nonliteral
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wmissing-prototypes
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
[... 148 lines cut ...]
{
    UA_Client *client = UA_Client_new();
    UA_ClientConfig_setDefault(UA_Client_getConfig(client));

    UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:62544");
    if (retval != UA_STATUSCODE_GOOD)
    {
        UA_Client_delete(client);
        return EXIT_FAILURE;
    }

    UA_NodeId node = UA_NODEID_STRING(1, "TNPZ.P_NEFT.R1.M");

    retval = UA_Client_HistoryRead_raw(client, &node, readHist, UA_DateTime_fromUnixTime(0), UA_DateTime_now(), UA_STRING_NULL, false, 10, UA_TIMESTAMPSTORETURN_BOTH, (void*) UA_FALSE);

    if (retval != UA_STATUSCODE_GOOD)
    {
        printf("Failed. %s\n", UA_StatusCode_name(retval));
    }

    UA_Client_disconnect(client);
    UA_Client_delete(client);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

jpfr, 2021-12-08:
You have to enable historical access in the build settings when the library is generated.
You can see that the functions are wrapped into an #ifdef.

AlexanderZhirov, 2021-12-08:
> You have to enable historical access in the build settings when the library is generated.

Yes, I added a compilation parameter, but with it another error is obtained

```
gcc -std=c99 -DUA_ARCHITECTURE_WIN32 -DUA_ENABLE_HISTORIZING -O0 -g3 -Wall -c -fmessage-length=0 -o main.o "..\\main.c" 
```

```
03:18:12 **** Rebuild of configuration Debug for project open62541 ****
Info: Internal Builder is used for build
gcc -std=c99 -DUA_ARCHITECTURE_WIN32 -DUA_ENABLE_HISTORIZING -O0 -g3 -Wall -c -fmessage-length=0 -o main.o "..\\main.c" 
gcc -o open62541.exe main.o -lopen62541 -lws2_32 
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: main.o: in function `main':
C:\msys64\home\User\open62541\Debug/../main.c:78: undefined reference to `__imp_UA_Client_HistoryRead_raw'
collect2.exe: error: ld returned 1 exit status

03:18:14 Build Failed. 2 errors, 0 warnings. (took 2s.452ms)
```

Trunkshot, 2021-12-08:
It is strange that the documentation does not specify the constant **UA_ENABLE_HISTORIZING**, which implies automatic inclusion when building this library. Because of this, there was ambiguity. Is it possible to update the documentation? Since implicit moments are misleading and generate issues.

After the compilation parameter was added during the build, the function for reading historical data works correctly

```
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_ENABLE_HISTORIZING=ON -DUA_NAMESPACE_ZERO=FULL ..
```

AlexanderZhirov, 2021-12-08:
> You have to enable historical access in the build settings when the library is generated.

> ```
> cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_ENABLE_HISTORIZING=ON -DUA_NAMESPACE_ZERO=FULL ..
> ```

This solution worked for me! Thank you very much!

> And I need to read the data on the specified tag - **TNPZ.P_NEFT.R1.M**.

[image]

closed 2021-12-08 by AlexanderZhirov