open62541/open62541 #4827: Wrong handling of SimpleAttributeOperand

basyskom-jvoe, 2021-12-08:
## Description
According to Part 9, 5.2.2, the Node ID of a condition instance can be retrieved by using the SimpleAttributeOperand from [Table 10](https://reference.opcfoundation.org/v104/Core/docs/Part9/5.5.2/#Table10) (leave the browse path empty and ask for the NodeId attribute).

The open62541 server returns the node id of the condition type instead of the node id of the condition instance.
This happens because the ReadValueId's node id is explicitly set to the type definition id from the SimpleAttributeOperand if the browse path is empty: https://github.com/open62541/open62541/blob/bb1abfc8810fa8226f3d1319911689c6b760c53a/src/server/ua_subscription_events.c#L172

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.2.3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-12-09:
Yes, we can fix on 1.2.
You create a PR?

basyskom-jvoe, 2021-12-10:
@jpfr I have created a PR

jpfr, 2021-12-13:
Thanks for the fix.

closed 2021-12-13 by jpfr