open62541/open62541 #4828: cmake macro for nodeset compiler results in "unrecognized arguments: --namespaceMap=123:http://..." from python script

jackik1410, 2021-12-08:
## Description
Trying to execute the cmake macros for `ua_generate_nodeset` or `ua_generate_nodeset` results in python errors. Running the python scripts directly with the provided arguments seems to work fine however.
The original version for the CMakeLists.txt is from the [opcua-modeling-tutorial-server repository](https://github.com/Pro/opcua-modeling-tutorial-server)

The only edit to the CMakeLists.txt that was done, was changing the to be used .csv, .bsd and .xml file, as well as replacing the `NAMESPACE_IDX` with the `NAMESPACE_MAP` directive in the arguments of the macro. There doesn't seem to be much documentation or previous issues with this. Very possible that this approach is flawed or naive.

## Background Information / Reproduction Steps

Folder Structure:
- root
  - open62541/ (cloned repo)
  - CMakeLists.txt

Either install open62541 according to the guide, or clone this repo into a local folder. (Tried both, made sure to `git submodule update --init` beforehand)
- Open a new file and call it CMakeLists.txt with these contents:
 ```CMake
cmake_minimum_required(VERSION 3.2)
project(opcua-modeling-tutorial-server)

# ~~open62541 must be installed.~~
# If in custom path, then use -DCMAKE_PREFIX_PATH=/home/user/install
# Use either an installed version of open62541 or a local clone of the repo for the scripts
# find_package(open62541 1.2 REQUIRED COMPONENTS FullNamespace)

SET(MY_OPEN62541 ${PROJECT_SOURCE_DIR}/open62541)
include("${MY_OPEN62541}/tools/cmake/macros_public.cmake")

# Output directory for Nodeset Compiler
set(GENERATE_OUTPUT_DIR "${CMAKE_BINARY_DIR}/src_generated/")

file(MAKE_DIRECTORY "${GENERATE_OUTPUT_DIR}")
include_directories("${GENERATE_OUTPUT_DIR}")

ua_generate_nodeset_and_datatypes(
        NAME "foo_flt"
        TARGET_PREFIX "${PROJECT_NAME}"
        # NAMESPACE_MAP "2:http://new.foo.com/zebra-compression/flattening-and-subspacefolding/UA/"
        # FILE_CSV "${modeldir}/FooFltModel.csv"
        # FILE_BSD "${modeldir}/FooFlt.Types.bsd"
        NAMESPACE_MAP "2:http://opcfoundation.org/UA/DI/"
        FILE_CSV "${PROJECT_SOURCE_DIR}/open62541/deps/ua-nodeset/DI/OpcUaDiModel.csv"
        FILE_BSD "${PROJECT_SOURCE_DIR}/open62541/deps/ua-nodeset/DI/Opc.Ua.Di.Types.bsd"
        OUTPUT_DIR "${GENERATE_OUTPUT_DIR}"
    # This namespace index must match the order in which you are adding the nodeset in the source code
        # NAMESPACE_IDX 2
        FILE_NS "${PROJECT_SOURCE_DIR}/open62541/deps/ua-nodeset/DI/Opc.Ua.DI.NodeSet2.xml"
        INTERNAL
)

# Previous macro automatically sets some variables which hold the generated source code files using the provided NAME
add_executable(opcua-modeling-tutorial-server
               ${UA_NODESET_FOO_FLT_SOURCES}
               ${UA_TYPES_FOO_FLT_SOURCES}
               main.c
               )

# Make sure the nodeset compiler is execute before compiling the main file
add_dependencies(opcua-modeling-tutorial-server
                 ${PROJECT_NAME}-ns-foo_flt
                 )

target_link_libraries(opcua-modeling-tutorial-server
  # open62541::open62541 # commented out in case of direct inclusion of the macros_public.cmake
  )

 ```
- create a build/ dir and enter it
- run `cmake ..` and `make` 
- Error log:
```log
$ make
[ 11%] Built target opcua-modeling-tutorial-server-ids-foo_flt
[ 22%] Generating src_generated/types_foo_flt_generated.c, src_generated/types_foo_flt_generated.h, src_generated/types_foo_flt_generated_handling.h
usage: generate_datatypes.py [-h] [-c <typeDescriptions>]
                             [--namespace NAMESPACE] [-s <selectedTypes>]
                             [--no-builtin] [--opaque-map <opaqueTypeMap>]
                             [--internal] [-t <typeBsds>] [-i <importBsds>]
                             <outputFile>
generate_datatypes.py: error: unrecognized arguments: --namespaceMap=2:http://opcfoundation.org/UA/DI/
make[2]: *** [CMakeFiles/opcua-modeling-tutorial-server-types-foo_flt.dir/build.make:65: src_generated/types_foo_flt_generated.c] Error 2
make[1]: *** [CMakeFiles/Makefile2:161: CMakeFiles/opcua-modeling-tutorial-server-types-foo_flt.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2 and 28f85f05
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu focal 20.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jackik1410, 2021-12-08:
Cmake seems to pretty aggressively swallow quotation marks from my tests with the above example, but I am not at all sure where the actual problem stems from.

jpfr, 2021-12-09:
The usage string says `[--namespace NAMESPACE]`.
Try replacing the `=` with a whitespace.

jackik1410, 2021-12-10:
Thank you for your response.

I'm using the provided CMAKE macro, I'm not using any `=`-sign myself. `NAMESPACE_MAP "2:http://opcfoundation.org/UA/DI/"` is how I pass that on through CMAKE.

Using the direct python method works, but is more prone to issues as it comes to blacklists and subsequent faults on a mistake.
Is this an issue with the cmake function arguments or how cmake passes them to the python scripts?

closed 2026-10-20 by o6-nbeier