open62541/open62541 #4837: API Question Client: How to receive the used SecurityPolicy

JOO200, 2021-12-09:
## Description

I implemented the open62541 OPC UA Client in an application and want to increase the information output.

The signed and encrypted connection works fine and the implemented logger will report the successful connection with informations about the used endpoint, SecurityMode (None, Sign, Sign and Encrypt) and the SecurityPolicy (URI of the SecurityPolicy).

I want to add user feedback to my application and use those informations in the stateCallback from the UA_Client.

However it's not possible to get those informations through the API as the structs are marked as internal, there are no getter methods and the used access for the log messages doesn't work:

```cpp
conf->stateCallback = +[](UA_Client * client, UA_SecureChannelState channelState, UA_SessionState sessionState, UA_StatusCode connectStatus) {
    if (client->channel && client->channel.securityPolicy->policyUri) {
        std::string policy;
        policy.assign(client->channe.securityPolicy->policyUri.length, client->channel.securityPolicy->policyUri.data);
        // add feedback such as "Connected encrypted with Policy 'Aes128Sha256'"
    };
}
```
This will fail with the error "invalid use of incomplete type 'UA_Client' { aka 'struct UA_Client' }

Maybe you can add some few getter methods to receive informations about the current connection and encryption state.

## Background Information / Reproduction Steps

I can add this and make a PR for that. Maybe I need some additional help and some reviews for that but if you don't have the time for that, feel free to request the PR from me.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.2.3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Debian 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-12-12:
Sure!
Go ahead and propose an API for that.

please let’s finalize the api first before you put effort into the implementation...

JOO200, 2021-12-13:
I guess this stuff might be interesting:

| Name | Datatype | Description  |
|---|---|---|
| ConnectionTime | UA_DateTime | The DateTime when the connection was accepted |
| RenewTime | UA_DateTime  | The DateTime when the current channel has to be renewed | was last renewed |
| RenewState | UA_SecureChannelRenewState | The renew state |
| SecurityMode | UA_MessageSecurityMode | The currently used SecurityMode |
| SecurityPolicy | UA_SecurityPolicy * | The currently active SecurityPolicy |
| RemoteCertificate | UA_ByteString | The currently active remote certificate (public key) from the server |

Either we add a getter for everything in the client.h or we create a struct, fill the struct on the api request and return it. The latter one has a bigger overhead.

I'm not too familiar with APIs in C, any help is welcome.

JOO200, 2021-12-29:
May I push this? I still need some feedback for the API design

JOO200, 2022-02-24:
Another push.

Currently I'm missing the feedback for rejected Client Certificates. The client connections work fine but when a server rejects the certificate there will be no callback for that.

The channel will be closed and no further information is outside the logs available:
The server answers with an error (BadSecurityChecksFailed):
Client Stacktrace:
UA_Client_run_iterate -> receiveResponse -> UA_SecureChannel_receive -> (over the hardcoded callback) processServiceResponse -> processERRResponse.

I don't see a way to change the used hardcoded callback (processServiceResponse) and I don't think that should be exposed through the api. I would like to see another callback there so I would be able to generate proper user feedback for that issue.