open62541/open62541 #4842: cmake build on windows 2019 VS2019

JamMika, 2021-12-12:
d:\dev\C++\OpcUa\open62541\build>"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL ..
-- Building for: Visual Studio 16 2019
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19042.
-- The C compiler identification is MSVC 19.29.30137.0
-- The CXX compiler identification is MSVC 19.29.30137.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: C:/Python37-32/python.exe (found version "3.7.9")
-- Found Git: C:/p64/dev/Git/cmd/git.exe (found version "2.34.1.windows.1")
-- open62541 Version: 1.2.3
-- The selected architecture is: win32
-- Could NOT find Sphinx (missing: SPHINX_EXECUTABLE)
-- Could NOT find LATEX (missing: LATEX_COMPILER)
RegularExpression::compile(): Nested *?+.
RegularExpression::compile(): Error in compile.
CMake Error at CMakeLists.txt:1534 (if):
  if given arguments:

    "D:/dev/C++/OpcUa/open62541/build/src_generated/open62541/config.h" "MATCHES" "D:/dev/C++/OpcUa/open62541/deps"

  Regular expression "D:/dev/C++/OpcUa/open62541/deps" cannot compile

-- Configuring incomplete, errors occurred!
See also "D:/dev/C++/OpcUa/open62541/build/CMakeFiles/CMakeOutput.log".
[CMakeOutput.log](https://github.com/open62541/open62541/files/7699925/CMakeOutput.log)

jpfr, 2021-12-14:
I think the problem is the `++` in the path which was evaluated as regex-sensitive.

The fix is in #4845. Can you verify before we merge?

o6-nbeier, 2026-10-20:
Closing as fixed (see #4845 and 6fd48c2).

closed 2026-10-20 by o6-nbeier