open62541/open62541 #4846: Is there a way to compile the .a and .h files of the arm architecture with arm-none-eabi-gcc?

ChaseWwwwind, 2021-12-14:
## Description

I am new to open62541, personal compilation environment is based on x86 ubuntu system. I want to compile open62541.c and open62541.h files for the arm architecture and embed open62541 into lightweight devices.

Is there anyone who can guide me to achieve this goal?

## Background Information / Reproduction Steps

Used CMake options:

cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DCMAKE_C_COMPILER=arm-none-eabi-gcc -DCMAKE_CXX_COMPILER=arm-none-eabi-g++

mlang-de, 2021-12-14:
I will not guide you, but maybe this helps https://askubuntu.com/questions/250696/how-to-cross-compile-for-arm.

mlang-de, 2021-12-14:
May check additionally https://github.com/Pro/open62541-arduino and https://github.com/Pro/open62541-esp32

ChaseWwwwind, 2021-12-15:
hi , thank you in advance for your answer. Following your help, I tried to compile with arm-none-eabi-gcc, but ran into the following problems:

x00512160@wuhmhispre00188:~/workspace/open62541/open62541/build2$ cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=REDUCED -DCMAKE_C_COMPILER=arm-none-eabi-gcc -DCMAKE_CXX_COMPILER=arm-none-eabi-g++
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- Check for working C compiler: /home/x00512160/buildTools/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc
-- Check for working C compiler: /home/x00512160/buildTools/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc -- broken
CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/home/x00512160/buildTools/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/x00512160/workspace/open62541/open62541/build2/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/make cmTC_f3277/fast && /usr/bin/make -f CMakeFiles/cmTC_f3277.dir/build.make CMakeFiles/cmTC_f3277.dir/build
    make[1]: Entering directory '/home/x00512160/workspace/open62541/open62541/build2/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_f3277.dir/testCCompiler.c.o
    /home/x00512160/buildTools/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc    -o CMakeFiles/cmTC_f3277.dir/testCCompiler.c.o   -c /home/x00512160/workspace/open62541/open62541/build2/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_f3277
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_f3277.dir/link.txt --verbose=1
    /home/x00512160/buildTools/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc      CMakeFiles/cmTC_f3277.dir/testCCompiler.c.o  -o cmTC_f3277
    /home/x00512160/buildTools/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /home/x00512160/buildTools/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): in function `exit':
    exit.c:(.text.exit+0x2c): undefined reference to `_exit'
    collect2: error: ld returned 1 exit status
    make[1]: *** [CMakeFiles/cmTC_f3277.dir/build.make:87: cmTC_f3277] Error 1
    make[1]: Leaving directory '/home/x00512160/workspace/open62541/open62541/build2/CMakeFiles/CMakeTmp'
    make: *** [Makefile:121: cmTC_f3277/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

-- Configuring incomplete, errors occurred!

mlang-de, 2021-12-15:
https://stackoverflow.com/questions/19419782/exit-c-text0x18-undefined-reference-to-exit-when-using-arm-none-eabi-gcc

closed 2026-10-20 by o6-nbeier