open62541/open62541 #4848: How to set for each server access login (password and user_name) one session ?

mohamedalielghoula, 2021-12-14:
Hallo all
**this my logins:**
static UA_UsernamePasswordLogin logins[5] = {
    {UA_STRING_STATIC("user1"), UA_STRING_STATIC("1234")},
    {UA_STRING_STATIC("user2"), UA_STRING_STATIC("abcd")},
    {UA_STRING_STATIC("user3"), UA_STRING_STATIC("0000")},
    {UA_STRING_STATIC("user4"), UA_STRING_STATIC("1111")}
};
**this function is to set logins:**
UA_AccessControl_default(uaServerConfig, false,
           &uaServerConfig->securityPolicies[uaServerConfig->securityPoliciesSize-1].policyUri, 5, logins);
           
**and hier to configure the max of sessions** 

uaServerConfig->maxSessions = 4;
This way I can access over 4 sessions regardless who is the login.
my question that if it is possible to set one session for each login ?
@Pro @jpfr

mlang-de, 2021-12-14:
Yes, but you have to implement it by your own in the access control plugin "open62541/plugins/ua_accesscontrol_default.c".
Beware, your array logins has the size 5, whereby you only define 4. So username "" and password "" could also be used imho.

o6-nbeier, 2026-10-20:
Thank you for the question.

`maxSessions` is server-global, so a per-login session limit has to be implemented in your own AccessControl plugin by tracking active sessions per username in the activate/close-session callbacks.

Closing for now.

closed 2026-10-20 by o6-nbeier