open62541/open62541 #4851: How do I make the node ID point to a string type (ns=1;s=pathToTag)?

vvpnet, 2021-12-15:
Hi!!
I am developing an opc ua server. How do I make clients (Uaexpert Prosys) in the nodeid field display "NODEID" the path to object. For example: ns=2;s=root/objects/test/child ?

mlang-de, 2021-12-15:
I'm not sure whether I understand your question right.
I string nodeId could be create like this
UA_NODEID_STRING(2, "root/objects/test/child")
Or you search a solution to generate this string automatically?

vvpnet, 2021-12-15:
I was just thinking that this is done automatically.
I think it's all about the server settings?

There are 4 types of NodeId:

Numeric (value is a UInteger)
String (value is a String)
Guid (value is a Guid/UUID)
Opaque (value is ByteString)
[image]

mlang-de, 2021-12-15:
"I was just thinking that this is done automatically.
I think it's all about the server settings?"
Imho not, which configuration should it be?

Is the NodeId 205943 from your example not given by you?
Only numerical NodeId will be generated automatically see function zipNsInsertNode() and/or UA_NodeMap_insertNode().

vvpnet, 2021-12-16:
Many thanks. Now everything is clear to me.