open62541/open62541 #4863: Web sockets tutorials : any examples ?

jackybek, 2021-12-20:
## Description
Can I know if there are examples on how to implement opc UA server with web sockets support ?

i only found 1 line of code  in the tutorial on encryption.

## Background Information / Reproduction Steps

After I added the following line:

UA_ServerConfig_addNetworkLayerWS(UA_Server_getConfig(server), 7681, 1048576, 1048576, &certificate, &privateKey);

when I run the server, ua expert fail to connect but the web socket client program can connect.

Is it possible to enable all modes at the same time
1. client server
2. Pub sub mqtt
3. Web sockets

What’s next ?

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jackybek, 2021-12-20:
Open62541 version : 1.2.2
OS: raspbian

i read from the release notes of 1.2.2 that support for websocket is available in this version.

Besides adding the 1 line of code to instantiate the websocket layer, is there any example how to allow a websocket client to read the opcua data in the nodes ?

jackybek, 2021-12-21:
I traced the source and realised that UA_parseEndpointUrl() did not cater to ws:// type of endpoint.
Hence although the websocket layer is initialised successfully using UA_ServerConfig_addNetworkLayerWS(), the subsequent calls to UA_parseEndpointUrl() failed.

I read that support for websockets is now available from v1.2.2 .. what does it mean then ?

jackybek, 2021-12-21:
I tried the following changes and now the websocket client from microsoft is able to connect successfully:

     UA_StatusCode UA_parseEndpointUrl(const UA_String *endpointUrl, UA_String *outHostname, u16 *outPort, UA_String *outPath) {
    UA_Boolean ipv6 = false;

    /* Url must begin with "opc.tcp://" or opc.udp:// (if pubsub enabled) */
    if(endpointUrl->length < 11) {
        return UA_STATUSCODE_BADTCPENDPOINTURLINVALID;
    }
    if (strncmp((char*)endpointUrl->data, "opc.tcp://", 10) != 0) {
             #ifdef UA_ENABLE_WEBSOCKETS
                 if (strncmp((char*)endpointUrl->data, "ws://", 5) != 0 &&
                strncmp((char*)endpointUrl->data, "wss://", 6) != 0) {
             #endif
             #ifdef UA_ENABLE_PUBSUB
        if (strncmp((char*)endpointUrl->data, "opc.udp://", 10) != 0 &&
                // using websocket client from microsoft to test : only support ws:// or wss://
                // refer to https://bami-tech.com/privacy_policy.html
                strncmp((char*)endpointUrl->data, "ws://", 5) != 0 &&
                strncmp((char*)endpointUrl->data, "wss://", 6) != 0 &&
                strncmp((char*)endpointUrl->data, "opc.mqtt://", 11) != 0) {
            return UA_STATUSCODE_BADTCPENDPOINTURLINVALID;
        }
             #else
        return UA_STATUSCODE_BADTCPENDPOINTURLINVALID;
             #endif
        #ifdef UA_ENABLE_WEBSOCKETS
         }
        #endif
         .......

To test add the following function call:

        #ifdef UA_ENABLE_WEBSOCKET_SERVER
                        // somehow after addNetworkLayerWS(...), OPCClient and UAExpert cannot connect to OPCServer
                        // error message in client is : AG_mainOPCUAClient.c : Cannot login securely to OPCUAServer : opc.tcp://192.168.1.33:4840

                        // example on enable websockets can be found here:
                        // https://github.com/open62541/open62541/blob/master/examples/tutorial_server_variable.c
                        // https://github.com/open62541/open62541/blob/master/examples/encryption/server_encryption.c
                        UA_UInt32 sendBufferSize = 1024 * 1024; // 1 Mbyte
                        UA_UInt32 recvBufferSize = 1024 * 1024; // 1 Mbyte
                        UA_StatusCode retVal;
                        retVal = UA_ServerConfig_addNetworkLayerWS(UA_Server_getConfig(uaServer1), 7681, sendBufferSize, recvBufferSize, &certificate, &privateKey);
                        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "retVal is %s", UA_StatusCode_name(retVal) );
        #endif

jackybek, 2021-12-27:
After I change my server code for OpenSSL encryption using self-signed certificate, I can now use UAExpert to connect to the opcua instance with websockets enabled.

jackybek, 2022-01-03:
I made the following changes to UA_parseEndpointUrl() to handle new connection string for:
1. mqtt connections : opc.mqtt://
2. ws connections : ws:// and wss://

It is tested on v1.3-rc1-128-g1bd54abc

source file : **"/src/ua_util.c"**

UA_StatusCode
UA_parseEndpointUrl(const UA_String *endpointUrl, UA_String *outHostname,
                    u16 *outPort, UA_String *outPath) {
    UA_Boolean ipv6 = false;
    UA_Boolean is_udp = false;
    UA_Boolean is_tcp = false;
    UA_Boolean is_mqtt = false;
    UA_Boolean is_ws = false;
    UA_Boolean is_wss = false;

printf("UA_parseEndpointUrl() : endpointUrl = %s, length = %d \n", endpointUrl->data, endpointUrl->length);

    /* Url must begin with "opc.tcp://" or opc.udp:// (if pubsub enabled) */
    //if(endpointUrl->length < 11) {
    //    return UA_STATUSCODE_BADTCPENDPOINTURLINVALID;
    //}

    if (strncmp((char*)endpointUrl->data, "opc.tcp://", 10) != 0) {
#ifdef UA_ENABLE_PUBSUB
      if (strncmp((char*)endpointUrl->data, "opc.udp://", 10) != 0) {
	 if (strncmp((char*)endpointUrl->data, "opc.mqtt://", 11) != 0) {
	    #ifdef UA_ENABLE_WEBSOCKET_SERVER
	        if (strncmp((char*)endpointUrl->data, "ws://", 5) != 0) {
		  if (strncmp((char*)endpointUrl->data, "wss://", 6) != 0 ) {
				return UA_STATUSCODE_BADTCPENDPOINTURLINVALID;
		  }
		  else {
			 is_wss = true;
			 printf("url prefix is a <wss://> \n");
		  }
		}
		else {
		       is_ws = true;
		       printf("url prefix is a <ws://> \n");
		}
	    #endif
            	//return UA_STATUSCODE_BADTCPENDPOINTURLINVALID;
	 }
	 else {
		is_mqtt=true;
		printf("url prefix is a <opc.mqtt://> \n");
	 }
      }
      else {
	     is_udp = true;
	     printf("url prefix is a <opc.udp://> or <opc.mqtt://> \n");
      }
#else
        return UA_STATUSCODE_BADTCPENDPOINTURLINVALID;
#endif
    }
    else {
	   is_tcp = true;
	   printf("strncmp endpointUrl->data with <opc.tcp://> is a match \n");
    }

// added by Jacky on 2 Jan 2022 to handle websockets ws:// or wss://
// to reuse the code below, change
//   1. wss:// to opc.wss://
//   2. ws:// to opc.ws1://

/*
[... 157 lines cut ...]

    /*
    if(endpointUrl->data[curr] != '/')
        return UA_STATUSCODE_BADTCPENDPOINTURLINVALID;
    if(++curr == endpointUrl->length) {
        printf("just about to exit UA_parseEndpointUrl() \n");
        return UA_STATUSCODE_GOOD;
    }

    outPath->data = &endpointUrl->data[curr];	// curr
    outPath->length = endpointUrl->length - curr;

    // Remove trailing slash from the path
    if(endpointUrl->data[endpointUrl->length - 1] == '/')
        outPath->length--;

    // Empty string?
    if(outPath->length == 0)
        outPath->data = NULL;

    return UA_STATUSCODE_GOOD;
    */

  }
}

jackybek, 2022-02-09:
Previously in my mqtt code i am able to make my OPCUA server periodically subscribe to data from a MQTT broker using the following functions:

     step 1:  create a function:
                  static void addSubscription(UA_Server *server, UA_PubSubConnection *connection)
                  UA_StatusCode rv = connection->channel->regist(connection->channel, &transportSettings, &callback)
                  if (rv == UA_STATUSCODE_GOOD)
                   {
                           UA_UInt64 subscriptionCallbackId;
                          UA_Server_addRepeatedCallback(server, (UA_ServerCallback)mqttYieldPollingCallback,
                                          connection, PUBLISH_INTERVAL_MQTT / 2, &subscriptionCallbackId);      // 9000 millisecond which is the same as sampli$
                   }
 
      step 2: create a function
                     /* Periodically refreshes the MQTT stack (sending/receiving) */
                       //https://github.com/open62541/open62541/blob/master/examples/pubsub/server_pubsub_subscriber_rt_level.c
                      static void
                      mqttYieldPollingCallback(UA_Server *uaServer, UA_PubSubConnection *connection)
                     {
                          //    connection->channel->yield(connection->channel, (UA_UInt16) 30);  // timeout : 30 seconds
                           addSubscribedVariables(uaServer, readerIdentifier); // repeated update the UA AddressSpace

                     }

Now, for websockets, i like to periodically write a web json payload to port 7681.  I tried the following:

      step 1: retVal = UA_ServerConfig_addNetworkLayerWS(&config1, 7681, sendBufferSize, recvBufferSize, &certificate, &privateKey);
      step 2: // create a connection (websockets) from config1
                        int index = config1.networkLayersSize - 1;
                        UA_ServerNetworkLayer *wsconfig = &config1.networkLayers[index];
                        UA_Connection *wsconnection;

                        UA_UInt64 websocketpublishingCallbackId;
                        UA_Server_addRepeatedCallback(uaServer1, (UA_ServerCallback)callback_opcua,
                                                                wsconnection, PUBLISH_INTERVAL_WEBSOCKETS /2, &websocketpublishingCallbackId);

Questions: 
1. how can i obtain the handle to the websocket  connection (wsconnection) after initialising the Websocket network layer (UA_ServerConfig_addNetworkLayerWS) ?
2. callback_opcua() function already exist in open62541.c and open62541.h (almagalmated).  But i still get 'Undefined reference to callback_opcua during linking.

Noface07, 2025-06-04:
Hey @jackybek , when I check the [FEATURES](https://github.com/open62541/open62541/blob/master/FEATURES.md)

I am unable to find anything related to websockets being enabled. I was wondering can I still use opc.wss?

if no, can I use any workaround to implement it?

jackybek, 2025-06-04:
Let me check and get back to you .. I recall I was able to instantiate a websocket connection in open62541 server.

jackybek, 2025-06-08:
I checked the repo and realised that the latest version of open62541 removed the support for libwebsockets.

jackybek, 2025-06-19:
I was thinking perhaps you can use the feature of callbacks in open62541 to implement native websockets for open62541 by calling the websockets native functions directly ?

For example,
1. Callback is initiated in open62541
2. Launch websocket listening port and listens for incoming request
3. Once the websocket request is accepted, you can use the callback to read data from open62541 nodes, package the data into a xml and send it back to the requestor.
4.you can set your callback to 10 msec and it will make your system as good as realtime.

Try it.

jackybek, 2025-08-19:
finally completed the wss module for open62541.

jackybek, 2025-09-03:
you are welcome.