open62541/open62541 #4883: Why inactivityCallback is run in process publish response?

ila162, 2022-01-04:
Hello everybody,

I have a problem with inactivity mechanizm, in documentation is:
/* When connectivityCheckInterval is greater than 0,
     * every connectivityCheckInterval (in ms), a async read request
     * is performed on the server. inactivityCallback is called
     * when the client receive no response for this read request
     * The connection can be closed, this in an attempt to
     * recreate a healthy connection. */
    UA_InactivityCallback inactivityCallback;

But why inactivityCallback is run when publish response get UA_STATUSCODE_BADTIMEOUT error? After all, it isn't releated with server inactivity, client get the response from server with timeout information for publish reqest, so the server is active.

Thanks!
Greetings

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

lv-der, 2022-10-12:
Good morning,

I ran into the same issue last week and took some time to dig deeper! I am using open62541 as backend with QtOpcUa and first thought it was a bug in the QtOpcUa, so I opened an issue in Qt bugtracker - see [QTBUG-107549](https://bugreports.qt.io/browse/QTBUG-107549), you can read my full description of the bug there!

And I looks very much like @ila162 is right. The documentation of open62541 states:
```cpp
/* When connectivityCheckInterval is greater than 0, every
 * connectivityCheckInterval (in ms), an async read request is performed on
 * the server. inactivityCallback is called when the client receive no
 * response for this read request The connection can be closed, this in an
 * attempt to recreate a healthy connection. */
void (*inactivityCallback)(UA_Client *client);

// [...]

/* If the client does not receive a PublishResponse after the defined delay
 * of ``(sub->publishingInterval * sub->maxKeepAliveCount) +
 * client->config.timeout)``, then subscriptionInactivityCallback is called
 * for the subscription.. */
void (*subscriptionInactivityCallback)(UA_Client *client, UA_UInt32 subscriptionId, void *subContext);
```

So if the PublishRequest messages sent by the client expire (getting ServiceFault response with BadTimeout code), the `subscriptionInactivityCallback` callback should be called - that's what the documentation says. But as seen here, the normal `inactivityCallback` is being called: https://github.com/open62541/open62541/blob/dabaabacd2217dc61b343458b2243c5ce05c16bd/src/client/ua_client_subscriptions.c#L1099
Later in the same file, the right callback is used:
https://github.com/open62541/open62541/blob/dabaabacd2217dc61b343458b2243c5ce05c16bd/src/client/ua_client_subscriptions.c#L1217

For your convenience I attached the open62541 log messages and a Wireshark network dump as suggested:
[open62541-ServiceFault.log](https://github.com/open62541/open62541/files/9761840/open62541-ServiceFault.log)
[open62541-ServiceFault.pcapng.gz](https://github.com/open62541/open62541/files/9761849/open62541-ServiceFault.pcapng.gz)

lv-der, 2022-10-12:
Added a PR which should fix the issue. I built QtOpcUa against my code changes and it works well for me.

PR #5379 (2022-10-26): fix(client): wrong callback in UA_Client_Subscriptions_processPublishResponse
Fix wrong callback being invoked in UA_Client_Subscriptions_processPublishResponse, which led to client disconnect on PublishRequest expiration.

closes #4883
commit dcde6ee5 (2022-10-26): fix(client): wrong callback in UA_Client_Subscriptions_processPublishResponse

Fix wrong callback being invoked in UA_Client_Subscriptions_processPublishResponse, which led to
client disconnect on PublishRequest expiration.

closes #4883