open62541/open62541 #4885: HasInterface --> Not instantiating childs of supertypes

janfelixklein, 2022-01-04:
## Description

I'm currenty working with the companion spec I4AAS and came across the following issue regarding Interfaces while setting up a server:

The spec defines some custom interface types: 
[image]

If one can see `IAASIdentifiableType` is a subtype of `IAASReferableType`. `IAASReferableType` itself has a mandatory property "Category":

[image]

However, when creating an object which has an `HasInterface` reference to `IAASIdentifiableType`, the category property is not instantiated:

[image]

When instantiating an object from type `IAASIdentifiableType` directly, there appears to be no problem.
[image]

I deepdived into the source code and came across the `addInterfaceChildren` function in the `ua_services_nodemanagement.c` which uses `copyAllChildren`. Maybe `copyAllChildren` doesn't check for supertypes and their children as well?

jpfr, 2022-01-10:
Right. `copyAllChildren` does not consider the supertypes.
For the normal object instances, we first collect all supertypes and then do a copyAllChildren for all of those.
We should add that for the upcoming v1.3 release.

Do you have capacity to work on that or does it have to go on the big-pile-of-stuff-for-v1.3?

jpfr, 2022-01-10:
So this function does the children instantiation for normal types:
https://github.com/open62541/open62541/blob/7ca561e4a3d38c651224c71cbe822ae6a3e1c7b4/src/server/ua_services_nodemanagement.c#L775

This function pulls out the children to be instantiated from the `hasInterface` definition.
https://github.com/open62541/open62541/blob/7ca561e4a3d38c651224c71cbe822ae6a3e1c7b4/src/server/ua_server_utils.c#L154

Whereas the object-type case pulls out the "type hierarchy", the "hasInterface" code pulls out the actual children to be copied over.

I think we should unify the two as much as possible.
Especially when it comes to recursion (children with children), etc. there should be a common logic to that.

janfelixklein, 2022-01-11:
> Right. `copyAllChildren` does not consider the supertypes. For the normal object instances, we first collect all supertypes and then do a copyAllChildren for all of those. We should add that for the [...]

I will probably take a look at this at the weekend and try some stuff out