open62541/open62541 #4892: Open62541 with FreeRTOS TCPIP Stack

Max15421, 2022-01-06:
## Description
Hello everybody. Did somebody port open62541 to use it with the FreeRTOS TCP/IP Stack instead of the lwIP Stack? I tried to do it myself but I have problems with some function, for example with the get and set adressinfo function. Is there a reason why open62541 was initially ported to FreeRTOS + lwIP instead of the FreeRTOS TCP/IP stack?

Cheers 

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: FreeRTOS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

henrik-the, 2023-09-04:
Did you have success with porting Open62541 to the Freertos TCP Stack?

jpfr, 2023-09-25:
The initial port was done due to internal project requirements.
We can add the built-in tcp/ip stack of FreeRTOS.
But somebody needs to step up and do the work.