open62541/open62541 #4893: Variabel: DataType and type of Value

Psoke, 2022-01-06:
## Description
Hey there

I ran into a small problem lately. My program uses an open62541 client to read certain variables of an OPC UA server. To gain access to the value of a variable, I use this function:

`
UA_Client_readValueAttribute(UA_Client *client, const UA_NodeId nodeId,
UA_Variant *outValue)
`

I then use outValue.type to find out which data type I am dealing with and outValue.data to gain access to the stored value. Depending on the type I then cast it with soemthing like (int32_t*) to store it internally as an integer (for example).

Till now I thought the lets call it "valueType" would always match the DataType of the variable. But on the current server, running on a Siemens PLC, the valueType is an int32 (or atleast thats what the function returns) and the DataType is of the type double.

Can someone explain this to me? I searched the OPC UA doc and did not find a "valuetype".

I would like to treat the data according to its DataType though. 
My best guess would be this function:

`UA_Client_readDataTypeAttribute(UA_Client *client, const UA_NodeId nodeId,
UA_NodeId *outDataType)`

And then comparing the outDataType.identifier.numeric to gain knowledge about the DataType I am dealing with. The DataType Double is NodeId i=11 for example. And then cast the value accordingly, no matter what. But that seams a bit unelegant.

Also the NodeId of the BaseDataType Double (i=11) does not fit the UA_TYPES_DOUBLE (10), they are all shifted by 1.

I anybody has some suggestions, please enlighten me. :)

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-01-10:
Hey @Psoke,

the `DataType` attribute describes constraints for the `Value` attribute.
But you do right in checking the actual `variant->type` datatype once the value has arrived.
That is the "source of truth" that is never wrong. Otherwise the value couldn't be decoded once received.
The `DataType` attribute can be off for the following reasons:

- The `DataType` attribute only gives constraints, not an "actual" data type. For example there exists an abstract "numeric" data type where the actual value can be an integer, float, etc. And the `BaseDataType` allows for any content. To looks this up, the hierachy of datatypes is represented in the information model itself.
- The server could update the `DataType` attribute over time. So you are not 100% guarantueed that your last read of that value is current.
- The server could badly implement the type-constraints or straight up "lie".

All in all, the `DataType` attribute gives good indications.
But in a memory-sensitive language like C, always change the `variant->type` field as well.

Psoke, 2022-01-17:
Thanks for the reply.

One more question regarding this. 
Lets say I use this code snippet:
```
UA_Variant value;
UA_Variant_setScalar(&value, &someDoublevalue, &UA_TYPES[UA_TYPES_DOUBLE]);
UA_NodeId requestedNewNodeId = UA_NODEID_NUMERIC(2, 1000);
UA_Server_writeValue(server, requestedNewNodeId, value);
```

What exactly does UA_Variant_setScalar() set? The DataType or the "valuetype"? What happens if the Node already exists as variable of type integer and I use this code snippet, lets say in a ValueCallback? Does the "valuetype" get overwritten to type double?

o6-nbeier, 2026-10-20:
Thank you for the question.

A Variant describes the actual storage type, the node's DataType only constrains what may be written (a Double into an Int32 node gives BadTypeMismatch), so read the value with its actual type and convert it into your own destination variable.

Closing for now.

closed 2026-10-20 by o6-nbeier