open62541/open62541 #4894: Library does not compile with SMP Debian 5.15.5-2

fantomsam, 2022-01-07:
## Description
Hello, the latest release (tag: c9064a43cded44036ef831d67db22144c246963c) of the open62541 library does not compile in Debian GNU/Linux bookworm/sid x86_64 with kernel 5.15.5-2 

When I try to compile I get the following error : /open62541/arch/eventloop_posix_epoll.c:83:18: error: implicit declaration of function ‘epoll_pwait2’; did you mean ‘epoll_pwait’? [-Werror=implicit-function-declaration]

Apparently the testing Debian release of the GNU OS does not currently support epoll_pwait2(), it's does not exist also in /sys/epoll.h
Can you improve a bit the checking at line 75 of file open62541/arch/eventloop_posix_epoll.c to check except of the kernel version also the existence of epoll_pwait2().

## Background Information / Reproduction Steps

I compile with this option enabled:
```bash
cmake -D BUILD_SHARED_LIBS=ON
```
## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: SMP Debian 5.15.5-2 (2021-12-18) x86_64 GNU/Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

jpfr, 2022-01-07:
I am surprised.
The kernel documentation says: epoll_pwait2() was added to Linux in kernel 5.11.
And we do test for that version.

But yeah, we should fix the build if Debian patches that away.

fantomsam, 2022-01-07:
If you mean the Linux kernel documentation it's doesn't write anything about epoll_pwait2() or epoll_pwait() in general.
There is a reference in the LWN about a patch that adding this function in the system call sets of the kernel (https://lwn.net/ml/linux-kernel/20201118144617.986860-1-willemdebruijn.kernel@gmail.com/).

However, the include file "sys/epoll.h", is not part of the Kernel but part of the glibc (g for gnu) (https://www.gnu.org/software/gnulib/manual/html_node/Glibc-sys_002fepoll_002eh.html). By browsing a bit the docs of glibc, I can see that it's does not have yet add this function in the glibc but only reference it as port in the kernel. 
Of course I don't know if other distributions using modified version of the glibc that have this function added. Or even if I use older version of the glibc. However the latest version that provided by the debian packages for distro: bookworm/sid (testing) is 2.33-2 witch have release date 2021-02-01.

Anyway, if it's possible to make a fix for Debian testing will be very good. 

Thanks in advance.

jpfr, 2022-01-09:
Solved via #4895.

This page claims the existence since Linux 5.11
https://man7.org/linux/man-pages/man2/epoll_pwait.2.html
But apparently the syscall is not exposed in the libc.

closed 2022-01-09 by jpfr

fantomsam, 2022-01-09:
Yes that was I wrote above, The link that you reference is that man pages for Linux Programmers, also it's direct you there from the documentation of the glibc.
```
      ... 
       epoll_pwait() was added to Linux in kernel 2.6.19.  Library
       support is provided in glibc starting with version 2.6.
       epoll_pwait2() was added to Linux in kernel 5.11.
      ...
```
Not any reference for support in glibc. 

However I think that in some of the next release of the glibc will support this function.