open62541/open62541 #4896: IPv6 LDS-ME

Muddasir-Shakil, 2022-01-10:
## Description
I was working on the discovery using LDS-ME specifically for ipv6 networks and I have few points to discuss and questions to ask.

1. Open62541 relies on the third party library to support ipv4, which is anyhow a forked repo by @Pro. It is also behind the original upstream (mdnsd). mdnsd is planning to add ipv6 support in release v1.0. 
Q. Is this a sustainable solution in a long run in your opinion? 
2.  mdnsd has planned to add ipv6 support in release v1.0, which has currently no due date. However, there are other solutions for mdns implementation. One implementation might be [mjansson/mdns](https://github.com/mjansson/mdns). It supports both ipv4 and ipv6 and it provides a single header only cross-platform file. 
Q. what do you think of switching from mdnsd to another implementation?
Q. what are the pitfalls and showstoppers in doing so?
Q. What other implementations are suitable for open62541 in your opinion?
Q. Apart from external deps, size and ip network support, what could be the criteria to select mdns implementation for open62541?
3. Current implementation of LDS-ME is tightly-coupled with mdnsd implementation. Switching to other implementation would require a great effort and re-implementation of some parts in Discovery manager in open62541. I think better approach and effort well spent  would be to define a common interface for discovery manager and have plugins to integrate external mdsn implementation (we have for security modules). This will also create opportunities to integrate avahi like implementations, if the system allows it.
Q. What do you think about it and if this a solid approach how should we start working on it?

Looking forward to your feedback and discussion.

jpfr, 2022-01-10:
Hey there, thanks for the remarks.

right now the LDS-ME subsystem has quite some historical baggage.
It should be moved over to the new EventLoop subsystem so it doesn't have to open a separate thread and pull in system dependencies.

Do you have the bandwith to work on this?
Otherwise we can keep the notes for posterity and @JanSurft who is also looking into this.

I have two must criteria. The rest is up for debate:
- Integrate with the new EventLoop subsystem for the communication.
- Use an actively maintained upstream for the mdns library.

Using the EventLoop subsystem means first to create UDP support.
That is needed for mdns and for PubSub.
The TCP support should be easy enough to copy/port over to UDP.

If the upstream mdnsd project is going to supoort IPv6, great!
We can move to mjansson/mdns if that simplifies the code.
I would trade off less complexity on our end against more complexity if the mdns library is well-maintained or stable enough.

Muddasir-Shakil, 2022-01-17:
Thank you for the feedback @jpfr 
> Do you have the bandwith to work on this?

Yes. I am looking in to this issue. However, before we start implementation on it could we discuss few details like starting point, event-loop integration and steps. Also I have some questions:

 1. Implement "eventloop_posix_udp.c" (UDP Connection Manager) -> Test it in Unit Tests as done for event_loop_posix_tcp.c, is this correct?
2. How/Where to Register UDP ConnectionManager with the running Server EventLoop? Probably in UA_Server_run_startup ?
3. I could not find anywhere "TCP connectioManager" being registered with the Server_eventloop? Does it has to be registered from the App? or Am I missing something? 
4. "connectionCallback" is this the Application specific callback? Meaning does it has to be defined in ua_discovery_manager.c for LDS-ME when something is received on the UDP socket?

Muddasir-Shakil, 2022-05-03:
Hi,
I wanted to update on the status. I have created a PR for Pro/mdnsd to support ipv6. The PR can be found here [devel/ipv6](https://github.com/Pro/mdnsd/pull/28). 
The changes regarding open62541 can be found [devel/ldsme-ipv6](https://github.com/Muddasir-Shakil/open62541/tree/devel/ldsme-ipv6). 
I could create a PR but It will not compile until @Pro accepts my ipv6 PR because I added a macro for ipv6 query type records. 
let me know what do you think or should I anyhow create a PR and you can look at it and be reviewed.

GoetzGoerisch, 2022-05-03:
@JanSurft is working on integrating the mdns upstream repo here #5032.
Maybe this can go hand and hand?

Muddasir-Shakil, 2022-05-11:
@GoetzGoerisch , yes I agree with you but upstream does not have support for ipv6 and does not want to include cmake build system. That is why I added support for ipv6 in @Pro mdnsd implementation for open62541. Where as on open62541 side my implementation would not require much changes. 
Let me know what are you plans about it.