open62541/open62541 #4901: Nodeset Compiler Issue: Value of a Union.

MurzynJ, 2022-01-17:
## Description
The generated code doesn't initialize the value according to the content of the XML.

Test code:
https://github.com/MurzynJ/open62541/tree/NcIssue_UnionValue

Example:
[image]
[image]

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): Latest master: b061554990ab38cd9062212aaa69c322f1415f33
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

MurzynJ, 2022-02-03:
This issue is still open, even after the latest commits by @NoelGraf. Value of a Union is not set in the generated code.

[image]

FYI @Fischer-Thomas

andreasebner, 2022-02-28:
@MurzynJ The issue should be solved, can we close it?

MurzynJ, 2022-02-28:
Sorry, I haven't had time to check the fix yet. But I trust that it's working. If I find anything in future, I will just create another issue.

closed 2022-02-28 by MurzynJ