open62541/open62541 #4903: Error=Cannot discover applications from opc.tcp://localhost:4840

jarkko-hautakorpi, 2022-01-19:
## Description
v0.3 examples (with amalgamation) produce working servers, where other clients like [Prosys OPC UA Client](https://downloads.prosysopc.com/opc-ua-client-downloads.php) could connect and browse/subscribe to nodeId.
v1.0 no longer produces compatible servers, maybe discovery service is incompatible/missing?
## Background Information / Reproduction Steps
after building and installing with below cmake otiopns, build examples with
```
cd examples && make build && cd build
cmake .. && make
```
Run any example server, and Prosys Client disconnects immediately. Server does show it connecting, but disconnects immediately.

```
./server_ctt (or any other server example)
info/network       Connection 5 | New connection over TCP from 127.0.0.1
info/channel       Creating a new SecureChannel
warn/securitypolicy        Security policy None is used to create SecureChannel. Accepting all certificates
info/channel       Connection 5 | SecureChannel 5 | Opened SecureChannel
warn/channel       Connection 5 | SecureChannel 5 | The server sends no timestamp in the request header. See the 'verifyRequestTimestamp' setting.
info/channel       Connection 5 | SecureChannel 5 | CloseSecureChannel
info/network       Connection 5 | Closed
```

Used CMake options:
```bash
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_SHARED_LIBS=ON \
  -DUA_NAMESPACE_ZERO=FULL \
  -DUA_ENABLE_DISCOVERY=ON \
  -DUA_ENABLE_DISCOVERY_MULTICAST=ON \
  -DUA_ENABLE_SUBSCRIPTIONS=ON \
  -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON \
  -DUA_ENABLE_QUERY=ON ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 18.04 and Debian11
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jarkko-hautakorpi, 2022-01-19:
v0.3 example servers can be browsed with client
[image]

jarkko-hautakorpi, 2022-01-19:
Seems Services are no longer on by default in examples/configurations.
_FindServersRequest_ response _FindServersResponse_:servers returns empty array, no servers found..

jpfr, 2022-02-01:
We have release the v1.1 and v1.2 release families since.
Do you have a real need to stay on the v1.0 version?

jarkko-hautakorpi, 2022-02-02:
Qt 5.15 qtopcua is tied to version "v1.0". Server version shouldn't matter tough, you are right..

closed 2022-02-04 by jarkko-hautakorpi