open62541/open62541 #4905: Pass context to method callback

Psoke, 2022-01-19:
## Description
Is it possible to pass data to a method callback? 

I managed doing this with a value callback using the UA_Server_setNodeContext() and UA_Server_setVariableNode_valueCallback() to pass a pointer of my own data through the void *nodeContext. I used this to place a universal callback on many variable nodes while incrementing the pointer, so I do not have to explicitly code every value callback.

Now I would like to do the same with a method callback, UA_Server_addMethodNode() even has a void *nodeContext in its arguments, but how do I access my information in the callback?

```
typedef UA_StatusCode
(*UA_MethodCallback)(UA_Server *server, const UA_NodeId *sessionId,
                     void *sessionContext, const UA_NodeId *methodId,
                     void *methodContext, const UA_NodeId *objectId,
                     void *objectContext, size_t inputSize,
                     const UA_Variant *input, size_t outputSize,
                     UA_Variant *output);
```

I tried the methodContext aswell as the objectContext but cant get it to work. Is this even possible?
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-01-20:
If it doesn't work, then it is a bug.

The method can be referenced from several objects.
Hence the methode node and object node are separate attributes in the callback.

So you get the nodeId and the context pointer from both the method and the object node.
You can set them in the usual way.

Psoke, 2022-01-20:
Here is a simplified part of my method and the coresponding callback. Maybe I am just doing something wrong.

```
static UA_StatusCode
MethodCallbackUniversal (UA_Server *server, const UA_NodeId *sessionId, void *sessionHandle, 
                         const UA_NodeId *methodId, void *methodContext, 
                         const UA_NodeId *objectId, void *objectContext, size_t inputSize,
                         const UA_Variant *input, size_t outputSize, UA_Variant *output)
{
	UA_StatusCode retval = UA_STATUSCODE_GOOD;

	int i = *(int *) methodContext;	

	printf("WUHU IT WORKED: %d\n", i);

	if (input == cool)
	{
		UA_Boolean ReturnValue = UA_TRUE;
		UA_Variant_setScalarCopy(&output[0], ...
		UA_Variant_setScalarCopy(&output[1], ...
		UA_Variant_setScalarCopy(&output[2], ...	
	}
	if (input == notcool)
	{
		UA_Boolean ReturnValue = UA_FALSE;
		UA_Variant_setScalarCopy(&output[0], ...
		UA_Variant_setScalarCopy(&output[1], ...
		UA_Variant_setScalarCopy(&output[2], ...	
	}
	return retval;
}
```

```
static UA_StatusCode
addMethod (UA_Server *server, UA_NodeId requestedNewNodeId, UA_NodeId parentNodeId)
{
	UA_StatusCode retval = UA_STATUSCODE_GOOD;

	UA_Argument inputArguments[4];

	UA_Argument_init ...

	UA_Argument outputArguments[3];

	UA_Argument_init ...

	UA_MethodAttributes attr = UA_MethodAttributes_default;
	attr.description = UA_LOCALIZEDTEXT("en-US", "Description");
	attr.displayName = UA_LOCALIZEDTEXT("en-US", "DisplayName");
	attr.executable = true;
	attr.userExecutable = true;

	UA_NodeId referenceTypeId = UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT);
	UA_QualifiedName browseName = UA_QUALIFIEDNAME(requestedNewNodeId.namespaceIndex, "BrowseName");

	int *pointer = &SOMEGLOBALVARIABLE;

	UA_Server_setNodeContext(server, requestedNewNodeId, pointer); //even necessary?

	retval = UA_Server_addMethodNode(server, requestedNewNodeId, parentNodeId, referenceTypeId,
                                         browseName, attr, MethodCallbackUniversal, 4, inputArguments,
                                         3, outputArguments, pointer, NULL);
	return retval;
}
```

I point to a global variable that exists throughout the entire program and has a value assigned at the moment of calling the method. I do not know if I even need to set the NodeContext via function or just pass it with the UA_Server_addMethodNode(), tried both, both failed. 

In the callback I then dereference the void *methodContext but it just outputs 0 or random values.

closed 2022-01-20 by Psoke