open62541/open62541 #4927: UA_Server_addVariableNode does not work on Ubuntu 20.04 -- BadNodeAttributesInvalid -- Works on Ubuntu 18.04

ftpronk, 2022-02-03:
## Description
A simple server example, adding an int32 node fails on `Ubuntu 20.04`, with error `BadNodeAttributesInvalid`.

- git commit: tag **v1.2.3**
- compiler: clang-10

## Background Information / Reproduction Steps
 - checkout git tag **v1.2.3** (commit bb1abfc881)
 - set clang-10:
 ```
export CC=/usr/bin/clang-10
```
- disable `check_add_cc_flag("-fno-ident")` in `CMakeLists.txt` (see bug #4559)
 - call cmake, with options: 
 ```bash
 cmake ../ -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=Release -DUA_NAMESPACE_ZERO=FULL -DCMAKE_INSTALL_PREFIX=~/local-opcua/ 
```
 - compile and install

The example program used (from the documentation):

```C
#include <signal.h>                                                                                 
#include <open62541/server.h>                                                                       
#include <open62541/server_config_default.h>                                                        
                                                                                                    
UA_Boolean running = true;                                                                          
void signalHandler(int sig) {                                                                       
    running = false;                                                                                
}                                                                                                   
                                                                                                    
int main(int argc, char** argv)                                                                     
{                                                                                                   
    signal(SIGINT, signalHandler); /* catch ctrl-c */                                               
                                                                                                    
    /* Create a server listening on port 4840 */                                                    
    UA_Server *server = UA_Server_new();                                                            
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));                                        
                                                                                                    
    /* Add a variable node */                                                                       
    /* 1) Define the node attributes */                                                             
                                                                                                    
    UA_VariableAttributes attr = UA_VariableAttributes_default;                                     
    UA_Int32 myInteger = 42;                                                                        
    UA_Variant_setScalar(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);                       
    attr.description = UA_LOCALIZEDTEXT("en-US","the answer");                                      
    attr.displayName = UA_LOCALIZEDTEXT("en-US","the answer");                                      
    attr.dataType = UA_TYPES[UA_TYPES_INT32].typeId;                                                
    //attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;                        
                                                                                                    
                                                                                                    
    /* 2) Define where the node shall be added with which browsename */                             
    UA_NodeId newNodeId = UA_NODEID_STRING(1, "the.answer");                                        
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);                          
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);                     
    UA_NodeId variableType = UA_NODEID_NULL; /* take the default variable type */                   
    UA_QualifiedName browseName = UA_QUALIFIEDNAME(1, "the answer");                                
                                                                                                    
    /* 3) Add the node */                                                                           
    UA_Server_addVariableNode(server, newNodeId, parentNodeId, parentReferenceNodeId,               
                              browseName, variableType, attr, NULL, NULL);                          
                                                                                                    
    /* Run the server loop */                                                                       
    UA_StatusCode status = UA_Server_run(server, &running);                                         
                                                                                                    
    UA_Server_delete(server);                                                                       
    return status;                                                                                  
} 
```

- compile with:
```bash
clang-10  server.c -o myServer -lopen62541 -L/home/user/local-opcua/lib
```
Output:

```
[2022-02-03 12:27:24.992 (UTC+0100)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2022-02-03 12:27:24.992 (UTC+0100)] info/server        AccessControl: Anonymous login is enabled
[2022-02-03 12:27:24.992 (UTC+0100)] warn/server        Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2022-02-03 12:27:24.992 (UTC+0100)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2022-02-03 12:27:24.992 (UTC+0100)] info/session       SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNode: Node could not create a node with error code BadNodeAttributesInvalid
[2022-02-03 12:27:24.992 (UTC+0100)] info/network       TCP network layer listening on opc.tcp://herring:4840/
^C[2022-02-03 12:27:29.740 (UTC+0100)] info/network     Shutting down the TCP network layer
```
Specifically:
```
AddNode: Node could not create a node with error code BadNodeAttributesInvalid
```

The same process works on Ubuntu **18.04**, also with `clang-10`.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): tag **v.1.2.3**, commit bb1abfc8810fa8226f3d1319911689c6b760c53a
 - [x] Operating system: `Ubuntu 20.04`
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached  (**Not added, no extra useful information**)
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue

ftpronk, 2022-02-03:
As extra information, it also fails with builds made using
- clang 8.0.1-9 
- clang 12.0.0
- gcc 9.3.0

Lukas1811, 2022-03-21:
I currently facing the same issue, everything worked without issues on 18.04 but throws "BadNodeAttributesInvalid" when compiled on 20.04. But the strange thing is that I'm not even executing it on the Ubuntu machine, I cross compile it using Linaro-gcc and run it on an Yocto Linux system.

In my application the error is created while adding the DI nodeset with the generated files. I assume the UA nodeset isn't added the right way  because if i try to get the namespace index for "http://opcfoundation.org/UA" I get an "BadNotFound" error.

Have you found any solution or fix yet? 

The cmake command I'm using:

```bash
cmake -DCMAKE_C_COMPILER=$LINARO_GCC \
      -DBUILD_SHARED_LIBS=ON \
      -DUA_NAMESPACE_ZERO=FULL \
      -DUA_ENABLE_ENCRYPTION=ON \
      -DUA_BUILD_EXAMPLES=OFF \
      -DUA_ENABLE_AMALGAMATION=OFF \
      -DUA_ENABLE_SUBSCRIPTIONS=YES \
      -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=YES \
      -DUA_ENABLE_METHODCALLS=ON \
      -DUA_MULTITHREADING=100 \
      -DMBEDCRYPTO_LIBRARY=$MBEDTLS_DIR/build/library/libmbedcrypto.so \
      -DMBEDTLS_INCLUDE_DIRS=$MBEDTLS_DIR/build/include \
      -DMBEDTLS_LIBRARY=$MBEDTLS_DIR/build/library/libmbedtls.so \
      -DMBEDX509_LIBRARY=$MBEDTLS_DIR/build/library/libmbedx509.so \
      -DUA_ENABLE_PAM=ON \
      -DPAM_LIBRARY=$PAM_LIBPAM_DIR \
      -DPAM_MISC_LIBRARY=$PAM_LIBPAM_MISC_DIR \
      -DPAM_INCLUDE_DIRS=$PAM_INCLUDE_DIR \
      -DPAM_MISC_INCLUDE_DIRS=$PAM_MISC_INCLUDE_DIR \
      -DPAMC_INCLUDE_DIRS=$PAMC_INCLUDE_DIR ..
```

I generate the DI nodeset using:
`make open62541-generator-ns-di`
To add it in the application I execute:
`namespace_di_generated(server)`
which results in the mentioned "BadNodeAttributesInvalid"

The only output of the server I get is:
`[2022-03-21 11:39:42.361 (UTC+0000)] info/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-00000000000d`

Another reason I think it has to do with the creation of the UA namespace is that if I ignore the error in my code the creation of event types inheriting from base event type also fail and they only inherit from UA namespace objects.

piperoc, 2022-09-07:
I ran into a similar issue.
The way I fixed this was by changing UA_Variant_setScalar to UA_Variant_setScalarCopy.
I'm still trying to understand the difference, but I guess they make different use of the memory.
Let me know if this works, I'm curious

happybruce, 2023-02-09:
> I currently facing the same issue, everything worked without issues on 18.04 but throws "BadNodeAttributesInvalid" when compiled on 20.04. But the strange thing is that I'm not even executing it on the Ubuntu machine, I cross compile it using Linaro-gcc and run it on an Yocto Linux system.
> 
> In my application the error is created while adding the DI nodeset with the generated files. I assume the UA nodeset isn't added the right way because if i try to get the namespace index for "http://opcfoundation.org/UA" I get an "BadNotFound" error.
> 
> Have you found any solution or fix yet?
> 
> The cmake command I'm using:
> 
> ```shell
> cmake -DCMAKE_C_COMPILER=$LINARO_GCC \
>       -DBUILD_SHARED_LIBS=ON \
>       -DUA_NAMESPACE_ZERO=FULL \
>       -DUA_ENABLE_ENCRYPTION=ON \
>       -DUA_BUILD_EXAMPLES=OFF \
>       -DUA_ENABLE_AMALGAMATION=OFF \
>       -DUA_ENABLE_SUBSCRIPTIONS=YES \
>       -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=YES \
>       -DUA_ENABLE_METHODCALLS=ON \
>       -DUA_MULTITHREADING=100 \
>       -DMBEDCRYPTO_LIBRARY=$MBEDTLS_DIR/build/library/libmbedcrypto.so \
>       -DMBEDTLS_INCLUDE_DIRS=$MBEDTLS_DIR/build/include \
>       -DMBEDTLS_LIBRARY=$MBEDTLS_DIR/build/library/libmbedtls.so \
>       -DMBEDX509_LIBRARY=$MBEDTLS_DIR/build/library/libmbedx509.so \
>       -DUA_ENABLE_PAM=ON \
>       -DPAM_LIBRARY=$PAM_LIBPAM_DIR \
>       -DPAM_MISC_LIBRARY=$PAM_LIBPAM_MISC_DIR \
>       -DPAM_INCLUDE_DIRS=$PAM_INCLUDE_DIR \
>       -DPAM_MISC_INCLUDE_DIRS=$PAM_MISC_INCLUDE_DIR \
>       -DPAMC_INCLUDE_DIRS=$PAMC_INCLUDE_DIR ..
> ```
> 
> I generate the DI nodeset using: `make open62541-generator-ns-di` To add it in the application I execute: `namespace_di_generated(server)` which results in the mentioned "BadNodeAttributesInvalid"
> 
> The only output of the server I get is: `[2022-03-21 11:39:42.361 (UTC+0000)] info/session Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-00000000000d`
> 
> Another reason I think it has to do with the creation of the UA namespace is that if I ignore the error in my code the creation of event types inheriting from base event type also fail and they only inherit from UA namespace objects.

hi Lukas1811, I run `make open62541-generator-ns-di` in build folder but fail, do I need other configuration?

sotoudi, 2025-07-24:
I had the same issue compiling on the Petalinux, what I did was to compile my code with the open62541.c together and not using the libraries from the Linux.
gcc my_server.c open62541.c -o opcua_server -lm
it worked for me, maybe it also works for you!

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We could not reproduce it on 1.4 or later, and versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier