open62541/open62541 #4928: compiling open62651.c/.h using gnu c++ in Ubuntu 18.04

jackybek, 2022-02-04:
## Description

## Background Information / Reproduction Steps

I want to recompile open62541.c using gnu c++ in Ubuntu 18.04.
The purpose is to link with some c++ libraries.
Is there any guide on how to go about this?

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-02-11:
Nothing special.
Any tutorial on the internet for mixing c and c++ applies.

and as a plus, the open62541.c is also valid C++98.
So you can also throw everything at the C++ compiler.

closed 2022-02-11 by jpfr

jackybek, 2022-02-12:
Thanks Julius.
In this case since you mentioned open62541 can be compiled by g++,  i tried to configure ccmake .. and change the following:

CMAKE_C_COMPILER = /usr/bin/g++
CMAKE_C_FLAGS = -x c++ -std=c++98 -v -fpermissive -Wno-literal-suffix -Wno-write-strings -Wno-long-long -Wno-return-type
CMAKE_EXE_LINKER_FLAGS = -x c++ -std=c++98 -v -fpermissive -Wno-literal-suffix -Wno-write-strings -Wno-long-long -Wno-return-type

After i hit 'c' i get the following messages:

    You have changed variables that require your cache to be deleted. Configure will be re-run and you may have to reset some variables.  The following variables have changed:
    CMAKE_C_COMPILER= /usr/bin/g++

    CMake Error at /usr/share/cmake-3.10/Modules/CMakeTestCCompiler.cmake:52 (message):
    The C compiler

        "/usr/bin/g++"

    is not able to compile a simple test program.
    It fails with the following output:

       Change Dir: /home/jackybek/open62541/build/CMakeFiles/CMakeTmp

       Run Build Command:"/usr/bin/make" "cmTC_4b334/fast"
       /usr/bin/make -f CMakeFiles/cmTC_4b334.dir/build.make CMakeFiles/cmTC_4b334.dir/build
      make[1] : Entering directtory '/home/jackybek/open62541/build/CMakeFiles/CMakeTmp'
       Building C object CMakeFiles/cmTC_4b334.dir/testCCompiler.c.o
       /usr/bin/g++      -o CMakeFiles/cmTC_4b334.dir/testCCompiler.c.o    -c /home/jackybek/open62541/build/CMakeFiles/CMakeTmp/testCCompiler.c
       /home/jackybek/open62541/build/CMakeFiles/CMakeTmp/testCCompiler.c:2:3: error #error "The CMAKE_C_COMPILER is set to a C++ compiler"
       # error "THE CMAKE_C_COMPILER is set to a C++ compiler"
       CMakeFiles/cmTC_4b334.dir/build.make:65: receipe for target 'CMakeFiles/cmTC_4b334.dir/testCCompiler.c.o' failed
       make[1]: *** [CMakeFiles/cmTC_4b334.dir/testCCompiler.c.o] Error 1
       make[1]: Leaving directory '/home/jackybek/open62541/build/CMakeFiles/CMakeTmp'
       Makefile:126: recipe for target 'cmTC_4b334/fast' failed
      make: *** [cmTC_4b334/fast] Error 2

      CMake will not be able to correctly generate this project.
      Call Stack (most recent call first):
          CMakeLists.txt:5 (project)