open62541/open62541 #4939: UA_Server_run_iterate(server, waitInternal=true) sometimes returns 65535ms

paddor, 2022-02-09:
I'm pretty sure that's a bug because that's just all bits of the `UA_UInt16` return value set to 1, and the documentation says the default max wait time is 50ms. I noticed my app hanging for 65s at a time and found this.

closed 2022-02-11 by jpfr

paddor, 2022-02-11:
I didn’t have a 65s callback scheduled though. Why would it return 65535?

What if I don’t want the function to wait (and block my own event loop) and still want to know when the next callback is scheduled? If I pass waitInternal=false, the return value is always zero.

paddor, 2022-02-14:
Why does it return 65535ms if I don't use any timers?

Why is the return value always 0ms if I pass waitInternal=false?

mlang-de, 2022-02-15:
Maybe check this line
https://github.com/open62541/open62541/blob/5d24520b078669a2212749642447d89de311cb45/src/server/ua_server.c#L715

paddor, 2022-02-15:
@mlang-de thank you.