open62541/open62541 #4943: Only allow for locally hosted applications to connect to the OPC-UA server

VictorWiedemann, 2022-02-11:
## Description
I hope it is fine to ask this here. I am trying to find information to do one of the following.

Connect via a PEM file as I see on UaExpert. I see that I can allow a trust list of the pem files on the server side, but I cannot find the way to initiate the connection on the client side with the pem file set.

OR

Since all of the applications are run locally on a single computer, only allow IP address 127.0.0.1 to be able to connect. This way, there is no issue with 

## Background Information / Reproduction Steps

Used CMake options:
```
cmake \
    -DUA_ENABLE_AMALGAMATION=ON \
    -DUA_ENABLE_METHODCALLS=ON \
    -DUA_ENABLE_SUBSCRIPTIONS=ON \
    -DUA_NAMESPACE_ZERO=FULL \
    -DUA_ENABLE_DA=ON \
    -DUA_ENABLE_DISCOVERY=OFF \
    -DUA_ENABLE_DISCOVERY_MULTICAST=OFF \
    -DUA_ENABLE_MICRO_EMB_DEV_PROFILE=OFF \
    -DUA_ENABLE_ENCRYPTION=ON \
    -DUA_ENABLE_ENCRYPTION_MBEDTLS=ON \
    -DUA_ENABLE_ENCRYPTION_OPENSSL=ON \
```

## Checklist

Please provide the following information:

 - [ x] open62541 Version (release number or git tag): v1.2.3
 - [ x] Other OPC UA SDKs used (client or server):  client and server
 - [ x] Operating system: Yocto
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

VictorWiedemann, 2022-02-11:
I have seen this post here: https://github.com/open62541/open62541/issues/3273 which shows that it is possible to get the trust list working, but I cannot find any information about how to set it up.

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We could not reproduce it on 1.4 or later, and versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier