open62541/open62541 #4945: a more detailed walk through of how to build on windows

om327, 2022-02-12:
Hi, Sorry for the most basic of questions but I'm new to programming in general (LabVIE background!) 
I've only built very simple programs in the past with gcc following online tutorials from microsoft. I've been using python opcua-async as a test bed but would like to install and build open62541. I've looked through the  README and install sections but they assume a lot of existing knowledge. I've installed cmake using the vs studio toolkit for vs code. I've then cloned the respository using git and tried to build using vs code - selecting the cmake builder 

[image]

Unfortaunly i keep hitting build errors. I've then tried the pre-build pacakages but i don't understand how to use them or where to save them?

Could some please explain what i need to do on a windows system to clone, build and use this library and possibly update the install instructions?

Many thanks!
Ollie

ccvca, 2022-03-24:
> [build] file cannot create directory: C:/Program Files/open62541/lib. Maybe need
> [build] administrative privileges.

You tried to install the library to `C:/Program Files/open62541/lib`. You may either only build the library, specify a different install location, where you have permissions (e.g. `-DCMAKE_INSTALL_PREFIX:PATH=C:/Users/Shadow/projects/install` ), or run the install command as an administrator.

Kaz040, 2024-07-08:
> I've managed to point cmaker to vs code and can now call the build command whicd generates abuild directory with a range of files including a .soln
> 
> the out put of the build shows there were some errors:
> 
> [main] Building folder: open62541 [main] Configuring folder: open62541 [proc] Executing command: "C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -Hc:/Users/Shadow/projects/open62541 -Bc:/Users/Shadow/projects/open62541/build -G "Visual Studio 17 2022" -T host=x64 -A x64 [cmake] Not searching for unused variables given on the command line. [cmake] -- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044. [cmake] -- The C compiler identification is MSVC 19.30.30709.0 [cmake] -- Detecting C compiler ABI info [cmake] -- Detecting C compiler ABI info - done [cmake] -- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.30.30705/bin/Hostx64/x64/cl.exe - skipped [cmake] -- Detecting C compile features [cmake] -- Detecting C compile features - done [cmake] -- Found Python3: C:/msys64/mingw64/bin/python3.exe (found version "3.9.10") found components: Interpreter [cmake] -- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.35.1.windows.2") [cmake] -- open62541 Version: v1.3.0-rc1-243-gdc8c9562 [cmake] -- CMAKE_BUILD_TYPE not given; setting to 'Debug' [cmake] -- The selected architecture is: win32 [cmake] -- Could NOT find Sphinx (missing: SPHINX_EXECUTABLE) [cmake] -- Could NOT find LATEX (missing: LATEX_COMPILER) [cmake] -- Configuring done [cmake] -- Generating done [cmake] -- Build files have been written to: C:/Users/Shadow/projects/open62541/build [kit] Patch Windows SDK bin path from C:\Program Files (x86)\Windows Kits\10\bin\x64 to C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\x64 for C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Auxiliary\Build\vcvarsall.bat [build] Starting build [proc] Executing command: "C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" --build c:/Users/Shadow/projects/open62541/build --config Debug --target install -j 10 -- [build] Microsoft (R) Build Engine version 17.0.0+c9eb9dd64 for .NET Framework [build] Copyright (C) Microsoft Corporation. All rights reserved. [build] [build] Checking Build System [build] Generating src_generated/open62541/nodeids.h [build] Generating src_generated/open62541/transport_generated.c, src_generated/open62541/transport_generated.h, src_generated/open62541/transport_generated_handling.h [build] Generating src_generated/open62541/statuscodes.h, src_generated/open62541/statuscodes.c [build] Building Custom Rule C:/Users/Shadow/projects/open62541/CMakeLists.txt [build] Building Custom Rule C:/Users/Shadow/projects/open62541/CMakeLists.txt [build] Generating src_generated/open62541/types_generated.c, src_generated/open62541/types_generated.h, src_generated/open62541/types_generated_handling.h [build] Building Custom Rule C:/Users/Shadow/projects/open62541/CMakeLists.txt [build] Generating src_generated/open62541/namespace0_generated.c, src_generated/open62541/namespace0_generated.h [build] INFO:**main**:Preprocessing C:/Users/Shadow/projects/open62541/tools/schema/Opc.Ua.NodeSet2.Reduced.xml [build] INFO:**main**:Preprocessing C:/Users/Shadow/projects/open62541/tools/schema/Opc.Ua.NodeSet2.EventsMinimal.xml [build] INFO:**main**:Preprocessing C:/Users/Shadow/projects/open62541/tools/schema/Opc.Ua.NodeSet2.Part8_Subset.xml [build] INFO:**main**:Generating Code for Backend: open62541 [build] INFO:**main**:NodeSet generation code successfully printed [build] Building Custom Rule C:/Users/Shadow/projects/open62541/CMakeLists.txt [build] Generating src_generated/open62541/statuscodes.h, src_generated/open62541/statuscodes.c [build] Building Custom Rule C:/Users/Shadow/projects/open62541/CMakeLists.txt [build] Generating src_generated/open62541/nodeids.h [build] Building Custom Rule C:/Users/Shadow/projects/open62541/CMakeLists.txt [build] Building Custom Rule C:/Users/Shadow/projects/open62541/CMakeLists.txt [build] ua_log_stdout.c [build] ua_types.c [build] ua_accesscontrol_default.c [build] ua_types_encoding_binary.c [build] ua_nodestore_ziptree.c [build] ua_types_print.c [build] ua_nodestore_hashmap.c [build] types_generated.c [build] ua_config_default.c [build] transport_generated.c [build] statuscodes.c [build] ua_pki_none.c [build] ua_util.c [build] ua_securitypolicy_none.c [build] ua_timer.c [build] ua_clock.c [build] ua_architecture_functions.c [build] ua_connection.c [build] network_tcp.c [build] ua_securechannel.c [build] ua_securechannel_crypto.c [build] eventloop_posix.c [build] ua_session.c [build] eventloop_posix_select.c [build] eventloop_posix_epoll.c [build] ua_nodes.c [build] eventloop_posix_tcp.c [build] ua_server.c [build] eventloop_posix_interrupt.c [build] ua_server_ns0.c [build] Generating Code... [build] ua_server_ns0_diagnostics.c [build] ua_server_config.c [build] open62541-plugins.vcxproj -> C:\Users\Shadow\projects\open62541\build\open62541-plugins.dir\Debug\open62541-plugins.lib [build] ua_server_binary.c [build] ua_server_utils.c [build] ua_server_discovery.c [build] Generating Code... [build] Compiling... [build] ua_server_async.c [build] ua_pubsub_networkmessage.c [build] ua_pubsub_writer.c [build [... cut ...] build] C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(156,5): error MSB3073: :VCEnd" exited with code 1. [C:\Users\Shadow\projects\open62541\build\install.vcxproj] [build] Build finished with exit code 1
> 
> When i open the .sln i don't understadn how i then add and use the the example code (myServer.c) from teh documentation?
> 
> [image]

Hi @om327 , Please how are you able to proceed with this problem? I am having the same issue.