open62541/open62541 #4950: How to login using certificates?

rexsum, 2022-02-14:
## Description

I have been struggle for several days now trying to figure out how to code my client to authenticate to a server using certificates. The provided examples show how to encrypt the channel using certificates and also how to login using username/password, but not, unless I'm missing something, how to login using a certificate.

With UA Expert I'm able to use my certificate to successfully authenticate and can confirm that it's working with an X509IdentityToken. However my client code always authenticates as AnonymousIdentityToken.

I've set securityMode, securityPolicyUri and applicationUri which does encrypt the channel, but doesn't authenticate as I need.

I've search the documentation, other Issues here, and the Google group, but none seem to fully answer this question. Others have ask this or similar questions but all w/o an adequate response. The closest I've seen is a response from Julius that mentions the need to do some manual setup but he doesn't go into detail.

Can someone please point me to a code snippet or some working code that demonstrates certificate authentication in a client? Thanks!

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_SHARED_LIBS=ON -DUA_LOGLEVEL=300 -DUA_MULTITHREADING=100 \
      -DUA_ENABLE_METHODCALLS=ON -DUA_ENABLE_SUBSCRIPTIONS=ON \
      -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_NAMESPACE_ZERO=REDUCED \
      -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON -DUA_ENABLE_PUBSUB_ETH_UADP=ON \
      -DUA_ENABLE_IMMUTABLE_NODES=ON -DUA_ENABLE_MALLOC_SINGLETON=ON -DUA_ENABLE_PUBSUB_BUFMALLOC=ON \
      -DUA_ENABLE_PUBSUB_ENCRYPTION=ON -DUA_ENABLE_ENCRYPTION_TPM2=ON -DUA_ENABLE_ENCRYPTION=OPENSSL \
      ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jackybek, 2022-03-18:
you can refer to github:
open62541/examples/encryption/client_encryption.c

rexsum, 2022-03-18:
@jackybek, as I explained above, that example simply encrypts the communications channel, it does not use the certificate to authenticate. Julius responded in the Google groups to my cross-post that it's not currently possible since client authentication using certs is not fully implemented.

jackybek, 2025-04-13:
Hi resume, have you resolve this ? Ping me if you need help.

tboeckel, 2026-01-30:
Just to be a bit more specific, this combination of commands works:
`./server_encryption_filestore server_cert.der server_key.der pki`
`./client_encryption opc.tcp://localhost:4840 client_cert.der client_key.der`

But this combination does not work:
`./server_encryption_filestore server_cert.der server_key.der pki`
`./client_encryption opc.tcp://localhost:4840 client_cert.der client_key.der  --serverCert server_cert.der`

Please note that the only difference is the usage of the --serverCert option for the client application. The server_cert.der file was put in all possible "trusted" directories, but it doesn't seem to be treated as trusted.

tboeckel, 2026-02-06:
I finally managed to implement the authentication via user+password and via certificate alternatively using the official 1.5.0 release. This required a different management of available endpoints and correctly setting up user identity tokens for each case in my C++ wrapper.

Many thanks for your hints anyway.

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier