open62541/open62541 #4960: How to distinguish variant data types safely?

paddor, 2022-02-16:
## Description

## Background Information / Reproduction Steps

I'm building a Ruby extension for Open62541. Sometimes I have to convert between Ruby and UA data types.

How to distinguish a variant's data type at runtime? Since the `switch` control structure in C can only work with integer types, this is what I've been doing (see `<-----`):

```c
static VALUE Server_m_update_uint_variable(VALUE self, VALUE node_id, VALUE ruby_value) {
  ServerData         *server_data;
  UA_NodeId          nodeId;
  UA_Variant         variant;
  UA_StatusCode      rc;
  unsigned long long value;

  TypedData_Get_Struct(self, ServerData, &ServerData_type, server_data); /* unwrap */

  nodeId = UA_NODEID(StringValueCStr(node_id));
  value  = NUM2ULL(ruby_value);
  rc     = UA_Server_readValue(server_data->server, nodeId, &variant);
  CHECK_STATUS_AND_RAISE(rc);

  switch (variant.type - UA_TYPES) // <----------
  {
    case UA_TYPES_BYTE:
      if (value > UA_BYTE_MAX)
        rb_raise(rb_eRangeError, "argument %+"PRIsVALUE" does not fit into UA_Byte", ruby_value);
      UA_Variant_setScalar(&variant, &value, &UA_TYPES[UA_TYPES_BYTE]);
      break;
    case UA_TYPES_UINT16:
      if (value > UA_UINT16_MAX)
        rb_raise(rb_eRangeError, "argument %+"PRIsVALUE" does not fit into UA_UInt16", ruby_value);
      UA_Variant_setScalar(&variant, &value, &UA_TYPES[UA_TYPES_UINT16]);
      break;
# ...
```

Is that safe? Is there a better way?

In the online API documentation I found that there's `UA_DataType.typeIndex`, but that doesn't actually exist anymore. There's also `UA_DataType.typeKind`, but it says it's used internally, and the constants to compare it to are not `UA_TYPES_XXX` but `UA_DATATYPEKIND_XXX`. Is this safe to use?

I'm using ce12009917fdce230f6b0175eb1332eaf9553480 (master).

jpfr, 2022-02-19:
No, this is not safe.
Because users can add custom types that are then not in the UA_TYPES array.
The data type kind can be safely used.
Besides that, if pointer comparisons are not possible, use the type nodeid or type name string.

paddor, 2022-02-19:
Thank you. I’ll `switch` against the `typeKind` then.

closed 2022-02-19 by paddor