open62541/open62541 #4961: OPCUA PUBSUB over TSN: Publish failed after configuring etf

ChangfengLi, 2022-02-17:
## Description
Hello
My system is ubuntu 18.04
The kernel is 4.19.37-rt19.
NIC: Intel I210
Before I setup etf, pubsub_ TSN_ Publisher works normally。
But after I setup etf, pubsub_ TSN_ Publisher failed to run。
The command to start ETF：
```
sudo tc qdisc add dev <I210 interface> parent root mqprio num_tc 3 map 2 2 1 0 2 2 2 2 2 2 2 2 2 2 2 2 queues 1@0 1@1 2@2 hw 0
MQPRIO_NUM=`sudo tc qdisc show | grep mqprio | cut -d ':' -f1 | cut -d ' ' -f3`
sudo tc qdisc add dev <I210 interface> parent $MQPRIO_NUM:1 etf offload clockid CLOCK_TAI delta 150000
sudo tc qdisc add dev <I210 interface> parent $MQPRIO_NUM:2 etf offload clockid CLOCK_TAI delta 150000
```
The command I run：
```
./bin/examples/pubsub_TSN_publisher -interface enp1s0
```
Error reporting information
```
[2022-02-17 19:05:25.672 (UTC+0800)] error/server	Publish failed. sendBufferedNetworkMessage failed. StatusCode BadInternalError
```
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_ETH_UADP=ON ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0f5bc8f(master)
 - [ ] Operating system: Ubuntu 18.04

o6-nbeier, 2026-10-20:
We were not able to reproduce this. If you still encounter this issue, please open a new issue with up to date information.

Closing for now.

closed 2026-10-20 by o6-nbeier

ChangfengLi, 2026-10-20:
邮件已收到。