open62541/open62541 #4964: Win32 Shutdown takes 120s when (unresponsive?) clients connected

ccvca, 2022-02-18:
## Description
When `running` is set to false, `UA_Server_run` takes ~120s to return, when a Client is still connected.

## Background Information / Reproduction Steps
The error was discovered here, after upgrading this test to the new open62541-Version with the event-loop:
https://github.com/umati/Sample-Server/blob/690cb0e590eeaaf3ee4c1e085995944b1b785fb4/tests/Test_BindWrongDatatype.cpp

- Start Server
- Connect Client
- Do some work
- Do not disconnect the client, do not call run_iterate (do not use the client anymore)
- Set running of `UA_Server_run` to false
- Wait ~120s until the Server is shut down

## Debugging

https://github.com/open62541/open62541/blob/5f5214427b62e96e066270f4c2798f68b1375213/arch/eventloop_posix_tcp.c#L814-L819
- The listening-Sockets are closed fast, as `TCP_close` is used 
- The client connections take the `TCP_shutdownConnection` path.

https://github.com/open62541/open62541/blob/5f5214427b62e96e066270f4c2798f68b1375213/arch/eventloop_posix_tcp.c#L503

- `UA_shutdown((UA_FD)connectionId, SD_BOTH)` (win32 `shutdown` - function is used)

(120s) gap

https://github.com/open62541/open62541/blob/5f5214427b62e96e066270f4c2798f68b1375213/arch/eventloop_posix_select.c#L162

 -  A new receive-event is issued and as there is an error, the connection is finally closed

https://github.com/open62541/open62541/blob/5f5214427b62e96e066270f4c2798f68b1375213/arch/eventloop_posix_tcp.c#L503

According to https://docs.microsoft.com/en-us/windows/win32/api/winsock/nf-winsock-shutdown the shutdown function is not used as expected in the library. As `SD_BOTH` is passed to  `shutdown`, any subsequent call to `recv` is disallowed (-1 is returned).

The proposed sequence is instead (taken from the [doc](https://docs.microsoft.com/en-us/windows/win32/api/winsock/nf-winsock-shutdown)):

1. Call shutdown with how=SD_SEND.
2. Call [recv](https://docs.microsoft.com/en-us/windows/desktop/api/winsock/nf-winsock-recv) or [WSARecv](https://docs.microsoft.com/en-us/windows/desktop/api/winsock2/nf-winsock2-wsarecv) until the function completes with success and indicates zero bytes were received. If SOCKET_ERROR is returned, then the graceful disconnect is not possible.
3. Call [closesocket](https://docs.microsoft.com/en-us/windows/desktop/api/winsock/nf-winsock-closesocket).

## Possible solutions

- Keep using `SD_BOTH` and directly call closesocket afterwards.
- Use the flow from the docs. (I don't know how to issue an event when there is no data to receive)

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 5f5214427b62e96e066270f4c2798f68b1375213
 - [x] Other OPC UA SDKs used (client or server): open62541
 - [x] Operating system: Win 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-02-18:
We only call shutdown initially and closesocket in the next iteration of the EventLoop.
That is on purpose because shutdown can be called everywhere, even if the socket is still used somewhere lower in the call stack.

But from your explanation, if we use SD_SEND for shutdown, then everything should work right away.
That means the subsequent call to `select` should signal a read or error condition.

Can you verify?

jpfr, 2022-02-18:
This is correct behavior.
`shutdown` initiates a graceful shutdown.
Windows will wait until either the shutdown handshake has completed or until a timeout occurs.
Linux will complete the handshake in the background and we can close the socket early.

That is the `time_wait` timeout baked into TCP.
You can set it manually in the Windows registry.
https://www.ibm.com/docs/en/records-manager/8.5.0?topic=performance-setting-tcp-timed-wait-delay-tcptimedwaitdelay

One way to get around this timeout is to add a shutdown flag to the internal structure with metadata for each socket.
And to manually test+close all sockets in every iteration.

But, frankly, if there is a 2-minute timeout in Windows for the remote side to answer, it is there for a good reason.
And it won’t impact other connections that are open in parallel.

ccvca, 2022-03-04:
So it's not desired to shorten this period in open62541, similar as in the UDP-Implementation:

https://github.com/open62541/open62541/blob/e974d4afe1f38ade8975e0fcee2c3b59d82a4071/arch/eventloop_posix_udp.c#L418-L434

JanSurft, 2022-03-04:
This is a very interesting finding thanks @ccva! So we really should decide whether we add an explicit shutdown to TCP that forces a closing in the next loop iteration or remove this workaround from the UDP connection manager as well @jpfr.

The following test case fails on windows if we do not explicitly shutdown via delayed callback, because the win32 UDP socket will not receive the `shutdown` signal within the next two iterations after `closeConnection`:

https://github.com/open62541/open62541/blob/e974d4afe1f38ade8975e0fcee2c3b59d82a4071/tests/check_eventloop_udp.c#L326-L334

So if we would iterate the eventloop long enough (> 2min) the shutdown signal would also arrive on the win32 socket?

ccvca, 2022-03-04:
> So if we would iterate the eventloop long enough (> 2min) the shutdown signal would also arrive on the win32 socket?

I've only verified this for TCP, not for UDP. But as TCP is stateful, there might be different results.

But in general I would prefer the solution from the UDP-Part (or a similar one), as the TCP-Socket is shutdown using "SD_BOTH", so trying to read/write anything on the socket should fail. And 120s is quite some time for shutting down gracefully.
https://github.com/open62541/open62541/blob/5f5214427b62e96e066270f4c2798f68b1375213/arch/eventloop_posix_tcp.c#L503

As TCP will send/receive the fin-Messages it might be ok, to wait for some (configurable) amount (e.g. 3s) before closing completely and not directly in the next cycle.

PR #5008 (2022-04-20): [REVIEW] fix(el): win32 shutdown takes 120 s
resolves #4964 

This fix resolves 120 sek timeout for win32 shutdown of tcp connections.

### HOW:

1. calls shutdown "SD_SEND" for connections to be shut down
2. adds a delayed callback that closes a tcp connection after a peek to recv showed no data on the socket
3. if there is peeked data on the socket the delayed callback adds itself again and waits another cycle of the eventloop to process the data that arrived on the socket

### WHY:

As mentioned in the according issue by @ccvca, according to win32 documentation this is the correct way of sh [... cut]
closed 2022-04-20 by jpfr