open62541/open62541 #4977: Certificate Authentication

rexsum, 2022-02-22:
## Description

I have a customer requirement to implement OPC UA authentication using certificates in my client code, but I have been unable to find a solution. I've searched extensively for an answer on how to accomplish this but have so far come up empty. All I've been able to find is a few tidbits of info and comments inside the library's code. The provided examples show how to encrypt the channel using certificates and also how to login using username/password, but not, unless I'm missing something, how to login using a certificate.

Can someone please point me to a code snippet or some working code that demonstrates certificate authentication in a client? Thanks!

## Background Information / Reproduction Steps

With UA Expert I'm able to use my certificate to successfully authenticate and can confirm that it's authenticating with an `X509IdentityToken`. However I can't seem to replicate that behavior in my client code. I've set `securityMode`, `securityPolicyUri` and `applicationUri` which does encrypt the channel, but doesn't authenticate as needed. 

If I allow anonymous on the server side my client code authenticates with an `AnonymousIdentityToken` .

When I disable anonymous on the server I receive the error: `ActivateSession failed with error code BadIdentityTokenInvalid`. 

When I attempt to use an `UA_X509IdentityToken` for the `userIdentityToken` (like I'm guessing that maybe I should from examining the library code) the error I receive is: `ActivateSession failed with error code BadApplicationSignatureInvalid`.

I've search the documentation, other Issues here, and the Google group, but none seem to fully answer this question. Others have ask this or similar questions but all w/o an adequate response. The closest I've seen is a response from Julius that mentions the need to do some manual setup but he doesn't go into detail.

Used CMake options:

```bash
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_SHARED_LIBS=ON -DUA_LOGLEVEL=300 -DUA_MULTITHREADING=100 \
      -DUA_ENABLE_METHODCALLS=ON -DUA_ENABLE_SUBSCRIPTIONS=ON \
      -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_NAMESPACE_ZERO=FULL \
      -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON -DUA_ENABLE_PUBSUB_ETH_UADP=ON \
      -DUA_ENABLE_IMMUTABLE_NODES=ON -DUA_ENABLE_MALLOC_SINGLETON=ON -DUA_ENABLE_PUBSUB_BUFMALLOC=ON \
      -DUA_ENABLE_PUBSUB_ENCRYPTION=ON -DUA_ENABLE_ENCRYPTION_TPM2=ON -DUA_ENABLE_ENCRYPTION=OPENSSL \
      ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master and 1.3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-02-22:
The certificate-based authentication has two parts.

You send your certificate in the X509IdentityToken as defined here:
https://reference.opcfoundation.org/v104/Core/docs/Part4/7.36.5/

You use that same certificate to sign you identity token to prove that you are in posession of the private key for that certificate.
See the userTokenSignature in the ActivateSessionRequest:
https://reference.opcfoundation.org/v104/Core/docs/Part4/5.6.3/

On the server side we "just" check if the signature is valid and then forward the certificate to the AccessControl plugin.
The AccessControl plugin (which can be extended by the user) checks the certificate itself.
The code is pretty straightforward: https://github.com/open62541/open62541/blob/931e0f0c0be04c311ef3c647d580b6eed01f40b3/src/server/ua_services_session.c#L738

The client side isn't that far.
You can set the X509IdentityToken, but it won't be used to generate the userTokenSignature.
Not because its complicated, only because we haven't gotten around to doing it.

rexsum, 2022-02-22:
When you say "the client side isn't that far", do you mean it's not fully implemented yet? Is that why I receive a BadApplicationSignatureInvalid error when I try to use an X509IdentityToken in my client code? 

When will this be ready? Is there anything I can code as a work-around until then? I really need to have certificate authentication working soon for a customer deliverable.

It would also be great to have a working example too so I could easily look at the implementation rather than trying to guess at how this should be coded (which is what I've been attempting to up till now).

kjeldflarup, 2022-10-11:
You state that the server checks that the certificate is valid. How much does that imply and how much should I implement my self.

- That the certificate has not expired?
- That the certificate matches a valid public certificate
- That the certificate belongs to a certain root CA?

How much is trustlistFolder, issuerlistFolder, revocationlistFolder in the context used for?

tobydox, 2024-09-03:
If I understand correctly, certificate authentication support is still missing in the open62541 side? If not, what needs to be done before calling `UA_Client_connect()`, especially regarding the `userIdentityToken`?

tobydox, 2024-09-03:
Nevermind, I got it working with the new `UA_ClientConfig_setAuthenticationCert()` API function introduced in v1.4.

jpfr, 2024-09-03:
Closing this as is has been implemented for 1.4.

closed 2024-09-03 by jpfr