open62541/open62541 #4983: Cannot connect Open62541 server and open62541 client to eachother using a CA-Signed certificate

VictorWiedemann, 2022-02-25:
## Description
I am currently trying to make it so that I can use CA-signed certificates for open62541 server and client.

I am able use the certificates generated by create_self-signed.py and the C code below works fine for that. My issue is when I am trying to use CA-signed certificates, I get vauge error codes and was looking for some insight. The way I am generating the CA-signed certificates can be seen below.

## Background Information / Reproduction Steps
When I use the Server cert (on the client) and the client cert (on the server) as the trusted certs, I get this error: 
`Verifying the certificate failed with error: The certificate is not correctly signed by the trusted CA`

This tipped me off that the trustList needed to be the master_cert, so I tried to use that and then I get.

`Processing the message failed with error BadCertificateRevocationUnknown`

```
openssl genrsa -out master_key.pem 2048
openssl req -new -x509 -days 36500 -key master_key.pem -subj "/CN=OPCUA_master_cert/O=companyName/C=US/ST=BC/L=city/OU=Engineering" -out master_cert.pem

openssl req -newkey rsa:2048 -nodes -keyout temp_server_key.pem -subj "/CN=OPCUA_server_cert/O=companyName/C=US/ST=BC/L=city/OU=Engineering" -out server_cert_request.csr
openssl x509 -req -days 365 -extfile <(printf "subjectAltName=URI:urn:open62541.server.application") -in server_cert_request.csr -CA master_cert.pem -CAkey master_key.pem -CAcreateserial -out temp_server_signed_cert.pem
#todo remove -CAcreateserial and set it with the serial number?

openssl x509 -in temp_server_signed_cert.pem -outform der -out ./server_cert.der 
openssl rsa -inform PEM -in temp_server_key.pem -outform DER -out ./server_key.der 

openssl req -newkey  rsa:2048 -nodes -keyout temp_client_key.pem -subj "/CN=OPCUA_client_cert/O=companyName/C=US/ST=BC/L=city/OU=Engineering"-out client_cert_request.csr
openssl x509 -req -extfile <(printf "subjectAltName=URI:urn:open62541.client.application") -days 365 -in client_cert_request.csr -CA master_cert.pem -CAkey master_key.pem -CAcreateserial -out temp_client_signed_cert.pem

openssl x509 -in temp_client_signed_cert.pem -outform der -out ./client_cert.der 
openssl rsa -inform PEM -in temp_client_key.pem -outform DER -out ./client_key.der ./out/{}/cert/client_key.der".format(folderLocation))
```

And then in the open62541 client and server, I have the following code:

```
#define CERT_FILE_PATH        "/data/cert/server_cert.der"
#define PRIV_KEY_FILE_PATH    "/data/cert/server_key.der"
#define CLIENT_CERT_FILE_PATH "/data/cert/client_cert.der" // I HAVE ALSO TRIED /data/cert/master_cert.der
....
int err = readFileToUABuffer(CERT_FILE_PATH, &certificate) |
              readFileToUABuffer(PRIV_KEY_FILE_PATH, &privateKey) |
              readFileToUABuffer(CLIENT_CERT_FILE_PATH, &trustList[0]);
....
status_code = UA_ServerConfig_setDefaultWithSecurityPolicies(
            config, 4840, &certificate, &privateKey, // Device certificates
            trustList, 1,                            // Trusted certs
            NULL, 0, // Issuer removed since adding it does nothing for us. refer to wiki above
            NULL, 0  // Revoked certs - not implemented in open62541
        );      
```

The client code is:

```
#define CERT_FILE_PATH        "/data/cert/client_cert.der"
#define PRIV_KEY_FILE_PATH    "/data/cert/client_key.der"
#define SERVER_CERT_FILE_PATH "/data/cert/server_cert.der" // I HAVE ALSO TRIED /data/cert/master_cert.der
...
int fileErr = readFileToUABuffer(CERT_FILE_PATH, &certificate) |
                  readFileToUABuffer(PRIV_KEY_FILE_PATH, &privateKey) |
                  readFileToUABuffer(SERVER_CERT_FILE_PATH, &trustList[0]);
....

    int uaError = UA_ClientConfig_setDefaultEncryption(clientConfig, certificate, privateKey, trustList,
                                                       trustListSize, revocationList, revocationListSize);
```

CMAKE linked libraries
target_link_libraries(program
    -lrt
    -lm
    -latomic
    -lmbedx509
    -lmbedcrypto
    -pthread
    ${CMAKE_CURRENT_BINARY_DIR}/lib/json-c/lib/libjson-c.a
)

    -DUA_ENABLE_AMALGAMATION=ON \
    -DUA_ENABLE_METHODCALLS=ON \
    -DUA_ENABLE_SUBSCRIPTIONS=ON \
    -DUA_NAMESPACE_ZERO=FULL \
    -DUA_ENABLE_DA=ON \
    -DUA_ENABLE_DISCOVERY=OFF \
    -DUA_ENABLE_DISCOVERY_MULTICAST=OFF \
    -DUA_ENABLE_MICRO_EMB_DEV_PROFILE=OFF \
    -DUA_ENABLE_ENCRYPTION=ON \
    -DUA_ENABLE_ENCRYPTION_MBEDTLS=ON \
    -DUA_ENABLE_ENCRYPTION_OPENSSL=ON \
    -DCMAKE_MESSAGE_LOG_LEVEL="WARNING" \

## Checklist

Please provide the following information:

 - [x ] open62541 Version (release number or git tag): v1.2.3 
 - [x ] Other OPC UA SDKs used (client or server): no, just open62541 for both sides
 - [x ] Operating system: Ubuntu 20.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jackybek, 2022-03-18:
i used .pem extensions and it worked.
Never tried .der extensions.
Perhaps you can generate the ca certs and server certs using pem extensions.

VictorWiedemann, 2022-03-18:
@jackybek to confirm what you did was make all of the certificates .pem files?: 

In the case of the Server side, you would use server_cert and server_key and then would the "trusted cert" be the client_cert file? I find it odd that every example shows the use of .der files and .pems actually work. see the example code provided below.

My goal here is to use a trust list to only allow certs signed by the CA-signed cert.

Also, did you run the tests of .pem files using a CA-signed cert? All of the examples for Open62541 are self-signed.

```
#define CERT_FILE_PATH        "/data/cert/server_cert.pem"
#define PRIV_KEY_FILE_PATH    "/data/cert/server_key.pem"
#define CLIENT_CERT_FILE_PATH "/data/cert/client_cert.pem"
....
int err = readFileToUABuffer(CERT_FILE_PATH, &certificate) |
              readFileToUABuffer(PRIV_KEY_FILE_PATH, &privateKey) |
              readFileToUABuffer(CLIENT_CERT_FILE_PATH, &trustList[0]);
....
status_code = UA_ServerConfig_setDefaultWithSecurityPolicies(
            config, 4840, &certificate, &privateKey, // Device certificates
            trustList, 1,                            // Trusted certs
            NULL, 0, // Issuer removed since adding it does nothing for us. refer to wiki above
            NULL, 0  // Revoked certs - not implemented in open62541
        );    
```

jackybek, 2022-03-20:
I think the trustlist is not yet implemented in open62541.
Yes, I use OpenSSL to generate CA certs and keys.

VictorWiedemann, 2022-03-21:
When you say that you are using OpenSSL (I am as well) to generated your .pem files. Are these self-signed or CA-signed?

I have implemented the TrustList with self-signed certs and have been able to get that to work without issue. I have tested and found only the certs that are in the trust list are the only ones that are able to connect.

jpfr, 2022-04-06:
@VictorWiedemann Do you use OpenSSL or mbedTLS for the crypto plugins?

> This tipped me off that the trustList needed to be the master_cert, so I tried to use that and then I get.
>> Processing the message failed with error BadCertificateRevocationUnknown

This should work if either
- the certificate itself is in the trust-list
- the issuing CA is in the issuer-list.

If not, then this is a bug.
And we need to tie it down with a unit test.

alex0598, 2023-12-11:
Any updates to this ?

dvnaz10, 2024-05-09:
Do we have any update here if using CA-signed certificate is possible?

otronarp, 2024-08-27:
@dvnaz10 I got it to work (client side accepting a CA signed server certificate) by also specifying a valid certificate revocation list.

otronarp, 2025-02-12:
What I got working was to have:
 - One CA certificate (and its revocation list)
 - One server certificate, signed by the CA certificate

Then on the client side I add the CA certificate to the trustlist and its revocation list to the list of revocation lists , that made the client accept the server certificate.

mcpat-pilz, 2025-02-13:
@aaghashr Please switch to a recent open62541 version. Certificate handling has changed a lot there including a more sophisticated CA detection. It now supports server certificates with the CA flag set (which is often the case for self-signed certificates) but which is not allowed to sign other certificates.

o6-nbeier, 2026-10-20:
Thank you for opening this issue.

CA-signed certificates are supported if you trust the CA and provide a current CRL (a missing CRL yields BadCertificateRevocationUnknown).

Closing for now.

closed 2026-10-20 by o6-nbeier