open62541/open62541 #4984: The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None

jackybek, 2022-02-26:
## Description
I amusing openssl 1.1.1m to generate SSL certs for the OPCServer.
After that i use the same set of cert and privatekey in the OPCClient.

During connecting using UA_Client_connectUsername(), i get the error (refer to the subject header).

I trace the program and found the trigger.  It failed at this function:
verifyClientApplicationURI() : BadSecurityChecksFailed

It seemed that it is looking for securitypolicy#None but when i check my code, only the following security policies are added:

Basic128Rsa15 
basic256 
basic256sha256 
Aes128Sha256RsaOaep 

How do i load the None security policy?

PS: I already verified that the ApplicationURI value in the client instance matches the server instance and also matches the certificate, which i also update open62541.c
#define ApplicationURI urn:virtualskies.com.sg 

## Background Information / Reproduction Steps

     ./myNewTcpClient 192.168.1.119 20004 opc.tcp://192.168.1.115:4840 cert115.pem
     [2022-02-13 09:19:48.037 (UTC+0800)] info/userland      NA_mainOPCUAClient : successfully loaded certificate </etc/ssl/certs/cert115.pem> and privateKey
     [2022-02-13 09:19:48.038 (UTC+0800)] info/userland      NA_mainOPCUAClient : successfully loaded trustList
     [2022-02-13 09:19:48.038 (UTC+0800)] info/userland      AG_mainOPCUAClient : config->securityPolicies : success
     [2022-02-13 09:19:48.038 (UTC+0800)] info/userland      Calling setDefaultEncryption
     [2022-02-13 09:19:48.039 (UTC+0800)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
     [2022-02-13 09:19:48.046 (UTC+0800)] info/securitypolicy        The Basic128Rsa15 security policy with openssl is added.
     [2022-02-13 09:19:48.050 (UTC+0800)] info/securitypolicy        The basic256 security policy with openssl is added.
     [2022-02-13 09:19:48.052 (UTC+0800)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
     [2022-02-13 09:19:48.054 (UTC+0800)] info/securitypolicy        The Aes128Sha256RsaOaep security policy with openssl is added.
     [2022-02-13 09:19:48.055 (UTC+0800)] info/userland      Finished calling setDefaultEncryption
     [2022-02-13 09:19:48.055 (UTC+0800)] info/userland      AG_mainOPCUAClient : successfully setDefaultEncryption
     [2022-02-13 09:19:48.055 (UTC+0800)] info/userland      AG_mainOPCUAClient.c : Just about to call UA_Client_connectUsername() at line 355
     [2022-02-13 09:19:48.055 (UTC+0800)] info/userland      OPCUAServerIP = opc.tcp://192.168.1.115:4840
     open62541.h : UA_Client_connectUsername()
     config->clientDescription.applicationUri = urn:virtualskies.com.sgIAZ+L 23
     config->clientDescription.applicationUri = urn:virtualskies.com.sg 23

     **verifyClientApplicationURI() : BadSecurityChecksFailed** 

     config.clientDescription.applicationUri : urn:virtualskies.com.sg
     [2022-02-13 09:19:48.056 (UTC+0800)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
     verifyClientApplicationURI() : Good
     config.clientDescription.applicationUri : urn:virtualskies.com.sg
     verifyClientApplicationURI() : Good
     config.clientDescription.applicationUri : urn:virtualskies.com.sg
     verifyClientApplicationURI() : Good
     config.clientDescription.applicationUri : urn:virtualskies.com.sg
     verifyClientApplicationURI() : Good
     config.clientDescription.applicationUri : urn:virtualskies.com.sg
     [2022-02-13 09:19:48.057 (UTC+0800)] info/eventloop     Starting the EventLoop
     [2022-02-13 09:19:51.152 (UTC+0800)] warn/network       Connection to opc.tcp://192.168.1.115:4840 failed with error: No route to host
     [2022-02-13 09:19:51.153 (UTC+0800)] info/client        Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadDisconnect
     [2022-02-13 09:19:51.153 (UTC+0800)] info/userland      AG_mainOPCUATcpClient.c : Cannot login securely to OPCUAServer : opc.tcp://192.168.1.115:4840, errcode : BadDisconnect

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

closed 2022-03-18 by jackybek

Samega7Cattac, 2022-03-25:
How did u solve ur issue? I'm having the same problem

jackybek, 2022-03-25:
It was resolved using the latest 1.3 patch