open62541/open62541 #4988: PubSub MQTT not working anymore!

Fischer-Thomas, 2022-02-28:
## Description

Currently the PubSub MQTT implementation is broken. 
After some investigation i found that the current problem lies in UA_Server_run_iterate().
There the pubSubChannel->yield() method is always called with **timeout** set to **0** and inside there we return with an error when **timeout equals 0**. After this the socket will set to an error state.

Also PubSub MQTT with TLS support is not even compiling!
It seems like that there is something wrong with the building/test pipeline on github. 
It doesn't show any problems with the build process! Also the **check_pubsub_connection_mqtt** test has a segmentation fault!

From my point of view the introduction of the eventLoop broke PubSubMQTT.
Also the latest changes to the mqtt source files (mqtt.c, mqtt_adapter.c, ...) broke the functionality of PubSubMQTT with TLS.
When [PR#3637](https://github.com/open62541/open62541/pull/3637) was merged i tested the functionality and everything worked. 
So my guess is that the latest changes broke the functionality.

jackybek, 2022-03-12:
Hi Fischer-Thomas,
I agree that your first observation that  introduction of the eventLoop broke PubSubMQTT.  I can attest to this.

As for your second observation : latest changes to the mqtt source files (mqtt.c, mqtt_adapter.c, ...) broke the functionality of PubSubMQTT with TLS., the reason is because the changes that are supposed to be committed for MQTT-TLS was done in 2 separate PR, and I am not able to create 1 PR that updates 2 different .c files.  

The changes are quite easy to make:
1. In deps/mqtt-c/mqtt-h change the following:

       mqtt_connect(struct mqtt_client *client,
                              size_t will_message_size,
                              const char* user_name,
                              const char* password,
       #ifdef UA_ENABLE_MQTT_TLS_OPENSSL
                              const char* caFilePath,
                              const char* caPath,
                              const char* clientCertPath,
                              const char* clientKeyPath,
                              bool useTLS,
        #endif
                              uint8_t connect_flags,
                              uint16_t keep_alive);

2.  missing useTLS parameter

       #ifdef UA_ENABLE_MQTT_TLS_OPENSSL
           /* try to pack the message */
           MQTT_CLIENT_TRY_PACK(rv, msg, client,
                         mqtt_pack_connection_request(
                                                      client->mq.curr, client->mq.curr_sz,
                                                      client_id, will_topic, will_message,
                                                      will_message_size,user_name, password,
                                                      caFilePath, caPath, clientCertPath, clientKeyPath, useTLS,
                                                      connect_flags, keep_alive
                                                      ),
                         1
                         );

3. change the following:

    enum MQTTConnectFlags {
    MQTT_CONNECT_RESERVED = 1u,
    MQTT_CONNECT_CLEAN_SESSION = 2u,
    MQTT_CONNECT_WILL_FLAG = 4u,
    MQTT_CONNECT_WILL_QOS_0 = (0u & 0x03) << 3,
    MQTT_CONNECT_WILL_QOS_1 = (1u & 0x03) << 3,
    MQTT_CONNECT_WILL_QOS_2 = (2u & 0x03) << 3,
    MQTT_CONNECT_WILL_RETAIN = 32u,
    MQTT_CONNECT_PASSWORD = 64u,
    MQTT_CONNECT_USER_NAME = 128u
    #ifdef UA_ENABLE_MQTT_TLS_OPENSSL
    ,
    MQTT_CONNECT_CAFILEPATH = 256u,
    MQTT_CONNECT_CAPATH = 512u,
    MQTT_CONNECT_CLIENTCERTPATH = 1024u,
    MQTT_CONNECT_CLIENTKEYPATH = 2048u,
    MQTT_CONNECT_USETLS = 4096u
    #endif

4. in mqtt_pack_connection_request() function, cast the following to (uint8_t):
      MQTT_CONNECT_CAFILEPATH
      MQTT_CONNECT_CAPATH
      MQTT_CONNECT_CLIENTCERTPATH
      MQTT_CONNECT_CLIENTKEYPATH
      MQTT_CONNECT_USETLS

5. correct the error from useTSL to useTLS

6. change strcpy(strTLS, "1", 1) and strcpy(strTLS, "0", 1)  to
            if (useTLS) /* useTLS = true */
        {
                strTLS[0] ='1';
                strTLS[1] ='\0';
        }
        else
        {
                strTLS[0] ='0';
                strTLS[1] ='\0';
        }

Fischer-Thomas, 2022-03-14:
Hello @jackybek,

thanks for your response.
For PR that introduces changes to multiple files maybe check this stackoverflow thread: https://stackoverflow.com/questions/67436427/how-to-create-a-pull-request-covering-multiple-files-in-different-directories

I tried your proposed changes and was able to build again.
However the functionality is still broken. I´m not able to send any MQTT with TLS packages at all.

As i mentioned in my post i know that i once was able to use PubSubMQTT with TLS support. I went back to check when exactly the functionality went broken. 
I went back to this [commit](https://github.com/open62541/open62541/commit/ee9d90d7ae3b03478625a6a16443d136bbfe36f8) that introduced changes to mqtt-c. (Before this commit i verified that MQTT TLS was still working)
I first had to fix all the compiler errors but without success the functionality is broken from this change. 
Can you maybe explain what you intended to change with this commit?

For the purpose of maintainability it would be better to add the [mqtt-c repository](https://github.com/LiamBindle/MQTT-C) as submodule, what do you think @jpfr?

jackybek, 2022-03-14:
The purpose of this commit is to handle the following use case:
1. pubSub Mqtt via port 8883 : MQTT over TLS/ SSL
2. pubSub Mqtt via port 8884 : MQTT over TLS/ SSL with client side certificate
currently in open62541, i tested 1883 and 1884.  no issues.
but for 8883, the code crashed.  for 8884 it returns MQTT_ERROR_SOCKET_ERROR

refer to issue#5005.
in previous version v1.2.2 both 8883 and 8884 works, albeit 8884 return MQTT_ERROR_SOCKET_ERROR after 10 mins.
in v1.3-rc2, both crashed.

jackybek, 2022-03-20:
I think I found some clues as to why using 8883 crashed and using 8884 caused mqtt_socket_error.
It is related to the mosquitto.conf configuration.
In open62541, there is no difference between using 8883 or 8884, since both requires TLS. In open62541, it packs the parameters and send to mqtt broker. Depending on whether you set username, password, and/ or pass in client certificate/key, the open62541 just send it to mqtt.
Now at mqtt end, depending on the mosquitto.conf settings you set for 8883 and 8884, you need to ensure there is no conflict.
At the moment, I manage to solve the 1st issue that using 8883 caused core dump, but as in 8884, I still get mqtt_connect_error.
I traced the runtime and found that the status M

ivan-ushakov, 2022-03-20:
@Fischer-Thomas 
Could you please tell me what previous version has working PubSub MQTT?

Fischer-Thomas, 2022-03-21:
> @Fischer-Thomas
Could you please tell me what previous version has working PubSub MQTT?

This is the latest working version for me: https://github.com/open62541/open62541/commit/44909b8cc2e0d62da8b308808c9028e859bb64ea

jackybek, 2022-03-21:
1.2.2 pubsub mqtt without TLS was working fine.

jpfr, 2022-04-04:
MQTT support is now restored on 1.3.
It was further added to the standard CI build to prevent future breakage.
The TLS support for MQTT compiles, but needs further testing and a more complete CI scaffolding.

Closing this.

closed 2022-04-04 by jpfr