open62541/open62541 #4991: Question: PubSub Publishing on demand

rexsum, 2022-03-03:
## Description

I need to create a PubSub Publisher that does not send out data at some predetermined interval (i.e. the publishingInterval) like all the examples show, but rather only on demand when my application determines that it is necessary. Is this possible? How can this be accomplished?

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Muddasir-Shakil, 2022-03-07:
Hi @rexsum 
What you are explaining sounds like Delta Frame DataSetMessages which is controlled KeyFrameCount, ( [Part 14](https://reference.opcfoundation.org/v104/Core/docs/Part14/6.2.3/) --> 6.2.3.3 KeyFrameCount. 
What I could gather from the specs is that if KeyFrameCount > 1, then send only changed values in the DataSetMessage. When the maxpublishinginterval == (publishing interval x KeyFrameCount) then send KeyFrame Message contains all the values. If there is no changed values at each publishing interval then send no Delta Frame DataSetMEssage but send KeepAlive Message. There you also have to configure KeepAlive Time.

Alternatively,  Event DataSetMessage can be sent which contains fields from an OPC UA Event, in the DataSet to be published. You can create your Event Types and in the Event Fields you can add fields so when ever event is triggered, the published event message contains the event fields (also your data fields). 

With open 62541 stack I tested the first approach. I could publish the delta frame datasetmessage, but I was not able to recieve then on subscriber side.  You may able to work it out by some tinkering on subscriber side.

rexsum, 2022-03-07:
Thank you @Muddasir-Shakil for taking the time to try and answer my question.

I often find the OPC UA spec very difficult to understand and to translate what it's saying into practical application. I read the section you pointed out but it's still not clear to me exactly how that would apply in my situation. It seems to indicate that I will still need some sort of publishing interval, but at the very least perhaps I won't need to flood the network with duplicate messages at each interval by using data frames. I will do some experimenting to see how I might be able to accomplish something close to what I need. Thanks again.

rexsum, 2022-03-07:
Hi @Muddasir-Shakil,

Would you mind sharing more details about the experiment you conducted?

I tried here on my end by turning on the delta frames CMAKE flag (-DUA_ENABLE_PUBSUB_DELTAFRAMES=ON) and then setting the keyFrameCount on my DataSetWriter, however I saw no effect at all. My publishingInterval is set to 1 second (slow enough so I could see what was going on) and set the keyFrameCount=5. The variable node that I'm publishing is set and not changing, so based upon what you said I would have expected to only see a full DataSet Message every 5th interval or 5 seconds and empty delta frames in between. Looking at Wireshark however I still see a complete packet published every second.

Not sure where to go from here, any further tips would be appreciated, thx.

Muddasir-Shakil, 2022-03-21:
Hi @rexsum 
Sorry for replying little late. I forgot to mention that standard also says that if PublishedDataSet has only one field then deltaframe should not be generated. It is also checked in the code here, I think this might be the reason you are getting KeyFrame Messages.
https://github.com/open62541/open62541/blob/5f5214427b62e96e066270f4c2798f68b1375213/src/pubsub/ua_pubsub_writer.c#L1342

DocDriven, 2024-08-06:
@Muddasir-Shakil I am looking for the same kind of functionality, but also ran into the same problem. I built my library with the -DUA_ENABLE_PUBSUB_DELTAFRAMES option and created a PublishedDataSet like this:

    UA_PublishedDataSetConfig publishedDataSetConfig;
    memset(&publishedDataSetConfig, 0, sizeof(UA_PublishedDataSetConfig));
    publishedDataSetConfig.publishedDataSetType = UA_PUBSUB_DATASET_PUBLISHEDITEMS;
    publishedDataSetConfig.name = UA_STRING(pdsName);
    UA_Server_addPublishedDataSet(server, &publishedDataSetConfig, publishedDataSetIdent);

Then I added fields to said dataset with this code snippet (note that I copied this from a function encapsulating this functionality, and thus I called this twice with different values for name and NodeId of the field):

    UA_DataSetFieldConfig dataSetFieldConfig;
    memset(&dataSetFieldConfig, 0, sizeof(UA_DataSetFieldConfig));
    dataSetFieldConfig.dataSetFieldType = UA_PUBSUB_DATASETFIELD_VARIABLE;
    dataSetFieldConfig.field.variable.fieldNameAlias = UA_STRING(fieldName);
    dataSetFieldConfig.field.variable.promotedField = UA_FALSE;
    dataSetFieldConfig.field.variable.publishParameters.publishedVariable = dataNodeIdent;
    dataSetFieldConfig.field.variable.publishParameters.attributeId = UA_ATTRIBUTEID_VALUE;
    UA_Server_addDataSetField(server, publishedDataSetIdent, &dataSetFieldConfig, NULL);

The outcome is exactly the same as described above, meaning that the keyFrameCount does not have an impact, it just sends at the configured intervals. Can you provide confirmation that everything has been done correctly?

o6-nbeier, 2026-10-20:
Thank you for the question.

Defer enabling the writer and call the publish callback on demand (1.3/1.4), or use a custom WriterGroup state machine without timer plus the public publish trigger (1.5/master).

Closing this for now.

closed 2026-10-20 by o6-nbeier