open62541/open62541 #5010: Adding callback pointer to existing MethodNode

paddor, 2022-03-14:
## Description

## Background Information / Reproduction Steps

I need to set the callback pointer to existing MethodNodes after importing nodesets from XML. Alternatively, is there a way to set a generic callback pointer that is called on ANY UA method call?

mlang-de, 2022-03-15:
To attach a callback to an existing method you can use UA_Server_setMethodNodeCallback(). 
https://github.com/open62541/open62541/blob/f73404023ddadfe6eeb9938f653cd6a62c8d72a8/include/open62541/server.h#L1129
I guess there exist no feature to add an generic callback pointer.

jpfr, 2022-03-15:
There is no "official way" to add a generic callback pointer.
You can
- The method indicated by @mlang-de to set individual callbacks.
- Use a (recursive) browse to get the identifiers of new nodes and their nodeclass.
- Or go directly to the NodeStore API to iterate over all nodes and find out which is a method missing a callback pointer.

Just reopen this issue if new questions come up.

paddor, 2022-03-15:
Thanks. `UA_Server_setMethodNodeCallback()` sounds like a solution. I'm now parsing the XML to find the `UAMethod` nodes, translate the namespace IDs to whatever they are in the server, and use this function to set the callbacks before starting the server (rc=good), but unfortunately I still don't get the callbacks when I call the method from UA Expert.

The method in XML:

```xml
  <UAMethod NodeId="ns=1;s=Foobar" BrowseName="1:foobar" ParentNodeId="ns=1;g=fc658c51-781b-4c13-8d4f-8c45f4054389">
    <DisplayName>foobar</DisplayName>
    <References>
      <Reference ReferenceType="HasComponent" IsForward="false">ns=1;g=fc658c51-781b-4c13-8d4f-8c45f4054389</Reference>
      <Reference ReferenceType="HasProperty">ns=1;s=Foobar.InputArguments</Reference>
      <Reference ReferenceType="HasProperty">ns=1;s=Foobar.OutputArguments</Reference>
    </References>
  </UAMethod>
  <UAVariable NodeId="ns=1;s=Foobar.InputArguments" BrowseName="InputArguments" ParentNodeId="ns=1;s=Foobar" DataType="Argument" ValueRank="1"                  ↪ArrayDimensions="0">
    <DisplayName>InputArguments</DisplayName>
    <References>
      <Reference ReferenceType="HasModellingRule">i=78</Reference>
      <Reference ReferenceType="HasProperty" IsForward="false">ns=1;s=Foobar</Reference>
      <Reference ReferenceType="HasTypeDefinition">PropertyType</Reference>
    </References>
    <Value>
      <uax:ListOfExtensionObject/>
    </Value>
  </UAVariable>
  <UAVariable NodeId="ns=1;s=Foobar.OutputArguments" BrowseName="OutputArguments" ParentNodeId="ns=1;s=Foobar" DataType="Argument" ValueRank="1"                ↪ArrayDimensions="0">
    <DisplayName>OutputArguments</DisplayName>
    <References>
      <Reference ReferenceType="HasModellingRule">i=78</Reference>
      <Reference ReferenceType="HasProperty" IsForward="false">ns=1;s=Foobar</Reference>
      <Reference ReferenceType="HasTypeDefinition">PropertyType</Reference>
    </References>
    <Value>
      <uax:ListOfExtensionObject/>
    </Value>
  </UAVariable>
```

paddor, 2022-03-15:
> Just reopen this issue if new questions come up.

I can't seem to be able to reopen this issue. There's no reopen button.