open62541/open62541 #5014: Async operations without multithreading

paddor, 2022-03-15:
## Description

## Background Information / Reproduction Steps

I wanna use async operations without multithreading. Our app is a single-threaded, multi-process, event-driven message passing app. No worker threads. But we still need to support async UA method calls.

Are there any plans to support async functions like `UA_Server_getAsyncOperationNonBlocking()` without the need to compile with `-pthread` and have mutexes all over the place? Maybe with `UA_MULTITHREADING=50`?

jpfr, 2022-03-18:
That is a great idea!
We can do that during the switch to the EventLoop based architecture.
Note that multithreaded only enables mutexes.
We don’t start Threads internally (with the exception of discovery).
So you can use the server_run_iterate method in a way that switches between server and worker in the same thread.

paddor, 2024-02-19:
Are there any updates here? Not sure but `struct UA_EventLoop` exists. Does that mean the architecture is now EventLoop based?

paddor, 2025-01-29:
Any updates on compiling without `-pthread`?