open62541/open62541 #5017: UA_findDataType() fails for data type UA_XVType

vdeters, 2022-03-16:
## Description
UA_FindDataType() returns null if passed typeId has identifierType = 26 (numeric), which is UA_TYPES_XVTYPE
## Background Information / Reproduction Steps
The typeId of 26 was returned from a call to UA_Client_readDataTypeAttribute() and then passed to UA_findDataType() to validate the returned data type. But it seems UA_findDataType() expects UA_NS0ID_XVTYPE (ie 12080) instead of UA_TYPES_XVTYPE. This is not an issue for the standard types. Not sure if this is actually a client bug, could it be a server bug in the typeId returned in UA_Client_readDataTypeAttribute()? Or is this call sequence not valid for non-standard types?

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.2-rc2
 - [ ] Other OPC UA SDKs used (client or server): open62541 client, Aspen InfoPlus.21 server
 - [ ] Operating system: linux x64
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) available on request
 - [ ] Wireshark network dump available on request
 - [ ] Self-contained code example available on request
 - [ ] Critical issue yes

vdeters, 2022-03-16:
Make options not showing correctly, they are:
CMAKE_FLAGS="-DUA_ENABLE_ENCRYPTION=ON -DUA_LOGLEVEL=700 -DUA_ENABLE_AMALGAMATION=ON -DMBEDTLS_INCLUDE_DIRS=$MBED/include -DMBEDTLS_LIBRARY=$MBED/lib/libmbedtls.a -DMBEDX509_LIBRARY=$MBED/lib/libmbedx509.a -DMBEDCRYPTO_LIBRARY=$MBED/lib/libmbedcrypto.a -DBUILD_SHARED_LIBS=OFF"

jpfr, 2022-03-18:
UA_TYPES_XVTYPE is an index for the UA_TYPES table.
The find method expects a NodeId from ns0.
So yes, you have arrived at the correct conclusion.

closed 2022-03-18 by jpfr

vdeters, 2022-03-18:
Thanks for your response Julius, but which one of my conclusions is the correct one:

- a server bug in the typeId returned in UA_Client_readDataTypeAttribute()? or

- this call sequence not valid for non-standard types?

Thanks,
Vince

vdeters, 2022-03-21:
If we assume the value of 26 returned from UA_Client_readDataTypeAttribute() is correct, that would correspond to type UA_NS0ID_NUMBER. What should we expect when passing this to UA_findDataType()? Is there no corresponding entry in the UA_TYPES table?