open62541/open62541 #5018: UA_Server_addTimedCallback() with date=UA_DateTime_now() never runs callback

paddor, 2022-03-17:
## Description

## Background Information / Reproduction Steps

`UA_Server_addTimedCallback()` seems broken. When I pass `UA_DateTime_now()`  as the `date` argument, it never runs the callback. It only runs the callback when I pass an integer like `5000`.

```c
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>

static volatile UA_Boolean running = true;
static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}

static void run_timed_callback(UA_Server *server, void *context)
{
  printf("Hello from timed callback\n");
}

int main(void) {
    UA_Server *server;
    UA_StatusCode rc;
    UA_DateTime time;

    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    time = UA_DateTime_now();
    /* time = 5000; */
    rc = UA_Server_addTimedCallback(server, run_timed_callback, NULL, time, NULL);
    if (rc != UA_STATUSCODE_GOOD) return EXIT_FAILURE;

    rc = UA_Server_run(server, &running);

    UA_Server_delete(server);
    return rc == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

Used CMake options:

```
-DUA_NAMESPACE_ZERO=FULL -DUA_MULTITHREADING=100 -DCMAKE_BUILD_TYPE=Debug -DUA_DEBUG=1 -O0 -DCMAKE_C_FLAGS_DEBUG='-fPIC -g' -DCMAKE_INTERPROCEDURAL_OPTIMIZATION=OFF
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 22883e54027366190a5c38abdfa28c9268fbbc1f
 - [X] Operating system: Ubuntu 20.04
 - [X] Self-contained code example attached

mister-potter, 2022-03-20:
@paddor 
`UA_DateTime_now()` returns a `long int` value which looks something like `132922552486583550`
Callback is set for this time. And if this time is in past, then the callback is executed in server's next iteration.

Since this is a huge value, you don't see the callback happening immediately.
```
UA_DateTime UA_DateTime_now(void) {
    struct timeval tv;
    gettimeofday(&tv, NULL);
    return (tv.tv_sec * UA_DATETIME_SEC) + (tv.tv_usec * UA_DATETIME_USEC) + UA_DATETIME_UNIX_EPOCH;
}
```

paddor, 2022-03-20:
> @paddor `UA_DateTime_now()` returns a `long int` value which looks something like `132922552486583550` Callback is set for this time. And if this time is in past, then the callback is executed in server's next iteration.

Then with the value of `UA_DateTime_now()` the callback should be executed in the server's next iteration, right?

> Since this is a huge value, you don't see the callback happening immediately.

The huge value means now. Or am I wrong?

For example, how do I register a callback that should be executed in 5 seconds?

mister-potter, 2022-03-21:
I suggest you to take a look at the example below. Callback is executed based on the value set for `timeInMilliSeconds`.
```
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>
#include <time.h>

static volatile UA_Boolean running = true;
static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}

static void run_timed_callback(UA_Server *server, void *context)
{
  printf("Hello from timed callback\n");
  // To know the callback interval
  struct timespec ts;
  timespec_get(&ts, TIME_UTC);
  char buff[100];
  strftime(buff, sizeof buff, "%D %T", gmtime(&ts.tv_sec));
  printf("Callback time: %s.%09ld UTC\n", buff, ts.tv_nsec);
}

int main(void) {
    UA_Server *server;
    UA_StatusCode rc;
    UA_DateTime timeInMilliSeconds;

    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    timeInMilliSeconds = 5000;

    rc = UA_Server_addRepeatedCallback(server, (UA_ServerCallback) run_timed_callback, NULL, timeInMilliSeconds, NULL);
    if (rc != UA_STATUSCODE_GOOD) return EXIT_FAILURE;

    rc = UA_Server_run(server, &running);

    UA_Server_delete(server);
    return rc == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

paddor, 2022-03-21:
That example doesn't use `UA_Server_addTimedCallback()`.

If you're suggesting that the `date` parameter is simply misnamed and actually a duration, that doesn't work either.

Fischer-Thomas, 2022-03-22:
@paddor this worked for me
 
```
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>

static volatile UA_Boolean running = true;
static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}

static void 
run_timed_callback(UA_Server *server, void *context) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Hello from timed callback");
}

static void
run_repeated_callback(UA_Server *server, void *context) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Hello from repeated callback");
}

int main(void) {
    UA_Server *server;
    UA_StatusCode rc;
    UA_DateTime time;
    UA_Double interval;

    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    UA_DateTime fiveSecondsInMs = 5000;

    time = UA_DateTime_nowMonotonic() + (fiveSecondsInMs * UA_DATETIME_MSEC);
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Hello from main");
    rc = UA_Server_addTimedCallback(server, (UA_ServerCallback)run_timed_callback, NULL, time, NULL);
    if (rc != UA_STATUSCODE_GOOD) return EXIT_FAILURE;
    interval = 5000;
    rc = UA_Server_addRepeatedCallback(server, (UA_ServerCallback)run_repeated_callback, NULL, interval, NULL);

    rc = UA_Server_run(server, &running);

    UA_Server_delete(server);
    return rc == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

See logs:
> [2022-03-22 10:41:25.689 (UTC+0100)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
> [2022-03-22 10:41:25.690 (UTC+0100)] info/server        AccessControl: Anonymous login is enabled
> [2022-03-22 10:41:25.690 (UTC+0100)] info/server        AccessControl: x509 certificate user authentication is enabled
> [2022-03-22 10:41:25.690 (UTC+0100)] warn/server        Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
> [2022-03-22 10:41:25.690 (UTC+0100)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
> **[2022-03-22 10:41:25.690 (UTC+0100)] info/userland      Hello from main**
> [2022-03-22 10:41:27.343 (UTC+0100)] info/network       TCP network layer listening on opc.tcp://md3bgm1c:4840/
> **[2022-03-22 10:41:30.692 (UTC+0100)] info/userland      Hello from timed callback**
> [2022-03-22 10:41:30.692 (UTC+0100)] info/userland      Hello from repeated callback
> [2022-03-22 10:41:35.692 (UTC+0100)] info/userland      Hello from repeated callback
>

Fischer-Thomas, 2022-03-22:
@paddor this worked for me
 
```
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>

static volatile UA_Boolean running = true;
static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}

static void 
run_timed_callback(UA_Server *server, void *context) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Hello from timed callback");
}

static void
run_repeated_callback(UA_Server *server, void *context) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Hello from repeated callback");
}

int main(void) {
    UA_Server *server;
    UA_StatusCode rc;
    UA_DateTime time;
    UA_Double interval;

    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    UA_DateTime fiveSecondsInMs = 5000;

    time = UA_DateTime_nowMonotonic() + (fiveSecondsInMs * UA_DATETIME_MSEC);
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Hello from main");
    rc = UA_Server_addTimedCallback(server, (UA_ServerCallback)run_timed_callback, NULL, time, NULL);
    if (rc != UA_STATUSCODE_GOOD) return EXIT_FAILURE;
    interval = 5000;
    rc = UA_Server_addRepeatedCallback(server, (UA_ServerCallback)run_repeated_callback, NULL, interval, NULL);

    rc = UA_Server_run(server, &running);

    UA_Server_delete(server);
    return rc == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

See logs:
> [2022-03-22 10:41:25.689 (UTC+0100)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
> [2022-03-22 10:41:25.690 (UTC+0100)] info/server        AccessControl: Anonymous login is enabled
> [2022-03-22 10:41:25.690 (UTC+0100)] info/server        AccessControl: x509 certificate user authentication is enabled
> [2022-03-22 10:41:25.690 (UTC+0100)] warn/server        Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
> [2022-03-22 10:41:25.690 (UTC+0100)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
> **[2022-03-22 10:41:25.690 (UTC+0100)] info/userland      Hello from main**
> [2022-03-22 10:41:27.343 (UTC+0100)] info/network       TCP network layer listening on opc.tcp://localhost:4840/
> **[2022-03-22 10:41:30.692 (UTC+0100)] info/userland      Hello from timed callback**
> [2022-03-22 10:41:30.692 (UTC+0100)] info/userland      Hello from repeated callback
> [2022-03-22 10:41:35.692 (UTC+0100)] info/userland      Hello from repeated callback
>

paddor, 2022-03-22:
@Fischer-Thomas Thanks! Using `UA_DateTime_nowMonotonic()` seems to work fine.

closed 2022-03-22 by paddor