open62541/open62541 #5019: Building error with multithreading

AVerutin, 2022-03-18:
## Building error with multithreading

When I building library with enable multithreading I have an errors like this:

```
E:/Projects/CPP/Libs/open62541/build/open62541.c:3448:5: error: unknown type name 'pthread_t'
 3448 |     pthread_t mdnsThread;
      |     ^~~~~~~~~
```

Used CMake options:

```bash
cmake -DUA_MULTITHREADING=110
```

## Checklist

 - [* ] open62541 Version (release number or git tag): 1.3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [* ] Operating system: Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [* ] Critical issue

jpfr, 2022-03-18:
You use Visual Studio or MinGW?

AVerutin, 2022-03-18:
> You use Visual Studio or MinGW?

I am using MinGW