open62541/open62541 #5033: Snapcraft container with open62 server

OsamaOAyoub, 2022-03-23:
## Description
Hello,

I have open62 installed and a running server with custom nodesets and everything is fine, I needed to create a snap using snapcraft for the working server but I am facing issues with running the server with the snap.

My YAML file contains the following apps and parts:

```
apps:
  opcserver:
    command: home/osama/OPCServer/src/server/parts/opcserver/src/OPCServer
    plugs:
    - network-bind
```

```
parts:

  opcserver:
    plugin: cmake
    source: ./  
    stage-packages:
      - libopen62541-1
 
  open62541:
    plugin: cmake
    cmake-parameters:
       - -DBUILD_SHARED_LIBS=ON
       - -DCMAKE_INSTALL_PREFIX=/home/osama/install
       - -DCMAKE_BUILD_TYPE=RelWithDebInfo
       - -DUA_NAMESPACE_ZERO=Full

        
    source-type: git
    source: https://github.com/open62541/open62541.git
    build-packages:
      - build-essential 
      - gcc
      - pkg-config  
      - cmake
      - python3
      - cmake-curses-gui
      - libmbedtls-dev
      - libsubunit-dev
```

With this setup, snapcraft can create the snap but running it leads to the following error:

```
Symbol `UA_TYPES' has different size in shared object, consider re-linking
[2022-03-23 14:06:09.655 (UTC+0100)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2022-03-23 14:06:09.655 (UTC+0100)] info/server        AccessControl: Anonymous login is enabled
[2022-03-23 14:06:09.655 (UTC+0100)] warn/server        Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2022-03-23 14:06:09.655 (UTC+0100)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
Segmentation fault
```

Is there a special procedure for the compilation of the library and the server so the error of different sizes would be fixed?

Thanks in advanced :)

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Always cloning the latest from the repository
 - [ ] Operating system: WSL

ccvca, 2022-03-24:
> Symbol `UA_TYPES' has different size in shared object, consider re-linking

This looks like a good hint, why there is a SegFault later. Is there any reason, why you prefer a shared library. Static linking should solve this. If you require shared library, make sure to use the same linker settings and the same version (git Version may change) for your application and the open62541 library.

OsamaOAyoub, 2022-03-24:
Thanks for your answer, I actually tried that before, but doing so will give me the following warning:
```
This part is missing libraries that cannot be satisfied with any available stage-packages known to snapcraft:
- usr/local/lib/libopen62541.so.1
These dependencies can be satisfied via additional parts or content sharing. Consider validating configured filesets if this dependency was built.
```

Running the snap gives me the following error this time:

```
error while loading shared libraries: libopen62541.so.1: cannot open shared object file: No such file or directory
```

ccvca, 2022-03-24:
You need to link your application (`OPCServer`) static.

In case of a shared library you should use the same codebase of open62541.

OsamaOAyoub, 2022-03-24:
I specified the source of the open62541 part as the local one which I normally use but I am still facing the same warning:

```
This part is missing libraries that cannot be satisfied with any available stage-packages known to snapcraft:
- usr/local/lib/libopen62541.so.1
These dependencies can be satisfied via additional parts or content sharing. Consider validating configured filesets if this dependency was built.
```

And when running the snap:

```
[2022-03-24 12:34:03.444 (UTC+0100)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2022-03-24 12:34:03.444 (UTC+0100)] info/server        AccessControl: Anonymous login is enabled
[2022-03-24 12:34:03.444 (UTC+0100)] info/server        AccessControl: x509 certificate user authentication is enabled
[2022-03-24 12:34:03.444 (UTC+0100)] warn/server        Username/Password Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2022-03-24 12:34:03.444 (UTC+0100)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
Segmentation fault
```

The following error message disappeared after changing - -DCMAKE_INSTALL_PREFIX to be equal /usr
```
Symbol `UA_TYPES' has different size in shared object, consider re-linking
```

ccvca, 2022-03-24:
You may need to run valgrind inside snap, to get a hint, when the Segmentation fault occurs. But as there is still a message, that the .so-File is missing it looks like the library is still linked shared instead of a static.

I do also not know any specifics about snap-Containers, but at least for docker-Containers this is a working project with open62541:
https://github.com/umati/Sample-Server/blob/964eb9024bbc290872006b222642be8eeb43c27f/Dockerfile

closed 2022-08-12 by OsamaOAyoub