open62541/open62541 #5040: Error compiling with UA_LOGLEVEL=200

Samega7Cattac, 2022-03-29:
## Description
Error compiling with `UA_LOGLEVEL=200`
```
FAILED: CMakeFiles/open62541-object.dir/src/server/ua_services_attribute.c.obj
C:\msys64\mingw64\bin\gcc.exe -DUA_ARCHITECTURE_WIN32 -DUA_DYNAMIC_LINKING_EXPORT -IC:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/src -IC:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/arch -IC:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/arch/win32 -IC:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/include -IC:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/plugins/include -IC:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/deps -IC:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/src/pubsub -IC:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/build-MINGW64/src_generated -IC:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/build-MINGW64 -march=x86-64 -mtune=generic -O2 -pipe -std=c99 -Wall -Wextra -Wpedantic -Werror -Wno-static-in-inline -Wno-overlength-strings -Wno-unused-parameter -Wc++-compat -Wformat -Wformat-security -Wformat-nonliteral -Wmissing-prototypes -Wstrict-prototypes -Wredundant-decls -Wuninitialized -Winit-self -Wcast-qual -Wstrict-overflow -Wnested-externs -Wmultichar -Wundef -fno-strict-aliasing -fexceptions -fPIC -g -MD -MT CMakeFiles/open62541-object.dir/src/server/ua_services_attribute.c.obj -MF CMakeFiles\open62541-object.dir\src\server\ua_services_attribute.c.obj.d -o CMakeFiles/open62541-object.dir/src/server/ua_services_attribute.c.obj -c C:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/src/server/ua_services_attribute.c
In file included from C:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/src/server/ua_server_async.h:20,
                 from C:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/src/server/ua_server_internal.h:25,
                 from C:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/src/server/ua_services_attribute.c:23:
C:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/src/server/ua_services_attribute.c: In function 'copyAttributeIntoNode':
C:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/src/server/ua_session.h:140:9: error: '%.*s' directive argument is null [-Werror=format-overflow=]
  140 |         UA_LOG_##LEVEL(LOGGER, UA_LOGCATEGORY_SESSION,                  \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  141 |                        "SecureChannel %" PRIu32 " | Session \"%.*s\" | " MSG "%.0s", \
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  142 |                        chanId, nameLen, nameStr, __VA_ARGS__);          \
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/src/ua_util_internal.h:35:5: note: in definition of macro 'UA_LOG_NODEID_INTERNAL'
   35 |     LOG;                                             \
      |     ^~~
C:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/src/server/ua_services_attribute.c:1527:5: note: in expansion of macro 'UA_LOG_NODEID_DEBUG'
 1527 |     UA_LOG_NODEID_DEBUG(&node->head.nodeId,
      |     ^~~~~~~~~~~~~~~~~~~
C:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/src/server/ua_session.h:154:5: note: in expansion of macro 'UA_MACRO_EXPAND'
  154 |     UA_MACRO_EXPAND(UA_LOG_SESSION_INTERNAL(LOGGER, DEBUG, SESSION, __VA_ARGS__, ""))
      |     ^~~~~~~~~~~~~~~
C:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/src/server/ua_session.h:154:21: note: in expansion of macro 'UA_LOG_SESSION_INTERNAL'
  154 |     UA_MACRO_EXPAND(UA_LOG_SESSION_INTERNAL(LOGGER, DEBUG, SESSION, __VA_ARGS__, ""))
      |                     ^~~~~~~~~~~~~~~~~~~~~~~
C:/Users/oncontrol/Desktop/AUR/MINGW-packages/mingw-w64-open62541/src/open62541/src/server/ua_services_attribute.c:1528:25: note: in expansion of macro 'UA_LOG_DEBUG_SESSION'
 1528 |                         UA_LOG_DEBUG_SESSION(&server->config.logger, session,
      |                         ^~~~~~~~~~~~~~~~~~~~
```
This is using MSYS2 but it should not be a problem since without defining `UA_LOGLEVEL` it compiles correctly

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -GNinja -DCMAKE_INSTALL=${MINGW_PREFIX} -DCMAKE_BUILD_TYPE=Debug -DBUILD_SHARED_LIBS=ON _DUA_NAMESPACE_ZERO=FULL _DUA_ENABLE_AMALGAMATION=OFF -DUA_ENABLE_ENCRYPTION_OPENSSL=ON -DUA_ENABLE_ENCRYPTION=OPENSSL -DUA_LOGLEVEL=200 ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.3_rc2_ef2_r61_g46afb7d24
 - [x] Other OPC UA SDKs used (client or server): OpenSSL 1.1.1n
 - [x] Operating system: Windows (MSYS2)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Samega7Cattac, 2022-03-31:
This is exclusive to MSYS2

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier