open62541/open62541 #5043: The node ID identifier.string. Data exists in Chinese and the client cannot subscribe

nihao-git, 2022-03-30:
## Description

The node ID identifier.string. Data exists in Chinese and the client cannot subscribe  

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-03-30:
The problem is not clear.
Please give more details.

nihao-git, 2022-03-31:
If the node ID exists in Chinese, the client cannot complete the subscription and returns that the current node ID does not exist.  The following code        
        UA_NodeId targetNodeId = UA_NODEID_STRING(2, "Data1.test_data.你");
	printf("NodeId:%s \n", targetNodeId.identifier.string.data);
	addMonitoredItemToVariable(client, &targetNodeId);
static void addMonitoredItemToVariable(UA_Client* client, UA_NodeId* target)
{
	printf("NodeId:%s length:%d \n", target->identifier.string.data, target->identifier.string.length);
	/* Create a subscription */
	UA_CreateSubscriptionRequest request = UA_CreateSubscriptionRequest_default();
	UA_CreateSubscriptionResponse response = UA_Client_Subscriptions_create(client, request, NULL, NULL, NULL);

	UA_UInt32 subId = response.subscriptionId;
	if (response.responseHeader.serviceResult == UA_STATUSCODE_GOOD)
	{
		UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_CLIENT, "Create subscription succeeded, id %u\n", subId);
	}

	UA_MonitoredItemCreateRequest monRequest = UA_MonitoredItemCreateRequest_default(*target);
	monRequest.requestedParameters.samplingInterval = 10.0;
	UA_MonitoredItemCreateResult monResponse = UA_Client_MonitoredItems_createDataChange(
		client, response.subscriptionId,
		UA_TIMESTAMPSTORETURN_BOTH,
		monRequest, (void*)target,
		handler_DataChanged, NULL);
	if (monResponse.statusCode == UA_STATUSCODE_GOOD)
	{
		UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_CLIENT, "Monitoring '%s', id %u\n", target->identifier.string.data, monResponse.monitoredItemId);
	}
	printf("statusCode:%d", monResponse.statusCode);
}

nihao-git, 2022-04-03:
Thank you, problem solved, but tell the compiler to encode in UTF-8

closed 2022-04-03 by nihao-git