open62541/open62541 #5044: Connect with user and pass return BadInternalError

Samega7Cattac, 2022-03-31:
## Description
I'm simply making a client to connect to a server using user and pass.
But I get `No suitable UserTokenPolicy found for the possible endpoints` and `BadInternalError`

### Code example

```
UA_Client* m_opc_client;
m_opc_client = UA_Client_new();
m_opc_status = UA_ClientConfig_setDefault(UA_Client_getConfig(m_opc_client));
m_opc_status = UA_Client_connectUsername(m_opc_client, url.c_str(), user.c_str(), pass.c_str());
```
Double checked the arguments are everything seems correct.
There was no need to set extra things on any of the examples

### Output:
```
[2022-03-31 11:48:53.080 (UTC+0100)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2022-03-31 11:48:53.081 (UTC+0100)] info/eventloop     Starting the EventLoop
[2022-03-31 11:48:53.081 (UTC+0100)] trace/eventloop    Iterate the EventLoop
[2022-03-31 11:48:53.081 (UTC+0100)] trace/client       Client connect iterate
[2022-03-31 11:48:53.081 (UTC+0100)] trace/eventloop    Iterate the EventLoop
[2022-03-31 11:48:53.081 (UTC+0100)] trace/client       Client connect iterate
[2022-03-31 11:48:53.081 (UTC+0100)] debug/network      Sent HEL message
[2022-03-31 11:48:53.081 (UTC+0100)] debug/client       Client Status: ChannelState: HELSent, SessionState: Closed, ConnectStatus: Good
[2022-03-31 11:48:53.081 (UTC+0100)] trace/eventloop    Iterate the EventLoop
[2022-03-31 11:48:53.081 (UTC+0100)] trace/client       Client connect iterate
[2022-03-31 11:48:53.083 (UTC+0100)] debug/network      Received ACK message
[2022-03-31 11:48:53.083 (UTC+0100)] debug/client       Client Status: ChannelState: AckReceived, SessionState: Closed, ConnectStatus: Good
[2022-03-31 11:48:53.083 (UTC+0100)] trace/eventloop    Iterate the EventLoop
[2022-03-31 11:48:53.083 (UTC+0100)] trace/client       Client connect iterate
[2022-03-31 11:48:53.083 (UTC+0100)] debug/channel      Connection 4 | SecureChannel 0 | Generating new local nonce
[2022-03-31 11:48:53.083 (UTC+0100)] debug/channel      Connection 4 | SecureChannel 0 | Requesting to open a SecureChannel
[2022-03-31 11:48:53.083 (UTC+0100)] debug/channel      Requesting to open a SecureChannel
[2022-03-31 11:48:53.083 (UTC+0100)] debug/channel      OPN message sent
[2022-03-31 11:48:53.083 (UTC+0100)] debug/client       Client Status: ChannelState: OPNSent, SessionState: Closed, ConnectStatus: Good
[2022-03-31 11:48:53.083 (UTC+0100)] trace/eventloop    Iterate the EventLoop
[2022-03-31 11:48:53.083 (UTC+0100)] trace/client       Client connect iterate
[2022-03-31 11:48:53.084 (UTC+0100)] trace/channel      Connection 4 | SecureChannel 21 | Verifying chunk signature
[2022-03-31 11:48:53.084 (UTC+0100)] trace/channel      Connection 4 | SecureChannel 21 | Generating new local keys
[2022-03-31 11:48:53.084 (UTC+0100)] info/channel       Connection 4 | SecureChannel 21 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2022-03-31 11:48:53.084 (UTC+0100)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2022-03-31 11:48:53.084 (UTC+0100)] trace/eventloop    Iterate the EventLoop
[2022-03-31 11:48:53.084 (UTC+0100)] trace/client       Client connect iterate
[2022-03-31 11:48:53.084 (UTC+0100)] debug/channel      Connection 4 | SecureChannel 21 | Sending request with RequestId 2 of type GetEndpointsRequest
[2022-03-31 11:48:53.084 (UTC+0100)] trace/channel      Connection 4 | SecureChannel 21 | Send from a symmetric message buffer of length 8196 a message of header+payload length of 122
[2022-03-31 11:48:53.084 (UTC+0100)] trace/channel      Connection 4 | SecureChannel 21 | Send from a symmetric message buffer of length 8196 a message of length 122
[2022-03-31 11:48:53.084 (UTC+0100)] trace/eventloop    Iterate the EventLoop
[2022-03-31 11:48:53.084 (UTC+0100)] trace/client       Client connect iterate
[2022-03-31 11:48:53.086 (UTC+0100)] trace/channel      Connection 4 | SecureChannel 21 | Generating new remote keys
[2022-03-31 11:48:53.086 (UTC+0100)] trace/eventloop    Iterate the EventLoop
[2022-03-31 11:48:53.086 (UTC+0100)] trace/client       Client connect iterate
[2022-03-31 11:48:53.086 (UTC+0100)] trace/eventloop    Iterate the EventLoop
[2022-03-31 11:48:53.086 (UTC+0100)] trace/client       Client connect iterate
[2022-03-31 11:48:53.086 (UTC+0100)] info/client        Rejecting endpoint 0: security policy not available
[2022-03-31 11:48:53.086 (UTC+0100)] info/client        Rejecting endpoint 1: security policy not available
[2022-03-31 11:48:53.086 (UTC+0100)] info/client        Rejecting endpoint 2: security policy not available
[2022-03-31 11:48:53.086 (UTC+0100)] info/client        Rejecting endpoint 3: security policy not available
[2022-03-31 11:48:53.086 (UTC+0100)] info/client        Rejecting endpoint 4: security policy not available
[2022-03-31 11:48:53.086 (UTC+0100)] info/client        Rejecting endpoint 5: security policy not available
[2022-03-31 11:48:53.086 (UTC+0100)] info/client        Rejecting endpoint 6: security policy not available
[2022-03-31 11:48:53.086 (UTC+0100)] info/client        Rejecting endpoint 7: security policy not available
[2022-03-31 11:48:53.086 (UTC+0100)] info/client        Rejecting UserTokenPolicy 0 in endpoint 8: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256' not available
[2022-03-31 11:48:53.086 (UTC+0100)] info/client        Rejecting UserTokenPolicy 1 in endpoint 8: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15' not available
[2022-03-31 11:48:53.086 (UTC+0100)] info/client        Rejecting endpoint 9: security policy not available
[2022-03-31 11:48:53.086 (UTC+0100)] info/client        Rejecting endpoint 10: security policy not available
[2022-03-31 11:48:53.086 (UTC+0100)] error/client       No suitable UserTokenPolicy found for the possible endpoints
[2022-03-31 11:48:53.086 (UTC+0100)] trace/channel      Connection 4 | SecureChannel 21 | Send from a symmetric message buffer of length 8196 a message of header+payload length of 57
[2022-03-31 11:48:53.086 (UTC+0100)] trace/channel      Connection 4 | SecureChannel 21 | Send from a symmetric message buffer of length 8196 a message of length 57
[2022-03-31 11:48:53.086 (UTC+0100)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError
[2022-03-31 11:48:53.086 (UTC+0100)] info/eventloop     Stopping the EventLoop
[2022-03-31 11:48:53.086 (UTC+0100)] debug/eventloop    All EventSources are stopped
[2022-03-31 11:48:53.086 (UTC+0100)] info/eventloop     The EventLoop has stopped
```

### Wireshark capture:
[opcua_userpass.zip](https://github.com/open62541/open62541/files/8388671/opcua_userpass.zip)
[image]

### Used CMake options:

```bash
cmake -DCMAKE_BUILD_TYPE=Debug -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION=OFF -DUA_ENABLE_ENCRYPTION_OPENSSL=ON -DUA_ENABLE_ENCRYPTION=OPENSSL -DUA_LOGLEVEL=100 -DBUILD_SHARED_LIBS=ON ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): db19f990d16e3ecac3934d67200d86d88bf49031
 - [x] Operating system: Arch Linux
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Wireshark network dump attached
 - [x] Self-contained code example attached

mustafacamurlu, 2022-04-07:
Hi There,

I have found an awkward setup in your server endpoint configuration.
In your client config you have set security policy none as accepted policy.

But when you check individual policies in the user token policy in the 8th response they are:
- Basic256
- Basic128Rsa15

which is perfectly reported by the client:

"
[2022-03-31 11:48:53.086 (UTC+0100)] info/client        Rejecting UserTokenPolicy 0 in endpoint 8: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256' not available
[2022-03-31 11:48:53.086 (UTC+0100)] info/client        Rejecting UserTokenPolicy 1 in endpoint 8: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15' not available
"

[image]

You need to check / correct your server endpoint configuration.

Samega7Cattac, 2022-04-07:
For more about that issue https://github.com/open62541/open62541/issues/5036

SplitGemini, 2022-05-13:
Same BadInternalError with user connect.

paddor, 2023-08-30:
I had this issue yesterday. I solved it by generating a certificate + key (using the [script](https://github.com/open62541/open62541/blob/master/tools/certs/create_self-signed.py) from this repo) and then preparing the `UA_ClientConfig` using `UA_ClientConfig_setDefaultEncryption()` instead of `UA_ClientConfig_setDefault()`. This will populate the `UA_ClientConfig.securityPolicies` with more policies so that one of them will match during the connect.

```
UA_ClientConfig_setDefaultEncryption(&clientConfig, localCertificate, privateKey,
                                     NULL, 0, // trustList, trustListSize
                                     NULL, 0); // revocationList, revocationListSize
clientConfig.clientDescription.applicationUri = UA_STRING_ALLOC("my_special_uri");
```

In my case, I had to make sure the applicationURI used to generate the certificate matches the `applicationUri` in the particular endpoint (one from the result of `UA_Client_getEndpoints()`) used during the connect. You'll get some warnings about `ApplicationURI does not match the URI specified in the certificate` from `UA_ClientConfig_setDefaultEncryption()` which can be ignored if you set the correct `UA_ClientConfig.clientDescription.applicationUri` afterwards (but before the connect).

xydan83, 2024-04-12:
I also have this problem.
But it helped to use signature or signature and encryption. It won't work without this.

This is log with only user/pass:
[image]

This is log with Sign&Encrypt:
[image]

closed 2025-08-25 by Samega7Cattac