open62541/open62541 #5047: double free / segmentation fault on failed ActivateSession requests

neojaw, 2022-04-01:
## Description

I'm running a client against a Siemens S7-1200 and I ran into sporadic program crashes, sometimes segmentation faults, sometimes what appeared to be a double free. The client does nothing special, it has one subscription and reads a variable in a loop. The crashes occured more often when the network connection was bad (or deliberately distorted). My guess is that it happens when several `ActivateSession` requests get cancelled at the same time.

I've tested with the v1.2.4 release. To test with master, I'd have to change my code since the API is different. Let me know if you need it.

## Background Information

Here is some log output (piped through my own logging mechanism, sorry):

```
[] DEBUG: [secure channel] Connection 4 | SecureChannel 1292966654 | Sending request with RequestId 33 of type ActivateSessionRequest
[] DEBUG: [client] Client Status: ChannelState: Open, SessionState: ActivateRequested, ConnectStatus: Good
[] DEBUG: [secure channel] Connection 4 | SecureChannel 1292966654 | Generating new remote keys
[] DEBUG: [secure channel] Connection 4 | SecureChannel 1292966654 | Verifying chunk signature
[] DEBUG: [secure channel] Connection 4 | SecureChannel 1292966654 | Calculated padding size to be 12
[] INFO: [client] Received a ServiceFault response
[] INFO: [client] The ServiceResult has the StatusCode BadSessionIdInvalid
[] ERROR: [client] ActivateSession failed with error code BadSessionIdInvalid
[] ERROR: [client] ActivateSession failed with error code BadSessionClosed
[] ERROR: [client] ActivateSession failed with error code BadSessionClosed
[] DEBUG: [secure channel] Connection 5 | SecureChannel 1292966524 | Sending request with RequestId 149 of type CreateSessionRequest
[] INFO: [client] Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[] ERROR: [client] Session cannot be activated. Create a new Session.
Segmentation fault (core dumped)
```

As you can see, there are several `ActivateSession` requests in transit. (I've seen this while working with the S7 before. Sometimes it just seems to not respond to `ActivateSession` requests, they pile up and when I shut down the client, I get a dozen failure notices.)
What I understand from reading the code is the following: When one of them fails (here with `BadSessionIdInvalid`), it callbacks to `responseActivateSession` where the session is closed, which, in the wake of cleaning up, cancels the other `ActivateSession` requests (with `BadSessionClosed`). And now they do the same thing again.
The immediate problem is that we now have two callbacks iterating through `client->asyncServiceCalls` and freeing stuff and the way that `LIST_REMOVE` works, it leaves dangling pointers behind, ready for the other ones to walk into.

Here is some gdb output to support this:

```
(gdb) bt
#0  0x00007f76a8ba095c in UA_Client_AsyncService_removeAll (client=0x7f7644022b60, statusCode=2149974016) at /XXX/open62541/src/client/ua_client.c:509
#1  0x00007f76a8ba5457 in closeSession (client=0x7f7644022b60) at /XXX/open62541/src/client/ua_client_connect.c:1178
#2  0x00007f76a8ba345d in responseActivateSession (client=0x7f7644022b60, userdata=0x0, requestId=143, response=0x7f765cff8220)
    at /XXX/open62541/src/client/ua_client_connect.c:491
#3  0x00007f76a8ba08e3 in UA_Client_AsyncService_cancel (client=0x7f7644022b60, ac=0x7f76440c9dc0, statusCode=2149974016) at /XXX/open62541/src/client/ua_client.c:500
#4  0x00007f76a8ba0975 in UA_Client_AsyncService_removeAll (client=0x7f7644022b60, statusCode=2149974016) at /XXX/open62541/src/client/ua_client.c:510
#5  0x00007f76a8ba5457 in closeSession (client=0x7f7644022b60) at /XXX/open62541/src/client/ua_client_connect.c:1178
#6  0x00007f76a8ba345d in responseActivateSession (client=0x7f7644022b60, userdata=0x0, requestId=148, response=0x7f765cff8440)
    at /XXX/open62541/src/client/ua_client_connect.c:491
#7  0x00007f76a8ba007e in processAsyncResponse (client=0x7f7644022b60, requestId=148, responseTypeId=0x7f765cff85c0, responseMessage=0x7f76440c7eb8, offset=0x7f765cff85a8)
    at /XXX/open62541/src/client/ua_client.c:294
[...]
(gdb) frame 0
#0  0x00007f76a8ba095c in UA_Client_AsyncService_removeAll (client=0x7f7644022b60, statusCode=2149974016) at /home/jaw/Desktop/pilot-core/vnx-opc-ua/open62541/src/client/ua_client.c:509
509	        LIST_REMOVE(ac, pointers);
(gdb) print ac->pointers
$11 = {le_next = 0x7f76440a4e00, le_prev = 0x0}
(gdb) print *ac->pointers.le_next
$12 = {pointers = {le_next = 0x656361006563, le_prev = 0x55}, requestId = 1141414240, callback = 0x0, responseType = 0x0, userdata = 0x7f76440d1d10, start = 0, timeout = 0, responsedata = 0xd9c324b217}
```

Sometimes the trap doesn't snap immediately, but then we free the same memory twice:

```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007f403dddd859 in __GI_abort () at abort.c:79
#2  0x00007f403de4829e in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7f403df72298 "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007f403de5032c in malloc_printerr (str=str@entry=0x7f403df745d0 "free(): double free detected in tcache 2") at malloc.c:5347
#4  0x00007f403de51f9d in _int_free (av=0x7f3fcc000020, p=0x7f3fcc049920, have_lock=0) at malloc.c:4201
#5  0x00007f403b6d57eb in UA_Array_delete (p=0x7f3fcc049930, size=1, type=0x7f403b786288 <UA_TYPES+11880>) at /home/jaw/Desktop/pilot-core/vnx-opc-ua/open62541/src/ua_types.c:1389
#6  0x00007f403b6d52e0 in clearStructure (p=0x7f3fcc04a850, type=0x7f403b7862d0 <UA_TYPES+11952>) at /XXX/open62541/src/ua_types.c:1235
#7  0x00007f403b6d559f in UA_delete (p=0x7f3fcc04a850, type=0x7f403b7862d0 <UA_TYPES+11952>) at /XXX/open62541/src/ua_types.c:1324
#8  0x00007f403b714826 in UA_PublishRequest_delete (p=0x7f3fcc04a850) at /home/jaw/Desktop/pilot-core/build/vnx-opc-ua/open62541/src_generated/open62541/types_generated_handling.h:5192
#9  0x00007f403b718cde in processPublishResponseAsync (client=0x7f3fcc001fb0, userdata=0x7f3fcc04a850, requestId=254, response=0x7f3fdb7fca60)
    at /XXX/open62541/src/client/ua_client_subscriptions.c:1204
#10 0x00007f403b70bf10 in UA_Client_AsyncService_cancel (client=0x7f3fcc001fb0, ac=0x7f3fcc0452c0, statusCode=2149974016) at /XXX/open62541/src/client/ua_client.c:498
#11 0x00007f403b70bfa2 in UA_Client_AsyncService_removeAll (client=0x7f3fcc001fb0, statusCode=2149974016) at /XXX/open62541/src/client/ua_client.c:508
#12 0x00007f403b710a50 in closeSession (client=0x7f3fcc001fb0) at /XXX/open62541/src/client/ua_client_connect.c:1178
#13 0x00007f403b70ea35 in responseActivateSession (client=0x7f3fcc001fb0, userdata=0x0, requestId=261, response=0x7f3fdb7fcc50)
    at /XXX/open62541/src/client/ua_client_connect.c:491
#14 0x00007f403b70bf10 in UA_Client_AsyncService_cancel (client=0x7f3fcc001fb0, ac=0x7f3fcc027ae0, statusCode=2149974016) at /XXX/open62541/src/client/ua_client.c:498
#15 0x00007f403b70bfa2 in UA_Client_AsyncService_removeAll (client=0x7f3fcc001fb0, statusCode=2149974016) at /XXX/open62541/src/client/ua_client.c:508
#16 0x00007f403b710a50 in closeSession (client=0x7f3fcc001fb0) at /XXX/open62541/src/client/ua_client_connect.c:1178
[...]
```

When I patch `LIST_REMOVE` to set `(elm)->field.le_next` to `NULL`, I still get crashes, but the stack trace changes and now I get a null pointer dereference from `client->channel.securityPolicy` in `createSessionAsync` (which comes right after `closeSession`). So I reckon my guess about the problem is correct, I just don't know how to fix it properly.

Used CMake options: default, except for
`UA_ENABLE_ENCRYPTION_OPENSSL=ON BUILD_SHARED_LIBS=ON UA_LOGLEVEL=100`

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.2.4
 - [x] Other OPC UA SDKs used (client or server): Siemens S7-1200 OPCUA server
 - [x] Operating system: Ubuntu 20.04
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue: I'd say yes, because it causes program crashes

jpfr, 2022-04-04:
Thanks for the report. We will fix that and make a patch release for 1.2 as well.
Just a quick clarification. You say:

> My guess is that it happens when several ActivateSession requests get cancelled at the same time.

Do you use the client SDK in multiple threads in parallel?
Or does "at the same time" mean "during the processing of a single server response"?

neojaw, 2022-04-05:
I'm using one client object in a single thread, by calling `UA_Client_run_iterate` regularly. When the connection breaks, I delete the client object and create a new one (by the docs _If the connection fails unrecoverably (`state->connectStatus` is set to an error), the client is no longer usable. Create a new client if required._). I don't know if there are supposed to be several open `ActivateSession` requests in that scenario, but that's what I observe.

Don't know if it's important, but the call to `UA_Client_run_iterate` may sometimes be delayed when I'm doing a variable read and the call gets stuck in the network and eventually times out. (I'm doing it synchronously, because asynchronous does not give a failure response at the moment, see #3268.)

o6-nbeier, 2026-10-20:
Thank you for the report.

The ownership of the outstanding async-call list was fixed on the EventLoop client and backported to 1.3 (see #7238 and #5114), so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier