open62541/open62541 #5048: Changing object nodeid

bittojoseph, 2022-04-01:
Hallo,
I would like to know, if I can change the Nodeid of an object dynamically, if it is preassigned by the server. 
I know that I can assign an id with addObjectNode method if the object is dynamically created but if it has subtypes, then the ids are automatically assigned by the server.

bearsh, 2022-10-31:
maybe you can use the `generateChildNodeId()` callback and change the id when a child node is being added.

http://www.open62541.org/doc/1.3/nodestore.html?highlight=generatechildnodeid#global-node-lifecycle

o6-nbeier, 2026-10-20:
Thank you for the question.

The NodeIds of instantiated children can only be chosen at creation time via `generateChildNodeId`, and changing an existing node's NodeId is not supported (the write returns BadWriteNotSupported).

Closing for now.

closed 2026-10-20 by o6-nbeier