open62541/open62541 #5049: Windows 10 VS 2022 UA_ENABLE_ENCRYPTION Build Error

ffreality, 2022-04-01:
## Description
I am working on an Unreal Engine 4 OPC UA plugin. I successfully created anonymous connections and read & write supports. But I need encryption, too. So, if I enable UA_ENABLE_ENCRYPTION with OpenSSL I got build error from open62541.sln

## Background Information / Reproduction Steps
I installed Win32OpenSSL-3_0_2 and Win64OpenSSL-3_0_2.
CMake can find LIB_EAY_RELEASE/DEBUG and SSL_EAY_RELEASE/DEBUG in these paths.

LIB_EAY_DEBUG = C:/Program Files (x86)/OpenSSL-Win32/lib/VC/libcrypto32MDd.lib
LIB_EAY_RELEASE = C:/Program Files (x86)/OpenSSL-Win32/lib/VC/libcrypto32MD.lib
SSL_EAY_DEBUG = C:/Program Files (x86)/OpenSSL-Win32/lib/VC/libssl32MDd.lib
SSL_EAY_RELEASE = C:/Program Files (x86)/OpenSSL-Win32/lib/VC/libssl32MD.lib

If I disable encryption it works perfectly.

Used CMake options:

Commandline options:
-DUA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL="1" -DUA_ENABLE_DISCOVERY:BOOL="1" -DUA_ENABLE_ENCRYPTION:STRING="OPENSSL" -DCMAKE_BUILD_TYPE:STRING="RelWithDebInfo" -DUA_ENABLE_AMALGAMATION:BOOL="1" -DUA_ENABLE_DIAGNOSTICS:BOOL="1" -DUA_ENABLE_HISTORIZING:BOOL="1" -DUA_ENABLE_PUBSUB_DELTAFRAMES:BOOL="1" -DUA_NAMESPACE_ZERO:STRING="FULL" -DUA_ENABLE_JSON_ENCODING:BOOL="1" -DUA_ENABLE_ENCRYPTION_TPM2:STRING="OFF" -DUA_ENABLE_PUBSUB:BOOL="1" -DCMAKE_CONFIGURATION_TYPES:STRING="Debug;Release;MinSizeRel;RelWithDebInfo" -DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL="1" 

Cache file:
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=1
UA_ENABLE_DISCOVERY:BOOL=1
UA_ENABLE_ENCRYPTION:STRING=OPENSSL
CMAKE_BUILD_TYPE:STRING=RelWithDebInfo
UA_ENABLE_AMALGAMATION:BOOL=1
UA_ENABLE_DIAGNOSTICS:BOOL=1
UA_ENABLE_HISTORIZING:BOOL=1
UA_ENABLE_PUBSUB_DELTAFRAMES:BOOL=1
UA_NAMESPACE_ZERO:STRING=FULL
UA_ENABLE_JSON_ENCODING:BOOL=1
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF
UA_ENABLE_PUBSUB:BOOL=1
CMAKE_CONFIGURATION_TYPES:STRING=Debug;Release;MinSizeRel;RelWithDebInfo
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=1

## Visual Studio Build Log
Build started...
1>------ Build started: Project: ZERO_CHECK, Configuration: RelWithDebInfo Win32 ------
1>Checking Build System
2>------ Build started: Project: open62541-generator-types, Configuration: RelWithDebInfo Win32 ------
3>------ Build started: Project: open62541-generator-transport, Configuration: RelWithDebInfo Win32 ------
4>------ Build started: Project: open62541-generator-statuscode, Configuration: RelWithDebInfo Win32 ------
5>------ Skipped Build: Project: open62541-generator-ids-ns0, Configuration: RelWithDebInfo Win32 ------
5>Project not selected to build for this solution configuration 
6>------ Skipped Build: Project: cpplint, Configuration: RelWithDebInfo Win32 ------
6>Project not selected to build for this solution configuration 
2>Generating src_generated/open62541/types_generated.c, src_generated/open62541/types_generated.h, src_generated/open62541/types_generated_handling.h
2>Building Custom Rule D:/C++/open62541/CMakeLists.txt
3>Generating src_generated/open62541/transport_generated.c, src_generated/open62541/transport_generated.h, src_generated/open62541/transport_generated_handling.h
7>------ Build started: Project: open62541-generator-namespace, Configuration: RelWithDebInfo Win32 ------
7>Generating src_generated/open62541/namespace0_generated.c, src_generated/open62541/namespace0_generated.h
4>Generating src_generated/open62541/nodeids.h
3>Building Custom Rule D:/C++/open62541/CMakeLists.txt
7>INFO:__main__:Preprocessing D:/C++/open62541/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
4>Generating src_generated/open62541/statuscodes.h, src_generated/open62541/statuscodes.c
4>Building Custom Rule D:/C++/open62541/CMakeLists.txt
7>INFO:__main__:Generating Code for Backend: open62541
7>INFO:__main__:NodeSet generation code successfully printed
7>Building Custom Rule D:/C++/open62541/CMakeLists.txt
8>------ Build started: Project: open62541-amalgamation-source, Configuration: RelWithDebInfo Win32 ------
9>------ Build started: Project: open62541-amalgamation-header, Configuration: RelWithDebInfo Win32 ------
8>Generating open62541.c
9>Generating src_generated/open62541/nodeids.h
9>Generating src_generated/open62541/statuscodes.h, src_generated/open62541/statuscodes.c
9>Building Custom Rule D:/C++/open62541/CMakeLists.txt
9>Generating open62541.h
9>Starting amalgamating file D:/C++/open62541/build/Win/open62541.h
9>Integrating file 'D:/C++/open62541/build/Win/src_generated/open62541/config.h' ... done.
9>Integrating file 'D:/C++/open62541/deps/ms_stdint.h' ... done.
9>Integrating file 'D:/C++/open62541/include/open62541/architecture_definitions.h' ... done.
[... 189 lines cut ...]
14>------ Skipped Build: Project: doc (open62541\doc\doc), Configuration: RelWithDebInfo Win32 ------
14>Project not selected to build for this solution configuration 
15>------ Skipped Build: Project: clang-tidy, Configuration: RelWithDebInfo Win32 ------
15>Project not selected to build for this solution configuration 
16>------ Build started: Project: ALL_BUILD, Configuration: RelWithDebInfo Win32 ------
17>------ Skipped Build: Project: doc_pdf, Configuration: RelWithDebInfo Win32 ------
17>Project not selected to build for this solution configuration 
16>Building Custom Rule D:/C++/open62541/CMakeLists.txt
18>------ Skipped Build: Project: PACKAGE, Configuration: RelWithDebInfo Win32 ------
18>Project not selected to build for this solution configuration 
19>------ Skipped Build: Project: INSTALL, Configuration: RelWithDebInfo Win32 ------
19>Project not selected to build for this solution configuration 
========== Build: 9 succeeded, 2 failed, 0 up-to-date, 8 skipped ==========

## Checklist

Please provide the following information:

 - [x ] open62541 Version (release number or git tag): Github Master Branch
 - [ ] Other OPC UA SDKs used (client or server):  None
 - [x ] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-04-04:
Hey,

please paste the content of the error lines.
```
11>D:\C++\open62541\build\Win\open62541.c(199649,52): error C2220: the following warning is treated as an error
11>D:\C++\open62541\build\Win\open62541.c(199649,52): warning C4090: 'function': different 'const' qualifiers
11>D:\C++\open62541\build\Win\open62541.c(199710,38): warning C4090: '=': different 'const' qualifiers
11>D:\C++\open62541\build\Win\open62541.c(199871,1): warning C4090: 'initializing': different 'const' qualifiers
```

These are probably function definitions.
Which function is it?

ffreality, 2022-04-08:
CMake Configurations
[image]
[image]
[image]

CMake Results
```
Selecting Windows SDK version 10.0.20348.0 to target Windows 10.0.19044.
open62541 Version: v1.3.0-rc2-ef2-153-gdb19f990
Architectures included in amalgamation: win32
The selected architecture is: win32
Could NOT find Sphinx (missing: SPHINX_EXECUTABLE) 
Could NOT find LATEX (missing: LATEX_COMPILER) 
Configuring done
Generating done
```

Error Code: 11>D:\C++\open62541\build\Win\open62541.c(199649,52): warning C4090: 'function': different 'const' qualifiers
Specific Line: get_pkey_rsa(privateKey),      /* private key      */
```
while (cipherOffset < data->length) {
        decryptedBytes = RSA_private_decrypt (keySize,
                           data->data + cipherOffset, /* what to decrypt  */
                           buf,                       /* where to decrypt */
                           get_pkey_rsa(privateKey),      /* private key      */
                           padding
                           );
        if (decryptedBytes < 0) {
            return UA_STATUSCODE_BADSECURITYCHECKSFAILED;
        }
        memcpy(data->data + outOffset, buf, (size_t) decryptedBytes);
        cipherOffset += (size_t) keySize;
        outOffset += (size_t) decryptedBytes;
    }
```

Error Code: 11>D:\C++\open62541\build\Win\open62541.c(199710,38): warning C4090: '=': different 'const' qualifiers
```
    /* get the encrypted block size */
    rsa = get_pkey_rsa (evpPublicKey);
```

Error Code: 11>D:\C++\open62541\build\Win\open62541.c(199871,1): warning C4090: 'initializing': different 'const' qualifiers
Specific Line:     RSA * rsa = get_pkey_rsa (evpKey);
```
/* return the key bytes */
UA_StatusCode
UA_Openssl_RSA_Public_GetKeyLength (X509 *     publicKeyX509,
                                    UA_Int32 * keyLen) {
    EVP_PKEY * evpKey = X509_get_pubkey (publicKeyX509);
    if (evpKey == NULL) {
        return  UA_STATUSCODE_BADINTERNALERROR;
    }
    RSA * rsa = get_pkey_rsa (evpKey);
    *keyLen = RSA_size(rsa);
    EVP_PKEY_free (evpKey);

    return UA_STATUSCODE_GOOD;
}
```

When I use generated open62541.h and .c with these errors in Unreal Engine 4, it gives me this error.
But as I say before, if I disable encryption from CMAKE, open62541 is being compiled without error and UE4 can use it perfectly both in editor, runtime and packaged project.

ffreality, 2022-04-11:
Is there any suggestion for it ?

ccvca, 2022-04-11:
Looks like, the tested OpenSSL Version is different to your used one. What OpenSSL Version do you use?
Do you require OpenSSL? Otherwise, you might try LibreSSL (Version 3.4.3 is working in my Project) or MBedTLS for the encryption.

ffreality, 2022-04-11:
https://slproweb.com/products/Win32OpenSSL.html
I used these ones.
Win64 OpenSSL v3.0.2
Win32 OpenSSL v3.0.2

I will try LibreSSL and share results, thank you !

MohammedShatit, 2022-04-27:
@ffreality, I was facing the same problem when building the library, but I was able to successfully build using MBedTLS 2.27.0 and open62541 master. If you got this to build without errors, were you successful in getting it to work in unreal engine?

ffreality, 2022-04-27:
I didn't try it yet because I need create another plugin but I think we need to add tls to engine, too. If we add tls's library location to build.cs and include them, we can solve the problem.

Other than that (without encryption) it works perfectly. I created history access, async readers writer and other stuffs.

ffreality, 2022-05-23:
I tried OpenSSL and MbedTLS. I passed x509h errors with copying ssl's include folder and its content to right beside open62541.c and h as you can see in screenshot.

Right now it gives me these errors.

`1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_md_info_from_type referenced in function policyContext_newContext_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_md_init referenced in function policyContext_newContext_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_md_free referenced in function clear_sp_aes128sha256rsaoaep
[image]

1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_md_setup referenced in function policyContext_newContext_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_md_get_size referenced in function mbedtls_generateKey
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_md_hmac_starts referenced in function mbedtls_generateKey
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_md_hmac_update referenced in function mbedtls_generateKey
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_md_hmac_finish referenced in function mbedtls_generateKey
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_rsa_set_padding referenced in function asym_decrypt_sp_basic128rsa15
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_rsa_rsaes_oaep_encrypt referenced in function mbedtls_encrypt_rsaOaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_rsa_rsaes_oaep_decrypt referenced in function mbedtls_decrypt_rsaOaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_pk_init referenced in function policyContext_newContext_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_pk_free referenced in function clear_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_pk_get_bitlen referenced in function asym_getLocalEncryptionKeyLength_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_pk_verify referenced in function asym_verify_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_pk_sign referenced in function asym_sign_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_pk_decrypt referenced in function asym_decrypt_sp_basic128rsa15
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_pk_encrypt referenced in function asym_encrypt_sp_basic128rsa15
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_pk_parse_key referenced in function UA_mbedTLS_LoadPrivateKey
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_x509_crl_parse referenced in function UA_CertificateVerification_Trustlist
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_x509_crl_init referenced in function UA_CertificateVerification_Trustlist
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_x509_crl_free referenced in function certificateVerification_clear
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_x509_crt_parse referenced in function UA_CertificateVerification_Trustlist
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_x509_crt_verify_info referenced in function certificateVerification_verify
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_x509_crt_verify_with_profile referenced in function certificateVerification_verify
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_x509_crt_init referenced in function UA_CertificateVerification_Trustlist
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_x509_crt_free referenced in function UA_mbedTLS_LoadLocalCertificate
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_aes_setkey_enc referenced in function sym_encrypt_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_aes_setkey_dec referenced in function sym_decrypt_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_aes_crypt_cbc referenced in function sym_decrypt_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_ctr_drbg_init referenced in function policyContext_newContext_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_ctr_drbg_seed referenced in function policyContext_newContext_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_ctr_drbg_free referenced in function clear_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_ctr_drbg_random referenced in function asym_encrypt_sp_basic128rsa15
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_entropy_init referenced in function policyContext_newContext_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_entropy_free referenced in function clear_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_entropy_func referenced in function policyContext_newContext_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_entropy_self_test referenced in function policyContext_newContext_sp_aes128sha256rsaoaep
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_sha1_ret referenced in function mbedtls_sign_sha1
1>open62541.c.obj : error LNK2019: unresolved external symbol mbedtls_sha256_ret referenced in function asym_sign_sp_aes128sha256rsaoaep`

ffreality, 2022-05-24:
Thank you very much we solved our problem.

closed 2022-05-24 by ffreality

gokhalehemanth, 2022-06-02:
Hi, what was the solution for rectifying the link error for openSSL. I am also facing similar issue when i try to integrate openSSL on windows platform. Below are the Link error i get:

Error	2	error LNK2019: unresolved external symbol _OPENSSL_sk_num referenced in function _sk_GENERAL_NAME_num	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	3	error LNK2019: unresolved external symbol _OPENSSL_sk_value referenced in function _sk_GENERAL_NAME_value	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	4	error LNK2019: unresolved external symbol _OPENSSL_sk_new_null referenced in function _sk_X509_CRL_new_null	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	5	error LNK2019: unresolved external symbol _OPENSSL_sk_pop_free referenced in function _sk_GENERAL_NAME_pop_free	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	6	error LNK2019: unresolved external symbol _OPENSSL_sk_push referenced in function _sk_X509_CRL_push	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	7	error LNK2019: unresolved external symbol _CRYPTO_free referenced in function _UA_OpenSSL_LoadLocalCertificate	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	8	error LNK2019: unresolved external symbol _OPENSSL_init_crypto referenced in function _UA_Openssl_Init	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	9	error LNK2019: unresolved external symbol _BIO_new referenced in function _UA_CreateCertificate	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	10	error LNK2019: unresolved external symbol _BIO_free referenced in function _UA_CreateCertificate	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	11	error LNK2019: unresolved external symbol _BIO_ctrl referenced in function _UA_CreateCertificate	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	12	error LNK2019: unresolved external symbol _BIO_s_mem referenced in function _UA_CreateCertificate	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	13	error LNK2019: unresolved external symbol _BIO_new_mem_buf referenced in function _UA_OpenSSL_LoadPemCertificate	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	14	error LNK2019: unresolved external symbol _BN_new referenced in function _UA_CreateCertificate	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	15	error LNK2019: unresolved external symbol _BN_set_word referenced in function _UA_CreateCertificate	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	16	error LNK2019: unresolved external symbol _BN_free referenced in function _UA_CreateCertificate	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	17	error LNK2019: unresolved external symbol _ASN1_OCTET_STRING_cmp referenced in function _UA_CertificateVerification_Verify	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	18	error LNK2019: unresolved external symbol _ASN1_INTEGER_set referenced in function _UA_CreateCertificate	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	19	error LNK2019: unresolved external symbol _EVP_MD_CTX_new referenced in function _UA_OpenSSL_RSA_Public_Verify	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	20	error LNK2019: unresolved external symbol _EVP_MD_CTX_free referenced in function _UA_OpenSSL_RSA_Public_Verify	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	21	error LNK2019: unresolved external symbol _EVP_DigestUpdate referenced in function _UA_OpenSSL_RSA_Public_Verify	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	22	error LNK2019: unresolved external symbol _EVP_EncryptInit_ex referenced in function _UA_OpenSSL_Encrypt	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	23	error LNK2019: unresolved external symbol _EVP_EncryptUpdate referenced in function _UA_OpenSSL_Encrypt	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	24	error LNK2019: unresolved external symbol _EVP_EncryptFinal_ex referenced in function _UA_OpenSSL_Encrypt	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	25	error LNK2019: unresolved external symbol _EVP_DecryptInit_ex referenced in function _UA_OpenSSL_Decrypt	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	26	error LNK2019: unresolved external symbol _EVP_DecryptUpdate referenced in function _UA_OpenSSL_Decrypt	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	27	error LNK2019: unresolved external symbol _EVP_DecryptFinal_ex referenced in function _UA_OpenSSL_Decrypt	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	28	error LNK2019: unresolved external symbol _EVP_DigestSignInit referenced in function _UA_Openssl_RSA_Private_Sign	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	29	error LNK2019: unresolved external symbol _EVP_DigestSignFinal referenced in function _UA_Openssl_RSA_Private_Sign	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	30	error LNK2019: unresolved external symbol _EVP_DigestVerifyInit referenced in function _UA_OpenSSL_RSA_Public_Verify	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	31	error LNK2019: unresolved external symbol _EVP_DigestVerifyFinal referenced in function _UA_OpenSSL_RSA_Public_Verify	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	32	error LNK2019: unresolved external symbol _EVP_CIPHER_CTX_new referenced in function _UA_OpenSSL_Decrypt	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	33	error LNK2019: unresolved external symbol _EVP_CIPHER_CTX_free referenced in function _UA_OpenSSL_Decrypt	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
[... cut ...]
Error	101	error LNK2019: unresolved external symbol _X509_get0_subject_key_id referenced in function _UA_CertificateVerification_Verify	VSProject\OpcUa\OpcUa\open62541.obj	OpcUaServer
Error	102	error LNK1120: 100 unresolved externals	VSProject\OpcUa\Debug\OpcUaServer.exe	OpcUaServer
	103	IntelliSense: expected a declaration	VSProject\OpcUa\OpcUa\open62541.c	55598	1008	OpcUaServer
	104	IntelliSense: identifier "X509_PURPOSE_CRL_SIGN" is undefined	VSProject\OpcUa\OpcUa\open62541.c	65145	44	OpcUaServer
	105	IntelliSense: identifier "GEN_URI" is undefined	VSProject\OpcUa\OpcUa\open62541.c	65271	29	OpcUaServer

gauravbvelex, 2024-06-06:
@ffreality Can you please suggest how did you resolve this ?

ffreality, 2024-06-07:
> @ffreality Can you please suggest how did you resolve this ?
> 
> open62541.c(199649,52): warning C4090: 'function': different 'const' qualifiers Specific Line: get_pkey_rsa(privateKey), /* private key */

I am using MBedTLS 3.6.0 LTS or OpenSSL 1.1.1t (my projects is an Unreal Engine 5 plugin. So, I can't use other OpenSSL in packaged projects. But OpenSSL 3 gave this error)

I didn't anything special for MBedTLS but for OpenSSL 1.x I needed to give applink.c location from source code. Sometimes binary installations doesn't have that file.

ffreality, 2024-06-07:
> @ffreality Can you please suggest how did you resolve this ?
> 
> open62541.c(199649,52): warning C4090: 'function': different 'const' qualifiers Specific Line: get_pkey_rsa(privateKey), /* private key */

@gauravbvelex I am using MBedTLS 3.6.0 LTS or OpenSSL 1.1.1t (my projects is an Unreal Engine 5 plugin. So, I can't use other OpenSSL in packaged projects. But OpenSSL 3 gave this error)

I didn't anything special for MBedTLS but for OpenSSL 1.x I needed to give applink.c location from source code. Sometimes binary installations doesn't have that file.

For certificate creation, you have to use OpenSSL but it can be another software. You don't need to integrate it to open62541