open62541/open62541 #5052: Multi-threaded Client & high CPU usage

dpazj, 2022-04-04:
## Description

I am writing a multi-threaded client application that subscribes to a large number of resources and also performs large amounts of read/write requests.
Currently, I have a thread that calls `UA_Client_connect` once and then calls `UA_Client_run_iterate(client,0)` (protected by a mutex) in a loop to handle any incoming subscription related messages. This loop is causing the high cpu usage, but I cannot run `UA_Client_run_iterate` with any timeout > 0 as that will interfere with other threads performing read/write requests due to the mutex protection. 

My first question is that perhaps I am using the library incorrectly and there are mechanisms in place to get around such an issue? 

If this is not the case and this is currently a limitation of the client implementation: is there any planned work or work in progress for a multi-threaded/thread-safe client to support this use case? It might be the case that I would end up working on a contribution to allow for a multi-threaded/thread-safe client if this is something that would benefit the project.

Thanks in advance. 

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): [v1.2.4](https://github.com/open62541/open62541/tree/v1.2.4)
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 20.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

xydan83, 2022-04-11:
Hi!
If you are subscribed to many nodes, I think, you are having problem with the high load CPU because of problem, describe in my issue: https://github.com/open62541/open62541/issues/5053.

dpazj, 2022-04-11:
Sorry I should have mentioned that in this case there is high CPU load even when I am not subscribed to any nodes. I am sure that your issue will also be causing high load when I am subscibed to the nodes so thanks for bringing that to my attention!

xydan83, 2022-04-11:
> Sorry I should have mentioned that in this case there is high CPU load even when I am not subscribed to any nodes. I am sure that your issue will also be causing high load when I am subscibed to the nodes so thanks for bringing that to my attention!

Well, perhaps it is another one problem. So far I haven't encountered this problem yet.

closed 2022-10-20 by dpazj