open62541/open62541 #5053: Big complexity in the *processDataChangeNotification* function.

xydan83, 2022-04-04:
## Description

Hello!

I have same trouble with Client. 
_Originally posted by @xydan83 in https://github.com/open62541/open62541/issues/4318#issuecomment-1087318664_

If I try to subscribe to 30,000 nodes that change at the same time once a second, then this filtering algorithm causes a very heavy load on the processor core. There is a quadratic complexity O(n ^ 2). It is quite possible to remove this problem by replacing the second loop with a lookup in a dictionary (hash table).
And it would also be cool if you could make a callback function that would return all the changes at once, in which I agree with the author of this issue.

Is it possible to replace list macro with hashmap?

## Checklist

Please provide the following information:

 - [v] open62541 Version (release number or git tag): v1.3-rc2-ef2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [v] Operating system: 20.04.1
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-04-04:
Hey @xydan83 ,

that is a good observation.
My recommendation then is to use the binary tree implementation from /deps/ziptree.h instead of the linked list.
The zip tree is used for example in /plugins/ua_nodestore_ziptree.c.

xydan83, 2022-04-05:
> Hey @xydan83 ,
> 
> that is a good observation. My recommendation then is to use the binary tree implementation from /deps/ziptree.h instead of the linked list. The zip tree is used for example in /plugins/ua_nodestore_ziptree.c.

Thanks for the answer.
I recently tried using [hashmap] (https://github.com/tidwall/hashmap.c) and it gave good results. CPU core load is reduced to 2-3%.

xydan83, 2022-04-14:
@jpfr Hi!
I'm trying to use your **ziptree** instead of a linked list in **ua_client_intrnal.h** and **ua_client_subscription.c** and I need help.
What I have done:
1.I made changes to two structures
[image]

2.In **ua_client_subscription.c** I added a comparison function and defined a macro **ZIP_FUNCTION**.
[image]

3. Added ZIP_INIT
[image]

4.Added ZIP_INSERT into **ua_MonitoredItems_create**.
[image]

5. Added ZIP_FIND to search for **UA_Client_MonitoredItem** in both **processData Change Notification** and **processEventNotification** functions.
 
[image]

[image]

With this minimum set, I'm trying to subscribe to a **UaCPPServer** test server, where I added 100,000 dynamic nodes that change their values once per second, and I get a good result. CPU usage is 10-15% if I print results to screen or 5-10% if I print results to file. When testing with **30,000 nodes**, the load is approximately 3-5%, if the result is output to a file, similar to hashmap.

But the big problem is how can I remove **UA_Client_MonitoredItem** from the tree if I compare **clientHandle** but some functions use **monitoredItemId** field to lookup.
Well, if we are removing all items from the subscription, I can use ZIP_ITER.
[image]

And then ZIP_REMOVE
[image]

But how to find **UA_Client_MonitoredItem** in **ua_MonitoredItems_delete** and **UA_Client_MonitoredItems_modify** functions?
Both are used to look for **monitoredItemId** and not **clientHandle**

[image]
[image]

I think using ZIP_ITER would be a bad idea.

Can you give me some advice or a hint on how to solve this situation?

xydan83, 2022-04-15:
**Upd:** Well, I tried **ZIP_ITER** in the delete and modify functions and it looks really good despite being O(n) complexity. **100,000** monitored items are removed in about 1-2 seconds.

jpfr, 2022-04-20:
Thanks for the fix @xydan83 !

closed 2022-04-20 by jpfr