open62541/open62541 #5055: 1.3: build fails on linking `bin/tests/check_subscription_event_filter`

kloczek, 2022-04-06:
## Description
Looks like 1.3 build fails on linking `bin/tests/check_subscription_event_filter`

## Background Information / Reproduction Steps
```console
[ 78%] Linking C executable ../bin/tests/check_subscription_event_filter
cd /home/tkloczko/rpmbuild/BUILD/open62541-1.3/x86_64-redhat-linux-gnu/tests && /usr/bin/cmake -E cmake_link_script CMakeFiles/check_subscription_event_filter.dir/link.txt --verbose=1
/usr/bin/gcc -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -std=c99 -fPIC -DNDEBUG -Wl,-z,relro -Wl,--as-needed -Wl,--gc-sections -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -flto=auto -flto-partition=none -fuse-linker-plugin -Wl,--build-id=sha1 CMakeFiles/check_subscription_event_filter.dir/server/check_subscription_event_filter.c.o ../CMakeFiles/open62541-object.dir/src/ua_types.c.o ../CMakeFiles/open62541-object.dir/src/ua_types_encoding_binary.c.o ../CMakeFiles/open62541-object.dir/src/ua_types_print.c.o ../CMakeFiles/open62541-object.dir/src_generated/open62541/types_generated.c.o ../CMakeFiles/open62541-object.dir/src_generated/open62541/transport_generated.c.o ../CMakeFiles/open62541-object.dir/src_generated/open62541/statuscodes.c.o ../CMakeFiles/open62541-object.dir/src/ua_util.c.o ../CMakeFiles/open62541-object.dir/src/ua_timer.c.o ../CMakeFiles/open62541-object.dir/src/ua_connection.c.o ../CMakeFiles/open62541-object.dir/src/ua_securechannel.c.o ../CMakeFiles/open62541-object.dir/src/ua_securechannel_crypto.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_session.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_nodes.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_server.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_server_ns0.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_server_ns0_diagnostics.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_server_config.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_server_binary.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_server_utils.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_server_discovery.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_server_async.c.o ../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_networkmessage.c.o ../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_writer.c.o ../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_writergroup.c.o ../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_reader.c.o ../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_readergroup.c.o ../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_manager.c.o ../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_ns0.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_view.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_method.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_session.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_attribute.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_discovery.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_subscription.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_monitoreditem.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_securechannel.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_nodemanagement.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_discovery_multicast.c.o ../CMakeFiles/open62541-object.dir/src/client/ua_client.c.o ../CMakeFiles/open62541-object.dir/src/client/ua_client_connect.c.o ../CMakeFiles/open62541-object.dir/src/client/ua_client_discovery.c.o ../CMakeFiles/open62541-object.dir/src/client/ua_client_highlevel.c.o ../CMakeFiles/open62541-object.dir/src/client/ua_client_subscriptions.c.o ../CMakeFiles/open62541-object.dir/deps/libc_time.c.o ../CMakeFiles/open62541-object.dir/deps/pcg_basic.c.o ../CMakeFiles/open62541-object.dir/deps/base64.c.o ../CMakeFiles/open62541-object.dir/deps/aa_tree.c.o ../CMakeFiles/open62541-object.dir/deps/ziptree.c.o ../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_config.c.o ../CMakeFiles/open62541-object.dir/src_generated/open62541/namespace0_generated.c.o ../CMakeFiles/open62541-object.dir/src/ua_types_lex.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_subscription.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_subscription_monitoreditem.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_subscription_datachange.c.o CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/openssl/securitypolicy_openssl_common.c.o CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/openssl/ua_openssl_basic128rsa15.c.o CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/openssl/ua_openssl_basic256.c.o CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/openssl/ua_openssl_basic256sha256.c.o CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/openssl/ua_openssl_aes128sha256rsaoaep.c.o CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/openssl/ua_openssl_create_certificate.c.o CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/openssl/ua_pki_openssl.c.o CMakeFiles/open62541-testplugins.dir/__/arch/network_tcp.c.o CMakeFiles/open62541-testplugins.dir/testing-plugins/testing_clock.c.o CMakeFiles/open62541-testplugins.dir/__/plugins/ua_log_stdout.c.o CMakeFiles/open62541-testplugins.dir/__/plugins/ua_config_default.c.o CMakeFiles/open62541-testplugins.dir/__/plugins/ua_accesscontrol_default.c.o CMakeFiles/open62541-testplugins.dir/__/plugins/ua_nodestore_ziptree.c.o CMakeFiles/open62541-testplugins.dir/__/plugins/ua_nodestore_hashmap.c.o CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/ua_securitypolicy_none.c.o CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/ua_pki_none.c.o CMakeFiles/open62541-testplugins.dir/testing-plugins/testing_policy.c.o CMakeFiles/open62541-testplugins.dir/testing-plugins/testing_networklayers.c.o CMakeFiles/open62541-testplugins.dir/__/arch/posix/ua_architecture_functions.c.o -o ../bin/tests/check_subscription_event_filter  -lcheck /usr/lib64/libssl.so /usr/lib64/libcrypto.so -lm -lpthread -lrt -lsubunit
/usr/bin/ld: /tmp/cc5fA3bv.lto.o: in function `eventSetup':
/home/tkloczko/rpmbuild/BUILD/open62541-1.3/tests/server/check_subscription_event_filter.c:279: undefined reference to `UA_Server_createEvent'
/usr/bin/ld: /tmp/cc5fA3bv.lto.o: in function `inListOperatorValidation_fn':
/home/tkloczko/rpmbuild/BUILD/open62541-1.3/tests/server/check_subscription_event_filter.c:329: undefined reference to `UA_Server_triggerEvent'
/usr/bin/ld: /tmp/cc5fA3bv.lto.o: in function `betweenOperatorValidation_fn':
/home/tkloczko/rpmbuild/BUILD/open62541-1.3/tests/server/check_subscription_event_filter.c:329: undefined reference to `UA_Server_triggerEvent'
/usr/bin/ld: /tmp/cc5fA3bv.lto.o: in function `orderedCompareOperatorValidation_fn':
/home/tkloczko/rpmbuild/BUILD/open62541-1.3/tests/server/check_subscription_event_filter.c:329: undefined reference to `UA_Server_triggerEvent'
/usr/bin/ld: /tmp/cc5fA3bv.lto.o: in function `ofTypeOperatorValidation_fn':
/home/tkloczko/rpmbuild/BUILD/open62541-1.3/tests/server/check_subscription_event_filter.c:329: undefined reference to `UA_Server_triggerEvent'
/usr/bin/ld: /home/tkloczko/rpmbuild/BUILD/open62541-1.3/tests/server/check_subscription_event_filter.c:329: undefined reference to `UA_Server_triggerEvent'
/usr/bin/ld: /tmp/cc5fA3bv.lto.o:/home/tkloczko/rpmbuild/BUILD/open62541-1.3/tests/server/check_subscription_event_filter.c:329: more undefined references to `UA_Server_triggerEvent' follow
collect2: error: ld returned 1 exit status
make[2]: *** [tests/CMakeFiles/check_subscription_event_filter.dir/build.make:248: bin/tests/check_subscription_event_filter] Error 1
make[2]: Target 'tests/CMakeFiles/check_subscription_event_filter.dir/build' not remade because of errors.
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/open62541-1.3/x86_64-redhat-linux-gnu'
make[1]: *** [CMakeFiles/Makefile2:1310: tests/CMakeFiles/check_subscription_event_filter.dir/all] Error 2
```
Used CMake options:
```console
[tkloczko@devel-g2v x86_64-redhat-linux-gnu]$ cmake -L
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
[... cut ...]
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_FORCE_WERROR:BOOL=ON
UA_LOGLEVEL:STRING=300
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=REDUCED
```
## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux x86/64
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

JohannesKauffmann, 2023-05-25:
Still there on v1.3.5. On master, it compiles fine with `UA_FORCE_WERROR:BOOL=OFF`.

@jpfr What is the policy for backporting fixes from master to versioned branches?

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier