open62541/open62541 #5059: Version 1.3 Nodeset example build error with CMAKE

estibalitz, 2022-04-11:
## Description

## Background Information / Reproduction Steps
Building nodeset examples in version 1.3. fails in my machine. Version 1.2 did also fail doing so. The error referes to an atribute of the addres map uploaded by .xml. (https://github.com/open62541/open62541/commit/69f1c5823ae4c7b3ddb0deeca0e1c5e777041b3b). The OS used is Windows 11 but with the WSL2 (Linux 5.10.102.1-microsoft-standard-WSL2 x86_64 x86_64 x86_64 GNU/Linux). I do not use any OPC UA SDKs in this process and the following error is obtained while building nodeset examples.  [[open62541]with]([open62541](https://github.com/open62541/open62541)/[examples](https://github.com/open62541/open62541/tree/master/examples)/nodeset/) actual CMakeLists without any modification. 

[ 72%] Linking C executable bin/examples/server_instantiation
[ 74%] Linking C executable bin/examples/tutorial_server_firststeps
[ 76%] Linking C executable bin/examples/client_connect
[ 78%] Linking C executable bin/examples/server_settimestamp
[ 80%] Linking C executable bin/examples/server_inheritance
[ 82%] Linking C executable bin/examples/tutorial_server_variable
[ 85%] Linking C executable bin/examples/client_async
[ 87%] Linking C executable bin/examples/tutorial_client_firststeps
[ 89%] Linking C executable bin/examples/server_loglevel
[ 91%] Linking C executable bin/examples/client
[ 93%] Linking C executable bin/examples/server_ctt
[ 93%] Built target tutorial_datatypes
INFO:__main__:Preprocessing /home/usr/open62541/examples/nodeset/server_nodeset.xml
Traceback (most recent call last):
  ****File "/home/usr/install/share/open62541/tools/nodeset_compiler/nodeset_compiler.py", line 169, in <module>
    ns.generateParser(args.existing, args.infiles, args.bsdFile)
  File "/home/usr/install/share/open62541/tools/nodeset_compiler/nodeset.py", line 431, in generateParser
    if (nd.hasAttribute("SymbolicName") and (re.match(r".*_BinarySchema", nd.attributes["SymbolicName"].nodeValue) or nd.attributes["SymbolicName"].nodeValue == "TypeDictionary_BinarySchema")) or (nd.hasAttribute("ParentNodeId") and not nd.hasAttribute("SymbolicName") and re.fullmatch(r"i=93", nd.attributes["ParentNodeId"].nodeValue)):
AttributeError: 'module' object has no attribute 'fullmatch'****
make[2]: *** [nodeset/CMakeFiles/server_nodeset.dir/build.make:70: src_generated/open62541/namespace_example_generated.c] Error 1
make[1]: *** [CMakeFiles/Makefile2:704: nodeset/CMakeFiles/server_nodeset.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 93%] Built target tutorial_server_firststeps
[ 93%] Built target server_mainloop
[ 93%] Built target client_connect_loop
[ 93%] Built target server_loglevel
[ 93%] Built target tutorial_server_datasource
[ 93%] Built target tutorial_server_variable
[ 93%] Built target client_connectivitycheck_loop
[ 93%] Built target client_async
[ 93%] Built target server_repeated_job
[ 93%] Built target tutorial_client_firststeps
[ 93%] Built target server_instantiation
[ 93%] Built target custom_datatype_client
[ 93%] Built target client_connect
[ 93%] Built target server_ctt
[ 93%] Built target client
[ 93%] Built target server_inheritance
[ 93%] Built target server_settimestamp
[ 93%] Built target tutorial_server_variabletype
[ 93%] Built target tutorial_server_object
[ 93%] Built target custom_datatype_server
make: *** [Makefile:130: all] Error 2

Used CMake options:
No change was done expect the path to the actual instalation folder. 
-- Cache values
CMAKE_BUILD_TYPE:STRING=
CMAKE_INSTALL_PREFIX:PATH=/usr/local
open62541_DIR:PATH=/home/usr/install/lib/cmake/open62541
open62541_NODESET_DIR:PATH=/home/usr/install/share/open62541/tools/ua-nodeset
open62541_TOOLS_DIR:PATH=/home/usr/install/share/open62541/tools

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [x] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mister-potter, 2022-04-11:
@estibalitz 
[fullmatch()](https://docs.python.org/3/library/re.html#re.fullmatch) is introduced in Python since 3.4 only. If the stack uses Python 2 executable for nodeset compilation, it may result in the error you mentioned. To avoid this, update your python symlink to point to version 3 by default.

Python 3 update alternative:
` sudo update-alternatives --install $(which python) python $(which python3) 10`

estibalitz, 2022-04-11:
Hi mister-potter,

I have just tried the command you suggested and it builds perfectly. Thank you very much.

closed 2022-04-11 by estibalitz

frameworker2019, 2022-10-26:
For me just calling python3 instead of python solves the problem:
```
root@7f457957c1c3:/build/open62541-server/xml#  python3 nodeset_compiler/nodeset_compiler.py --types-array=UA_TYPES --existing ua-nodeset/Schema/Opc.Ua.NodeSet2.xml --xml myNS.xml myNS
INFO:__main__:Preprocessing (existing) ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing myNS.xml
INFO:__main__:Generating Code for Backend: open62541
INFO:__main__:NodeSet generation code successfully printed
```

eziosella, 2023-01-17:
> For me just calling python3 instead of python solves the problem:
>
> ```
> root@7f457957c1c3:/build/open62541-server/xml#  python3 nodeset_compiler/nodeset_compiler.py --types-array=UA_TYPES --existing [...]

Yes, me too!

I was following the instructions:
>https://www.open62541.org/doc/0.3/nodeset_compiler.html

and I have got the same problem, after read this issue, using python3 has worked perfectly!